WEC-Sim/WEC-Sim #454: Simulation failing after 14 seconds

smervosh, 2020-11-09:
Hello!

I am trying to run a simulation of a sphere and keep running into the following error after about 14 seconds:

Error using wecSim (line 292)
['BETA/Global Reference Frame/Solver Configuration']: Error evaluating equations at time 14.750000000000002.  Stopping simulation.
There may be a singularity in the solution.  If not, try reducing the step size (either by reducing the fixed step size or by
tightening the error tolerances)
Caused by:
    Error using wecSim (line 292)
    ['BETA/Floating (3DOF)/CONSTRAINT (Planar Joint)']: 'BETA/Floating (3DOF)/CONSTRAINT (Planar Joint)' has a degenerate mass
    distribution on its follower side.

Any help with this error would be greatly appreciated! I believe I may have made in error in how I initialized my model but I have been unable to figure out exactly where I have gone wrong from looking at it myself and with a teammate.

[BETA.zip](https://github.com/WEC-Sim/WEC-Sim/files/5507500/BETA.zip)

sugarl14, 2020-11-11:
I'm not an expert in WEC-Sim, but in the past I have seen it suggested to reduce simu.dt from 0.1 to 0.01. 
simu.dt can be found around line 10 of the wecSimInputFile.m code.

Or, possibly try changing the wave height wave period values. Maybe try something like H=1 and T= 5?

If that doesn't work, maybe check your simulink file to make sure you have the "base" and "follower" connections properly defined.

dforbush2, 2020-11-11:
Hi @smervosh 

I am unfortunately unable to open your simulink model (I'm currently running 2020a, its in 2020b). Is it possible you could re-upload an earlier version?

In the mean time, it appears that you have both a PTO and a 3DOF floating constraint defined in your wecSimInputFile with only a single body. I'm assuming it is connected something like this:

[image]

This can lead to the error you're seeing a lot because the multi-body solver is trying to resolve the motion between the PTO and the Constraint block, which doesn't have any inertia. This leads to a "divide by zero" sort of situation, and can cause that error. Could you describe in a bit more detail what it is you are trying to model? I am assuming this is a 3-DOF floating sphere with a PTO that acts in heave only.

If I've got this right then the quickest fix is to make another dummy body block (body(2)) between the PTO and the Constraint blocks.  This body is not excited by wave motion and is neutrally buoyant, so its a source of non-zero inertia and nothing else.

[image]

And then add this to the wecSimInputFile:

`body(2) = bodyClass('');  % no hydroData needed
body(2).nhBody = 1; % this says it's a non-hydro body
body(2).name= 'dummy'; 
body(2).mass = 1; % 1 kg, and displacement of 1/1025 m^3 --> neutral buoyancy
body(2).momOfInertia = [1 1 1];
body(2).dispVol = 1/1025; % m^3
body(2).cg = [0 0 0]; % coincident with big sphere
body(2).cb = [0 0 0]; % coincident with big sphere
body(2).viz.color = [0.8 0 0]; % makes red for visualization`

Apologies if this is off-base...I'm attempting help without seeing your model :)

dforbush2, 2020-11-11:
Two other ways to do this: 

1). Use a mooring matrix as a proxy for your PTO. Add this to your wecSimInputFile and a mooring matrix block to your model.

Note that the vertical spring force on the mooring block will act as your PTO.

% Mooring Matrix
mooring(1) = mooringClass('mooring');       % Initialize mooringClass
mooring(1).matrix.k = zeros(6,6);
mooring(1).matrix.k(3,3) = 0.0054; % applies spring force in heave, like PTO
mooring(1).matrix.c = zeros(6,6);
mooring(1).matrix.preTension = zeros(1,6);

[image]

2). USE CAUTION AS THIS WILL BREAK YOUR LIBRARY LINK: Apply the PTO spring force to the internal mechanics of the 3DOF constraint block.

[image]

smervosh, 2020-11-11:
Thank you @sugarl14 and @dforbush2  you guys have been a lot of help!

I have tweaked my model to include a mooring matrix and removed the PTO. This method has seemed to work the best out of everything I have tried so far. My simulation can make it to 17.6 seconds now until I get this error:

Error using wecSim (line 292)
Derivative of state 'BETA.Floating_3DOF.CONSTRAINT_Planar_Joint.Px.v' in block 'BETA/Floating (3DOF)/CONSTRAINT (Planar Joint)' at
time 17.6 is not finite. The simulation will be stopped. There may be a singularity in the solution.  If not, try reducing the step
size (either by reducing the fixed step size or by tightening the error tolerances)

I will include both a 2020b and 2020a version of my Simulink model this time.

[BETA.zip](https://github.com/WEC-Sim/WEC-Sim/files/5527117/BETA.zip)

smervosh, 2020-11-11:
Also, just for reference, I'm using this program for a senior design project. My team and I are creating a wave energy converter with a spherical float connected to a spring-loaded shaft.

dforbush2, 2020-11-12:
Ah I can see a few things now that I can open the model. Thanks for the version swap.

Firstly, I am able to run your model as-is by switching to a variable time-step solver (simu.solver = 'ode45'), so nice call @sugarl14, since this is effectively reducing the solver time step. 

However, I am getting some very strange dynamics from your model: excitation does not seem to align with your excitation frequency. I suspect there is a disagreement in your dimensions somewhere....for instance, you specify a body mass of 8.3 kg, but a displaced volume of 216 m^3. Secondly, although your *.stl file does not affect calculations, running with simu.explorer = 'on' will show you an animation of your sphere in the wave field, and will highlight what I mean about possible dimension issues. This could present a problem for calculations, however, if your mesh used for BEM calculations was generated from this *.stl or vice versa.

Additionally, unless your device is completely free-floating, I would recommend you add some small restoring force in surge to your mooring (i.e., mooring(1).matrix.k(1,1) = 0.005) so that the device does not continually drift to one side.

dforbush2, 2020-11-12:
I would also try running a longer wave period. These will be more representative of actual sea-states (assuming this is a full scale device). I got better looking results with T = 6 s, so I would recommend using long waves relative to device spatial extent for your debugging purposes. Otherwise, the spatial extents of your device are so large that on short-period waves, multiple  wave crests are interacting with your device at once, and it will make your output results more difficult to interpret, in addition to being on shakier ground with respect to the assumptions inherent to linear solution methods.

smervosh, 2020-11-13:
Oh I see. I had not noticed the issue with the volume. That is concerning since the device I'm trying to simulate is only 0.12 m^3. Where should I go in the code to fix the issue of the volume of the sphere?
The .stl was made with this volume of 0.12 m^3. Perhaps that is what is causing the issues?

dforbush2, 2020-11-13:
This normally is inherited from your *.h5 data file...taking a look in your *.zip folder, this is indeed the case. This is concerning because the hydrodynamic coefficients dictating hydrostatics, excitation, and radiation are calculated from this mesh, so I would recommend double-checking your boundary element method run. If your mesh is problematic I would recommend you take a look at MeshMagick https://lheea.github.io/meshmagick/command_line/cli_usage.html#translations, which can handle scaling and transformations pretty easily, especially for a simple geometry like a sphere. A sphere also has some semi-empirical solutions for excitation and radiation coefficients: I would check https://www.researchgate.net/publication/245195840_Radiation_and_diffraction_of_water_waves_by_a_submerged_sphere_in_finite_depth, and some of the cited works if you pursue this avenue.

An additional option: check the Free-Decay cases of the WEC-SimApplications repository, which uses a sphere geometry https://github.com/WEC-Sim/WEC-Sim_Applications 

You can specify displacement directly by specifying in your wecSimInputFile: body(1).dispVol = 0.12, but I would highly recommend that you double-check the hydrodynamic coefficients of your BEM run first.

dforbush2, 2020-11-13:
Note that your *.stl file is used ONLY for visualization (and often the units are not inherited: I'm guessing that you perhaps modeled in mm, but outside of solidworks, these will usually be interpreted as m) unless you are using the non-linear hydro option (simu.nlHydro = 2). So to revise your hydrodynamics, you'll need to modify the mesh file used in your BEM code.

smervosh, 2020-11-16:
Ok sweet thank you I'll try that. Do you recommend any BEM code in particular to use?

sugarl14, 2020-11-16:
@smervosh 
I've been using ANSYS AQWA and I'm a pretty big fan of it. It has a very user friendly GUI, and your university may even have an ANSYS subscription. Reach out to your school's tech desk to see if your school has an academic license. 

I know a lot of people in industry use WAMIT. There's a ton of info on the internet you can find. But you'd need to purchase a license. The last BEM option that is compatible with WEC-Sim is NEMOH, which is also code based, and is free. Code based things (for me) are little more complicated to understand conceptually, especially if you're new to BEM (that's why I prefer AQWA). But it all does the same thing.

Let me know if y'all end up using AQWA. I'm more than happy to be of any assistance to y'all. Good luck!

dforbush2, 2020-11-16:
I second @sugarl14 suggestions. It is really likely that your University has licenses available for commercial such as AQWA or WAMIT. NEMOH is also useful, has a MATLAB interface, and for asymmetric geometries like a sphere, has mesh-creation utilities. 

Whatever you end up using, you can check against the BEM data found here: https://github.com/WEC-Sim/WEC-Sim_Applications/tree/master/Free_Decay. These also used a sphere, albeit of different scale, and should be characteristically similar.

dforbush2, 2020-11-19:
Hi @smervosh 

I am going to close this issue for now. Feel free to re-open or open another issue if necessary as your work progresses. Good luck!

closed 2020-11-19 by dforbush2

hossein-mehdipour, 2020-11-20:
@dforbush2 I had a question if you don't mind,
Is WEC-Sim code Stiff? (I mean can we use stiff solvers for it?)

dforbush2, 2020-11-20:
Most features were successfully tested a couple of years ago using I believe ode15s alongside the standard ode4 and ode45, so in most cases I suspect the answer is yes. Please let us know if you encounter issues using stiff solvers.