WEC-Sim/WEC-Sim #455: Docs Restructuring Proposal

H0R5E, 2020-11-10:
In #369, I said that I thought the organisation of the documentation could be better. I wanted to put an issue together to discuss how I think that reorganisation might happen and whether the development team think what I have in mind is good / bad / needs improvement.

I would like to split the documentation into 3 tables of content (TOCs), rather than just one as it is at the moment, which reflect the broad scope of the documentation. These three TOCs would be "Introduction", "User Manual" and "Development" (I'm not wedded to these titles). Below these categories, I think the docs should be organised like this:

* Introduction
    -  Overview
        + This is a new section that would describe things like the advantages of using WEC-Sim over other methods
        + Show some examples of what WEC-Sim can do
    - Webinars
    - Publications
    - Acknowledgements
    - Release Notes
        + Not sure about where this one should go...
* User Manual
    - Getting Started
    - Tutorials
        + I always think tutorials should be upfront and self-supporting. They should offer a taster and direct people to dive deeper, where needed.
    - Workflow
        + Was Overview, but with WEC-Sim Development removed
    - Code Structure
    - Advanced Features
    - Theory
    - Terminology
* Development
    - Overview
        + This can be the WEC-Sim Development section for starters
    - API
       + My feeling is that this is more aimed at developers rather than the user. The Code Structure section can direct here, if necessary.
    - License

For me, this (mostly) neatly splits the documentation into three types of people: people who want to know about WEC-Sim, but maybe not use it, yet; people who want to use WEC-Sim, but not develop it; and people who want to develop WEC-Sim.

Interested to hear your thoughts.

akeeste, 2020-11-17:
I like your idea of restructuring the documentation to better reflect who is viewing it. Introducing the code and its capabilities upfront will be a good improvement and will better attract new users. This is a good opportunity to make use of the applications repo, as it already contains many examples of what WEC-Sim can do. With some more description and Paraview visualizations, the samples from the applications repo would fit well in an Introduction/Overview/Examples section.

The Development/Overview section can include more information on the team's workflow as I started on in PR #453. It can include instructions on best practices for fork-based workflows, PRs, etc so that any external developments are easily integrated.

kmruehl, 2020-12-17:
@akeeste and @H0R5E is this issue resolved or do we want to keep it open to consider this docs restructuring?

H0R5E, 2020-12-17:
No, I think it's still for consideration. I just don't want to commit to such a big job without a clear mandate!

akeeste, 2020-12-18:
I agree we should still consider this. I am willing to help out with this and work on a couple samples of new organizations. @kmruehl Should we go ahead and get started on this?

kmruehl, 2020-12-18:
@akeeste and @H0R5E sounds good to me. Let's keep it open and discuss when/how to implement these changes during our next meeting.

kmruehl, 2021-01-06:
Based on discussion with @WEC-Sim/wec-sim-lab-team on 1/6/21

* Introduction
    -  Overview
        + This is a new section that would describe things like the advantages of using WEC-Sim over other methods
        + Show some examples of what WEC-Sim can do
    - Webinars
    - Publications
    - Acknowledgements
    - Release Notes
        + Not sure about where this one should go...

* Theory 

* User Manual
    - Getting Started
    - Tutorials
        + I always think tutorials should be upfront and self-supporting. They should offer a taster and direct people to dive deeper, where needed.
    - Workflow
        + Was Overview, but with WEC-Sim Development removed
    - Code Structure
    - Advanced Features
    - Terminology
* Development
    - Overview
        + This can be the WEC-Sim Development section for starters
    - API
       + My feeling is that this is more aimed at developers rather than the user. The Code Structure section can direct here, if necessary.
    - License

H0R5E, 2021-01-07:
So, I think the only difference here is that the Theory section has been moved to the top level?

 I can see the case for having a top level "theory manual", but I believe it should go after the user manual. If the idea of putting it above the user manual is to "encourage" users to read it before using the code, then I think that's misguided - they will just skip over it.

I also don't think all theory should be isolated from the user manual. Clearly, some theoretical information is important to know to be able to use the tool, but I think the deeper stuff should be separated and made as an optional for people who really want to know.

kmruehl, 2021-01-07:
@H0R5E Yes, I think there are a few other minor comments, but that's the biggest change. Definitely open to swapping the order, and putting Theory after User manual, but we liked the idea of having a separate Theory manual that we could cross reference from the manuals, etc. It's more of a stand-alone section as it is.

H0R5E, 2021-01-07:
@kmruehl, thanks for the heads up, seems sensible to me. Shall we discuss implementing this in the meeting next week, or would you like me to start putting it together now?

dav-og, 2021-01-07:
Hi Matt,

Just adding a couple of points:

> I can see the case for having a top level "theory manual", but I believe it should go after the user manual. If the idea of putting it above the user manual is to "encourage" users to read it before using the code, then I think that's misguided - they will just skip over it.

IMO having the structure as "intro -> theory -> user -> developer" is just a logical, familiar structure that makes it easier for users to navigate the docs and quickly find what they are looking for 👍. Also, in the User section, we will refer back to the theory - so IMO it makes more sense that the Theory section would come first.

> I also don't think all theory should be isolated from the user manual. Clearly, some theoretical information is important to know to be able to use the tool, but I think the deeper stuff should be separated and made as an optional for people who really want to know.

Personally I would prefer a consolidated Theory section, and just refer back to this from the User section when/where appropriate. I don't like the idea of have theoretical content separated into different sections as it may become harder for the user to find what they're looking for (quickly, at least).

Cheers :)
Dave

H0R5E, 2021-01-13:
Dave, thanks for the feedback. I just wanted to offer a counter opinion, before the meeting today, to help with any discussion we might have about this.

> IMO having the structure as "intro -> theory -> user -> developer" is just a logical, familiar structure that makes it easier for users to navigate the docs and quickly find what they are looking for .

I wouldn't necessarily agree that this is the case for user documentation. I can find several high profile packages where their theory sections are somewhat later than their user sections. See, e.g. [Dakota](https://dakota.sandia.gov/content/manuals) or [OpenFOAM](https://www.openfoam.com/documentation/). No doubt there are counter examples, but I believe user documentation should present the user with the most relevant information as quickly as possible, and unless they are developers of the code or need to understand more about their results, they shouldn't have to know the exact mechanics of the algorithms at work.

>Also, in the User section, we will refer back to the theory - so IMO it makes more sense that the Theory section would come first.

I'm not sure this is relevant in an age of hyperlinking. And then even in an academic paper you might refer forward to an appendix. Ultimately then, it doesn't matter about the order because the user will choose where to start in the documentation, but I would bet that time limited ocean energy companies are more likely to read the user manual before the theory manual.

> Personally I would prefer a consolidated Theory section, and just refer back to this from the User section when/where appropriate. I don't like the idea of have theoretical content separated into [...]

I think there are two "types" of theory for software - the theory of _what_ the software does and the theory of _how_ the software does it. I would argue that the _what_ part is important to present within the user manual, because that's important for the user to know (i.e these inputs transform to those outputs). On the other hand, I think the _how_ theory is only relevant to those that want to take a deeper dive to understand results or to those that want to develop the code. This part of the documentation is important as some kind of "static" verification of the algorithms, certainly, but not every user will want to check it for themselves, I think.

Ultimately, there is a bit of the [hard vs soft wrapping](https://github.com/SNL-WaterPower/WecOptTool/pull/158) about the debate, and I'm definitely a hard wrapping kind of person, but, you know, I think there are compelling arguments for it. :poodle:

kmruehl, 2021-01-13:
Based on discussion with @WEC-Sim/wec-sim-lab-team on 1/13/21

* Introduction
    -  Overview
        + This is a new section that would describe things like the advantages of using WEC-Sim over other methods
        + Show some examples of what WEC-Sim can do
        + Highlight some applications cases with images and the repo 
        + Highlight some WEC-Sim features
    - Webinars 
    - Publications
    - Acknowledgements
    - Release Notes
        + Not sure about where this one should go...
    - License
* Theory 
    - Terminology
* User Manual
    - Getting Started (and running example)
    - Workflow
        + Was Overview, but with WEC-Sim Development removed
    - Tutorials
        + I always think tutorials should be upfront and self-supporting. They should offer a taster and direct people to dive deeper, where needed.
    - Applications (pulled out from Tutorials)
    - Code Structure
    - Advanced Features
* Development
    - Overview
        + This can be the WEC-Sim Development section for starters
    - API
       + My feeling is that this is more aimed at developers rather than the user. The Code Structure section can direct here, if necessary.

H0R5E, 2021-01-13:
I'm happy to get this started, if you want.

kmruehl, 2021-01-16:
@H0R5E yes, please do!

H0R5E, 2021-01-20:
@kmruehl, I will try and get the basics of this set up on branch of my fork before next week's working meeting. I probably can't do much more in this quarter, but I will give @akeeste write access to the fork, so that he can contribute to it. Just give a me buzz if you want anyone else to be added.

H0R5E, 2021-01-22:
OK, so I have the new doc structure set up on the [new_docs_structure](https://github.com/H0R5E/WEC-Sim/tree/new_docs_structure) branch of my fork. That branch is building automatically onto my github pages for both the [original docs](https://h0r5e.github.io/WEC-Sim/) which contain a link on the frontpage to [the new docs](https://h0r5e.github.io/WEC-Sim/new/index.html). I just wanted to have them side by side, for now, so we could compare and contrast.

The new docs source code is in the `docs_new` directory of the `new_docs_structure branch` and new stuff should be added in there. I've tidied up the existing content so that it builds without error. On every push to the `new_docs_structure` branch in my fork, the docs will automatically rebuild.

So far I've given @akeeste  and @kmruehl write access. I can talk you through it on Monday, if that would be helpful.

akeeste, 2021-02-04:
Hi @H0R5E
You intended for us to push updates for the new structure to your branch right? I'm having trouble pushing my commits right now. The error when pushing is:
"...
remote: Resolving deltas: 100% (53/53), completed with 17 local objects.
To https://github.com/H0R5E/WEC-Sim/
 ! [remote rejected] new_docs_structure -> new_docs_structure (permission denied)
error: failed to push some refs to 'https://github.com/H0R5E/WEC-Sim/' "

All I'm using is
git push h0r5e new_docs_structure

The remote should be set-up just fine:
$ git remote -v
h0r5e   https://github.com/H0R5E/WEC-Sim/ (fetch)
h0r5e   https://github.com/H0R5E/WEC-Sim/ (push)

Can you check the permissions? If its an issue I could pull your branch to one of mine and push the changes there.

H0R5E, 2021-02-05:
@akeeste, I sent you an invitation to collaborate on the repo, but you haven't responded to it. Until you accept it, you won't be able to push.

akeeste, 2021-02-08:
@H0R5E Sorry I must have missed that invite from a couple weeks ago. I found the email but it has expired. Can you add me as a collaborator again? I will watch for it this time

H0R5E, 2021-02-08:
@akeeste, no problem, I've just sent it again.

H0R5E, 2021-02-10:
@akeeste, I just fixed the build issues (be careful with case sensitivity in your image paths) and tidied up a little, so make sure you pull down my changes, otherwise you'll have a tricky merge on your hands.

kmruehl, 2021-05-27:
Closing this based on #557

closed 2021-05-27 by kmruehl