WEC-Sim/WEC-Sim #457: Error reading geometry .stl file

t-osu, 2020-11-12:
Hi all,
I'm having troubles modeling my device. All the errors occur due to my non-hydro body.
Yes, it is specified as a non-hydro body, and due to other errors, the mass, displaced volume, COG, and moment of inertia have all been defined. The .h5 file string is left as blank as indicated in webinars.

Now the problems:
1) The .stl geometry file cannot be read, yielding error "['Pend_simplified/Pendulum/Non-Hydro Body/Structure/Body Properties']: The unit could not be read from the file  \\depot.engr.oregonstate.edu\users\dizonc\Windows.Documents\My
    Documents\MATLAB\WEC-Sim-4.1\ORISE\Pend_simplified\geometry\Pend_real.STL. A unit should be specified in the block instead. Resolve this issue in order to simulate the model."

2) Upon commenting this out I get the error "(Revolute Joint)' has a degenerate mass distribution on its follower side." occurring from one of my constraint blocks. I'm hoping this can be resolved once the .stl file can be read (?).

I create my .stl file from my Solidworks model (just save as-> file type -> .stl) and while this gives me a disproportionate size visual compared to the water when WEC-Sim simulates, it still loads.
It is only this one part that cannot be loaded. 

My model is quite small because I'm modeling a small WEC. The hull is a cylinder 0.8m diameter, 0.4m high (this loads fine), and the pendulum is 0.35m length and 0.1m depth, and 0.05m high (this does not load). The hull has a small 0.05m cylinder extruding 0.05m out the top that the pendulum "swings" around in yaw. 
Could the small size be an issue? 
Thanks, haven't found much online that I haven't already tried!
- Chris

nathanmtom, 2020-11-12:
@t-osu Thanks for describing your issue and hopefully our team can help find a solution. 

First, can you provide more details on the model you have built? It sounds like you have a two- or -three body system that is likely connected through 2 or more joints/constraints? In the past our team has observed that if you connect PTOs or constraints in series the 2nd error you listed comes up because there is no inertia to be included in the PTO and constraint equations. Placing an inertial block between the two PTOs/constraints with a negligible mass has more often than not solved that issue. 

Your first question is a bit more difficult to answer as it would appear somehow the .stl is either scaled incorrectly or MATLAB is having difficulty finding the file you are pointing to in the input file. 

If possible, could you share your working files so that our team might be able to reproduce the error and work to find a fix?

Cheers, 
Nathan
National Renewable Energy Laboratory

t-osu, 2020-11-12:
Thanks for the speedy reply @nathanmtom!
I'll provide all my files through a Google Drive link, including a picture of my design, the main WEC-Sim files (zipped), and some other files that might be helpful (Solidworks and AQWA).
https://drive.google.com/drive/folders/13QlMK2jlfNpvJnpS2pGEE9LToPaH0Wj5?usp=sharing

**note:** I also want to stress that this isn't the final shape or anything. I just want to get a simple model of simulating a pendulum WEC in WEC-Sim before moving on to more complicated shapes/designs/control schemes/etc. This could mean it might behave oddly, but that's on my end to fix with mass/COG/COB/and other stability issues, or maybe it will only work moored, etc.

1) Design
In the link you'll find _design.jpg_. My thesis is designing a small WEC that will help monitor HABs at sea. We decided on using a vertical-axis pendulum design given its simplicity and robustness. In _design.jpg_ the pendulum is modelled as a cylinder with another cylinder extruding out the top that the pendulum swings around in yaw. 

2) Design represented in Simulink
If you look at my Simulink file, _Pend_simp.slx_ I have 2 rigid bodies, body(1) the hull, body(2) the pendulum. I use a rotational PTO constraint to attach the pendulum to the hull so it can swing around in yaw. I want the hull to be able to move in pitch and roll, and I read on the WEC-Sim homepage that one could connect two constraints in series to make this happen, but I guess that is not allowed anymore from your comment?

3) wecSimInputFile
In _wecSimInputFile.m_ I'm starting out with no waves just to make sure my body doesn't behave out of the ordinary (like float away to space!). Of course, given my note, maybe my shape/mass/COG&COB just won't allow this and it needs to be changed or moored. I'll deal with that later, for now, getting a running simulation is priority.
**bodies**
You can see that body(1) has its .h5, .stl, mass, and MOI defined. body(2) is defined as a nonhydro and is as described in my first post.
**constraints**
constraint(1) is my roll, and is re-oriented as such by:
constraint(1).orientation.y = [1,0,0];
constraint(1).orientation.z = [0,1,0];
constraint(2) is pitch and left as is since this is the default.
**pto**
pto(1) is yaw and re-oriented as such by:
pto(1).orientation.y = [0,0,1];
pto(1).orientation.z = [0,1,0];
since it is located at the top of the cylinder, the location is:
pto(1).loc = [0 0 0.4];

4) Noteworthy items
**.stl size**
I create my _.stl_ file using Solidworks because that is the only modelling program I have access to. I have put other designs in the past into WEC-Sim and it's always the same thing; the visual is HUGE...I am at a loss as to what the issue could be, just the way Solidworks makes the _.stl_ file? My friend used Rhino and it got imported fine, however I don't have a license for that.

**AQWA**
I would like to use WAMIT, but I'm awaiting a license so to all WAMIT lovers, take it easy on me! My frequency solutions aren't exactly pretty, but again, this is an item I'll work on later with COG/COB/shape/stability/etc

I appreciate the response and willingness to re-create the error. Please let me know if there is any other information I can provide.
- Chris

t-osu, 2020-11-14:
Just an update. Dimensions shouldn't be an issue, but I made everything 5x bigger (no fix) and then got rid of the extrude cut in the pendulum so it is now just a rectangle and offset it so it's not interfering with the extrude of the hull (still no fix), but at least that's tried.

nathanmtom, 2020-11-16:
@t-osu Thank you for sharing your files. I'll do my best to answer your questions and provide guidance. 

First, can you confirm that you want your main hull to only roll and pitch, rather than use a 6 DOF joint between the seabed and the hull?

My previous comment did not imply you cannot use two constraints in series, only that often when you have two constraints in series with no inertia in between, be it a hydrodynamic body or mass, then we often get the degenerate mass issue you listed in your initial issue post. 

After downloading your files, I unfortunately was unable to reproduce your error as the simulation did run as provided. I am using MATLAB version 2020b, so perhaps you are using a different MATLAB version. If so, let us know what it is and we can see if the issue is with a specific MATLAB version. 

From reviewing the run, it appears that the model oscillates in roll a full 360 degrees which make it appear the mass properties or the hydrodynamics are causing constant rolling despite having no wave excitation. I'm not sure if you have already done this check, but perhaps simplify your model to just your hull and re-run the `noWaveCIC` case to see if you have expected motion with the roll and pitch constraints in series. It would also be possible to replace the roll and pitch constraints with a [spherical joint](https://www.mathworks.com/help/physmod/sm/ref/sphericaljoint.html) from MATLAB SimMechanics. Although, not a default constraint in the WEC-Sim library, you can replace this block under the rotational constraint mask. This may simplify the number of constraints you need and perhaps be closer to modeling the freely floating system. 

I hope this helps guide you towards next steps and update us on progress to see if we can help continue troubleshooting your issue. 

Cheers, 
Nathan 
National Renewable Energy Laboratory

t-osu, 2020-11-17:
@nathanmtom thanks for the input, I've been playing around with the simulation the past few days.
After I downloaded Matlab 2020b, the simulation still produced the same error. I did delete the files and then re-download them from the shared folder and it started to work. Don't know how that fixed it, but glad it did.

Ideally the body will move in only roll and pitch. I made everything bigger and this produced better frequency results in AQWA which I am now using in the simulation.

With only the hull under `noWaveCIC` it is stable.

I have resolved the _.stl_ Solidworks issue after following @enrjcutl solution from https://github.com/WEC-Sim/WEC-Sim/issues/391, thanks!

I am wondering about 3 other items though. 
1) As the pendulum rotates, so will the COG, but this location needs to be defined, yet I cannot let it be fixed to one spot. Is there a way around this? I also noticed if you change the COG location, it moves the part in the animation...but I guess you can just offset the model in Solidworks.

2) Pto.loc sometimes works and sometimes doesn't in assigning/defining the pendulum part to know that it rotates. I searched for issues similar to this on the forum, but didn't find any. WEC-Sim API for pto.loc also did not provide much information. Does this location have to be located on the part you want moving? For example the pendulum is at (after the COG definition moved it) [1 0 1.2], and I define the pto.loc [1 0 1.2] which resulted in no movement. I then tried a variety of locations (a BIG variety!), all with no result. I have gotten it to move in the past before though, under different conditions (e.g. when my previous _.stl_ files did not upload 'correctly' in WEC-Sim)

3) When it was working, the pendulum rotated around the extrusion in yaw, but it always strayed far away from it instead of what you would expect. FYI, I changed the layout a bit, originally I thought maybe it was because the extrusion and the hull were one part so maybe the pendulum and PTO constraint was referenced to the diameter of the hull, but after making a new non-hydro part (the extrusion), the pendulum still does not rotate tightly around it. 

Thoughts? The only thing I've got is trial and error so far,
Thanks,
- chris

t-osu, 2020-11-17:
I have fixed my issues, although I would still appreciate an explanation as to why :]
I set my pto.loc = [0 0.1 1.2] which results in the pendulum rotating tightly around the extrusion. I am wondering why setting the y coordinate worked, but if x, y left zero, or if only x and z have a value, it doesn't. Would this have to do from changing its the orientation?

closed 2020-11-17 by t-osu