WEC-Sim/WEC-Sim #458: Running wecSim in a parfor loop

eliesh17, 2020-11-12:
Hi, I am trying to run wecSim in a parfor loop but keep getting transparency errors (which is expected considering the number of variables) 
I was wondering if there is a workaround for this. 

Thank you.

yuyihsiang, 2020-11-18:
Please switch to the Dev branch and check the wecSimPCT.m file. We added a wecSimFcn.m file to call wecSim from a function to get around this issue.  If you are planning a series WEC-Sim run using MATALB parallel computing toolbox, you can use the run wecSimPCT.m instead of wecSimMCR.m, see the user manual. 
[image]

eliesh17, 2020-11-24:
> 
> 
> Please switch to the Dev branch and check the wecSimPCT.m file. We added a wecSimFcn.m file to call wecSim from a function to get around this issue. If you are planning a series WEC-Sim run using MATALB parallel computing toolbox, you can use the run wecSimPCT.m instead of wecSimMCR.m, see the dev branch user manual (bottom left corner).
> 
> [image]

Thank you for your reply!!

I'm gutting the following error when I run wecSimPCT: 

Error using wecSimPCT (line 100)
The source code (X:\...\wecSimPCT.m) for the parfor-loop
that is trying to execute on the worker could not be found.

Caused by:
    Undefined function 'hydroData' for input arguments of type 'double'.
    Error using remoteParallelFunction (line 84)
    Worker unable to find file.
        Undefined function 'hydroData' for input arguments of type 'double'.

If I set simu.reloadH5Data=1, then I get the following error: 

parfor imcr=1:length(mcr.cases(:,1))

Caused by:
    Error using wecSim (line 257)
    Variant control 'sv_b1_hydroBody' of variant block 'C6M/BA' must be a condition expression or name of a
    Simulink.Variant object in the global scope workspace of the model. Ensure that the condition does not
    match the model name.
    Error using wecSim (line 257)
    Variant control 'sv_b1_nonHydroBody' of variant block 'C6M/BA' must be a condition expression or name of
    a Simulink.Variant object in the global scope workspace of the model. Ensure that the condition does not
    match the model name.

yuyihsiang, 2020-11-25:
We have tested using the RM3_MCROPT3 in the WEC-SIm Applications repository. If you have same issue when running the RM3_MCROPT3 using wecSImPCT.m, you may want to check two things

1. Which MATLAB version r u using?

2. Can you confirm your parallel setting?
[image]
[image]

Note: simu.reloadH5Data=1 should not matter. The *.h5 hydrodynamic data is reload, regardless the setting for simu.reloadH5Data in the WEC-Sim input file.

yuyihsiang, 2020-12-09:
Did it resolve your issue?

closed 2020-12-16 by kmruehl