WEC-Sim/WEC-Sim #459: Setting up ParaView NL Hydro Visualization

Aquaharmonics, 2020-11-16:
Hello,

We have run the Paraview NL hydro tutorial and visualization works in Paraview for us via the OSWEC_Nonlinear_viz.pvsm file when loading the state and after running the complete tutorial.

However, we are not sure how to get everything working when starting with the OSWEC.pvd, or any other .pvd from the WEC-Sim output. 

We run the WEC-Sim script, which works fine, and then we run the pressureGlyphs script, which creates glyph items in the tree, glyph 1-glyph 5 like the image below for our own device output:

[image]

in the  OSWEC_Nonlinear_viz.pvsm, the output looks like below, with the 5 glyphs and associated calculators:

[image]

however, when we try to run the calculator for the different glyphs we get the following errors, both with our own device and the OSWEC.pvd whose files worked for running the OSWEC_Nonlinear_viz.pvsm:

[image]

It seems we are misunderstanding the steps on how exactly to generate the glyph data with the calculator function in Para view. We couldn't find anything in the repo this, do you have any suggestions? Did we miss where this process is described?

Thanks,

Alex

kmruehl, 2020-11-19:
@Aquaharmonics I'll look into this. Just to clarify, you're havingthis issue when using the 
https://github.com/WEC-Sim/WEC-Sim_Applications/tree/master/Paraview_Visualization/OSWEC_NonLinear_Viz case, and also when you run your own case, correct?

Aquaharmonics, 2020-11-19:
Hi @kmruehl ,

The 

https://github.com/WEC-Sim/WEC-Sim_Applications/tree/master/Paraview_Visualization/OSWEC_NonLinear_Viz  

case works fine when we load the .OSWEC_Nonlinear_viz.pvsm state first and then point it to the directory where the output files from WEC-Sim are.

However, if we just open the OSWEC.pvd file in Paraview and then run the WEC-Sim script and pressureGlyphs script, we get the same errors if we try it on our own output files.

We are not very versed in ParaView, and it seems that  is where the issue lies. 

One thing we did try that seemed to be a short term solution, maybe not an elegant one was this:

1)  load the .OSWEC_Nonlinear_viz.pvsm state and point it to the directory where the OSWEC paraview files are:

[image]

2) open our own .pvd output file, which then shows up in the bottom of the tree in the pipeline browser. Notice also the buoy shows up in the bottom render window in red:

[image]

3) run the WEC-Sim and pressureGlyph scripts:

[image]

4) Delete all the "body 2" items from the tree:
[image]

5) Select body_1:flap in tree, right click and use the "change input" option. In the window, select our .pvd as the new input (we are a single body in this simulation) and hit ok.

[image]

6) Note how the flap is now replaced with our device body and its hydrostatic normal vectors per below:

[image]

7) carefully replacing all the various elements in the pipeline browser using the "change input option" we finally arrive at what you see below. It needs some help, but looks like overall it functions and runs without error. Again, the issue seems to be our lack of experience with paraview, but the data structures, etc from WEC-Sim look to be correct, and the pressure glyphs show up for all of the glyph calculations 1-5. 

[image]

Also just wanted to confirm the units for pressure are Pascal? Is this absolute scale? In addition, do you have any advice on how to output the pressure data generated to a .csv in paraview using the export inspector? 

[image]

Thanks again for the help.

Best,
Alex

Aquaharmonics, 2020-11-19:
Also, if it helps we can upload/send files. Basically, just trying to get our .pvd files to load with the same render and section views like the OSWEC tutorial.

Cheers,
Alex

kmruehl, 2020-12-11:
@Aquaharmonics I just ran the [OSWEC_NonLinear_Viz ](https://github.com/WEC-Sim/WEC-Sim_Applications/tree/master/Paraview_Visualization/OSWEC_NonLinear_Viz) case, and I have the exact same issue. When I load the `OSWEC.pvd` file and run `pressureGlpyhs` I get a series of warnings and a bunch of glyphs in the pipeline but the presssure glyphcs do not render properly. But the visualizations loads properly fine when I apply the `OSWEC_Nonlinear_viz.pvsm` state file. In short, I do not think this is a user error, I think this is actually a version compatibility issue/bug. What version of Paraview are you using? I have Paraview 5.9.0/ 

I'm going to look into a few things, to try to debug this case:
- [ ] run a previous release of Paraview and see if I get the same error, to confirm this is a version comparability issue
- [ ] debug the `pressureGlyphs.py` macro.

kmruehl, 2020-12-11:
@Aquaharmonics I was able to confirm that this is a version comparability issue. I just downloaded Paraview 5.4.1, and was able to load the `OSWEC.pvd` file, run the `WEC-Sim` macro, and then successfully run the `pressureGlpyhs` macro on body_1. For now, I would recommend using the workaround you've found by loading a previously saved state, or downloading a previous version of Paraview (version 5.4.1 worked for me). 

[image]

Aquaharmonics, 2020-12-11:
Hi @kmruehl ,

Thanks for looking into this! 

I currently have Paraview version 5.8. I will download 5.4.1. and report back with results. I have made a little more progress with version 5.8 with the workaround, and will compare the results with what I find from 5.4.1. 

Cheers,
Alex

closed 2021-07-02 by kmruehl

kmruehl, 2021-07-10:
@Aquaharmonics this issue should be fully resolved on https://github.com/WEC-Sim/WEC-Sim, and the Applications are being updated via https://github.com/WEC-Sim/WEC-Sim_Applications/pull/8.

Aquaharmonics, 2021-07-14:
> Closed #459 <https://github.com/WEC-Sim/WEC-Sim/issues/459> via #595

Hi @KelleyRuehl,

Thanks for following up on this, we will give Paraview version 5.4.1 a try!

Best,
Alex

commit 2a63daa7 (2021-07-02): Merge pull request #595 from cmichelenstrofer-forks/master

Fixes #459. Update ParaView macros for Paraview 5.9
commit 7c19f183 (2021-07-02): Fixes #459. Update ParaView macros to work with latest version of ParaView (5.9.1).