WEC-Sim/WEC-Sim #460: Error in reading AQWA generated h5 file

abhisheks865, 2020-11-17:
Hi there,
When I am running the wecSim code after creating h5 file from AQWA, I am getting below error. 

[image]

I have checked my H5 file and it is of 8 MB so i think it has all the data required to proceed in wecSim. Can someone help me to resolve this error?

sugarl14, 2020-11-18:
can you send a screenshot of the following things:

1) BEMIO script in matlab
2) first few lines of both .LIS and .AH1 files
3) output options in AQWA (Analysis Settings > Output File Options)

abhisheks865, 2020-11-18:
@sugarl14 Thanks for responding to my query. I think I troubleshot this problem. wecSim command was showing an error because AQWA generated h5 file had only one body. I set up only one body while setting up my model in AQWA. I grouped floating and submerged bodies into two different bodies in AQWA and generated the h5 file again. That's how I was able to read h5 file in wecSim.

closed 2020-11-18 by abhisheks865