WEC-Sim/WEC-Sim #461: Plotting Power Output for Linear PTO

sugarl14, 2020-11-18:
Hi,

I'm modeling an OWSC using a linear pto (not ptoSim) and am currently trying to determine which combination if pto.c and pto.k values results in the greatest power output. 

I'm trying to plot the power output in the userDefinedFunctions.m code, but I think I'm missing something, conceptually. I've tried reading other posts about this issue, but I think I'm just missing something that is keeping me from understanding. So here are a few of my questions:

1) Is `output.ptos.powerInternalMechanics(:,5)` the power output of the OWSC that I am looking for? If yes, are the units in Watts or kW?  If not...

2) how would I calculate the OWSC power output? since Power = Force * Velocity, would it just be: 
`power = output.ptos.forceActuation(:,5)   *    output.bodies(1).velocity(:,5)` ? If yes, I'm assuming these units are Watts.

3) Please correct me if I am wrong, but I believe `output.ptos.forceActuation(:,5)` is the torque (in Newton*meters) that the linear PTO experiences, correct? If yes, whenever I run wecSim and then navigate to ` output.ptos.forceActuation`, it gives me a matrix of zeros.  I tried debugging this by opening the `Revolute Joint: Pitch PCC` mask, expanding `Actuation`, and changing Torque to be Automatically Computed. However, wecSim gave me this error:

[image]

From this error, how would I "increase the number of joint primitives with motion from inputs" or "reduce the number of joint primitives with automatically computed force or torque?" 

Here is a snippet of the Joint Mask:

[image]

Thank you for the help!!

dforbush2, 2020-11-18:
I am a little confused here as to how your PTO is setup. It sounds as if you are using pto.c and pto.k to act as your PTO. If this is the case, the forces/powers that result will be in the appropriate InternalMechanics fields. The forceActuation field you mention would be expected to be zero since, if I understand your model, there is not additional actuation force applied externally.  Specifically, I am guessing you're using this block:

[image]

and not the other ones. The forceActuation field logs the force applied via the additional inputs of the right two blocks (that I have assumed you are not using). But I agree this is confusing looking at the block diagram. We are in the process of improving our documentation to clarify the parameter --> output structure relationship. 

With respect to the units of power, check out the annotation within ../PTOs/Rotational PTO/InternalMechanics block. Power should be in W, and force should be in N (in translation) or N-m (in rotation), but do pay attention to the sign convention!

I am not sure about the error you are seeing when you interact with the block mask, but this should be a totally solvable problem without manipulating these fields.

sugarl14, 2020-11-19:
Yes sir, I am using pto.c and pto.k as my PTO. Thank you for the explanation. That's where my mind was heading, but your explanation definitely cleared things up for me. 

There's one more thing that's confusing me a little. Here is what's happening when I run the OSWEC tutorial:

When I plot `output.ptos.powerInternalMechanics(:,5)`, here is what I get:

[image]

But, whenever I solve for Power using the equation P = F .* v 
where F = `output.ptos.forceInternalMechanics(:,5)`  and  v = output.bodies(1).velocity(:,5) 
I get this:

[image]

So for the OSWEC tutorial, it's the same magnitude, but different signs. 

When I try that same plotting process for my OWSC geometry, I get the following results. Same magnitude _and_ sign convention.

[image]

Any ideas why this is?

dforbush2, 2020-11-19:
I'm reasonably certain that this is the same issue as #247 . It was a legacy sign convention that was not super well documented. @jleonqu is in process of reviewing/revising this for consistency.

sugarl14, 2020-11-19:
Ahhh gotcha. Thank you!

closed 2020-11-19 by sugarl14