WEC-Sim/WEC-Sim #462: Quick Question about wave force

sugarl14, 2020-11-18:
Hi,

Does WEC-Sim calculate the force of the wave that acts on the device being studied? I'm looking to model my OWSC in SolidWorks FEA and I'd like to know how much force to apply. I wasn't sure if this info was included in WEC-Sim's calculations. 

thanks!!

dforbush2, 2020-11-18:
Hi @sugarl14 

Assuming you are running with non-linear hydrodynamics turned off (simu.nlHydro = 0), the output.bodies.forceExcitation field will include the wave excitation force on the body. However, if you are aiming at an FEA analysis, it might make more sense to consider the total force on the body (ouput.bodies.forceTotal), since this will include the sum of all forces on the body. Of particular concern would be any actuation force applied to the body, since this may add with an excitation.

Note that this force is an aggregate over the whole body, based on your BEM results, but for a device like the OSWEC in a realistic wave field, these forces will not be evenly distributed over the body surface, which may be an additional consideration for an FEA model.

sugarl14, 2020-11-19:
Thank you very much!

closed 2020-11-19 by sugarl14