WEC-Sim/WEC-Sim #463: Multiple Condition Runs Tutorial

eirinikatsido, 2020-11-19:
Hello, 

I have noticed that the file wecSimMCR.m is missing from here https://github.com/WEC-Sim/WEC-Sim_Applications/tree/master/Multiple_Condition_Runs. 

I would appreciate if you could provide it!

Best Regards, 
Eirini

kmruehl, 2020-11-19:
@eirinikatsido Thanks for your post. We do not include `wecSimMCR.m` with the [MCR applications cases](https://github.com/WEC-Sim/WEC-Sim_Applications/tree/master/Multiple_Condition_Runs) because it is included in the WEC-Sim source here: https://github.com/WEC-Sim/WEC-Sim/blob/master/source/functions/wecSimMCR.m. Once you've added the WEC-Sim source directory to your MATLAB path, you should be able to call `wecSimMCR` from any case directory, including the MCR application cases.

eirinikatsido, 2020-11-19:
@kmruehl Thank you for your reply! Now it makes sense..

kmruehl, 2020-11-20:
No problem. I'm going to close this issue.

closed 2020-11-20 by kmruehl