WEC-Sim/WEC-Sim #465: Degenerate mass distribution error

abhisheks865, 2020-11-19:
Hello all,
I am a beginner at WEC-SIM and trying to run the simulation for two-body buoy very similar to RM3 mode. For that, I have taken RM3 example base file and replaced data with my files as follow:
1. Changed rm3.h5 file with my h5 file which I have created using AQWA.
2. Changed geometry file of float and spar.
3. changed the following file name in wecSiminputfile.m and tried to run the model.
However, my model is showing degenerate mass distribution error. Also model is showing upside down in Mechanics Explorer (Attached below). I did not change any constrained, so I was not expecting 

[image]

Can anyone guide me how can I fix this error and will be able to generate the power output from this geometry. 
Also, can anyone please guide me how we decide the value of pto(1).k &  pto(1).c. 
I am attaching the H5 file which I have generated from AQWA, Float and spar stl file, wecsiminputfile.m for your readdy reference. 

[zip.zip](https://github.com/WEC-Sim/WEC-Sim/files/5569693/zip.zip)

dforbush2, 2020-11-20:
1). Please see issue #454 for specific suggestions to resolve this error. Without seeing your simulink model *.slx, it is difficult to offer more particular advice. Please double check that your units are as-intended in your mesh file used for your AQWA run, since sometimes incorrect units can cause this error due to the unintentionally disparate magnitudes of force and inertia quantities even when the model is constructed correctly. 
2). This is a complicated question .Please see #422, where it is handled in detail.
3). The visualization uses your *.stl files. (These do not affect the calculation unless simu.nlHydro is not equal to 0). It appears that the geometry was modeled about a non-zero origin, and likely was modeled in mm. By default, many *.stl interpreters outside of solidworks presume the units are m. The simMechanics visualization is thus likely 1000x as large as intended and displayed at its non-zero origin. Recreating the STL file at origin 0,0,0 and in units of m should resolve this issue.

abhisheks865, 2020-11-23:
1. Thanks @dforbush2 for responding. I am attaching my .slx model for your reference so that you can have a look on it. I have tried remodeling my whole design in meters to avoid any problem related to size but getting the same results.
2. I have also corrected the orientation and modelled the design in positive z direction (attached) but same error. Is there something am I missing ?

[zip.zip](https://github.com/WEC-Sim/WEC-Sim/files/5584596/zip.zip)

dforbush2, 2020-11-25:
Hi @abhisheks865 

I have plotted your *stl files in meshmagick and have noted that they are modeled with the xy plane likely not coincident with the body.cg, and their z-axis is not coincident with the origin. As mentioned, I would recommend meshmagick  (https://lheea.github.io/meshmagick/command_line/cli_usage.html#translations) to directly manipulate and visualize your *.stl files. But perhaps that zip file did not contain the most recent stl files?

I am suspicious of your *h5 and input files: with the specified mass and displaced volumes, the combined (2-body) device does not float. It is typical for the heave plate/spar to sink, but the float must have adequate additional flotation to support the combined body. Additionally, the cg of both bodies is above the waterline, while that cb is below the waterline. This is certainly atypical, and may not be what was intended. This perhaps could prevent the model from simulating, since the initial condition is thereby unstable, so I do recommend double-checking these conditions, but I am still working finding the definitive issue causing your error.

abhisheks865, 2020-11-25:
Hello @dforbush2,
I will have a look on stl files again. But, I tried to export them with new coordinate system on CG. Not sure why it is not aligning with z-axis as I have created it on orgin. But, will have a look on it and also on Meshmagic. For your reference, I am attaching updated stl file. 

[newstl.zip](https://github.com/WEC-Sim/WEC-Sim/files/5598285/newstl.zip)

kmruehl, 2020-12-24:
@abhisheks865 can you save your SImulink model to a previous version of MATLAB? I am unable to open your `onedrum.slx` file because I'm running 2020a. Thank you!

``` matlab
Warning: Model 'onedrum' was created with a newer version (R2020b) of Simulink
To create a model that is compatible with this version of Simulink, load the model in Simulink
R2020b and select Save > Export Model to > Previous Version. 
```

kmruehl, 2021-01-13:
@abhisheks865 following up with my previous post. First, I'm going to have to apologize for previous response. While there are still some unresolved issues with the STL files, they will not affect your WEC-Sim solution. The STL files are only used by WEC-Sim for visualization (unless you're running [non-linear hydro](http://wec-sim.github.io/WEC-Sim/dev/man/advanced_features.html#nonlinear-buoyancy-and-froude-krylov-excitation)). Please disregard my prior response. 

After looking into this issue a big more I have to agree with @dforbush2's conclusion that the core issue here is the BEM data from AQWA. While I am able to reproduce the error using the AQWA data you provided, if I simply swap out that hydroData from the [rm3.h5 hydroData file](https://github.com/WEC-Sim/WEC-Sim/blob/master/examples/RM3/hydroData/rm3.h5), the case runs without issue. I've zipped up this case using the rm3 hydroData here: [OneDrum.zip](https://github.com/WEC-Sim/WEC-Sim/files/5810519/OneDrum.zip)

closed 2021-01-13 by kmruehl