WEC-Sim/WEC-Sim #468: Postprocessing Paraview Error

Rodimus88, 2020-11-21:
I got the following error when I try to create paraview file for wecsim example RM3.

Post-processing and saving...   
    ...writing ParaView files...   
Error using fprintf
Invalid file identifier. Use fopen to generate a valid file identifier.

Error in paraViewVisualization (line 55)
            fprintf(fid,[bodyname '\n']);

Error in wecSim (line 308)
paraViewVisualization

[image]

akeeste, 2020-12-02:
Hi @Rodimus88 

Thank you for bringing this bug to our attention. In a previous commit, the way that hspressure, wpressurenl, and wpressurel are passed internally in the post-processing functions was changed. These three references were not fully updated when creating Paraview output. 

I have created a pull request (#472) to fix these references to the cell pressures. It will soon be reviewed by another member of the WEC-Sim team and merged with master to fix this issue. Please let me know if the same problems occurs after this update.

Adam

Rodimus88, 2020-12-02:
Hi @akeeste 

thank you for the attention given, So i just need to update my wec-sim directory to avoid this error. I will inform if any error occurs afterwards.

closed 2020-12-02 by Rodimus88