WEC-Sim/WEC-Sim #469: a crush in run in matlab

mozhganaghanezhad94, 2020-11-24:
I am using NEMOH for modelling a awavestar WEC. but right after discretisation it crushes with the error below.
: [ mass, inertia, KH, XB, YB, ZB]=aximesh (r, z, length (r))
and can not go to upper command anymore.
how can i fix it?

nathanmtom, 2020-11-25:
@mozhganaghanezhad94 Thank you for your question, unfortunately this issues page is related to WEC-Sim issues and NEMOH is a separate code that we do not support. I am sure you have already reviewed the information on the NEMOH [website](https://lheea.ec-nantes.fr/research-impact/software-and-patents/nemoh-presentation), but suggest revisiting and see if any support can be received from that community. 

We have a few researchers with experience using NEMOH, but cannot guarantee they can provide additional comments. 

Cheers, 
Nathan 
National Renewable Energy Laboratory

dav-og, 2020-12-02:
Hi mozhganaghanezhad94,
Please can you share your Nemoh project files? (including files generated by Nemoh)
Cheers,
Dave

nathanmtom, 2020-12-09:
@mozhganaghanezhad94 Our team wanted to check-in on your issue? If possible could you let us know if you were able to solve your issue or upload your files for our team members to review? This will help us determine if we can close this issue.

Cheers, 
Nathan 
National Renewable Energy Laboratory

closed 2020-12-16 by nathanmtom