WEC-Sim/WEC-Sim #470: Question about force plotting

Rodimus88, 2020-11-25:
I was try to simulate a small circular bouye of 50cm diameter to see how  much forces acting upon the objet. i just use the regular wave condition of 2.5m wave height and 8s wave period.  when i got the heave force plot of the bouye, i compare it with the exapmle RM3's plot. i was suprised to see the plots were almost similar. just only few were beign different. can that be posssible, shouldent the smaller object expereince less forces than the one in the examples. if i have done any erroe in my hydrodata or any other error. it would be grateful if any one can educate me about this.

the two figues in the below show the results plots. left one is the RM3 and right one is my created object.
[image]

dforbush2, 2020-11-25:
Hi @Rodimus88 

Physically, you are correct. All else being equal, a buoy like the one you are describing should generally experience smaller excitation, radiation, and hydrostatic forces than the RM3 model, so I am also surprised by these results.

Just to rule out the easy stuff:
1). Are you running 'wecSim' from the intended directory, and with a properly configured path pointing to the correct *.h5 fiel? I've definitely made this mistake and run a different case than intended. A way to check is, from the directory you want to run 
a) `edit wecsim`
b) place a debug point on line beginning sim(simu.simMechanicsFile...
c) run wecsim, and then check out your workspace variables, particularly the 'body' variable, to make sure they are pointing at the *.h5 file you intend. Checking body.dispVol is a good idea.

2) If the above doesn't work, then your *h5 file is suspect. How were your BEM results for this geometry produced? How was the mesh generated? It is a good idea to double-check the units in the mesh file (i.e., to make sure your 50 cm isn't modeled as 5 m) 

Please recall that if simu.nlHydro = 0 (as it does by default) then the *stl file is only used for visualization, and the *h5 file alone determines the hydrodynamic coefficients involved with producing the above plots. 

Attaching a *zip of your run directory will be helpful to follow up, if these suggestions have not solved your problem.

Rodimus88, 2020-11-26:
Hi @dforbush2 

I check for the points you have mentioned in the comments, path dirrectory is to the .h5 was set up correctly. the wecsim file also was corretly calling the simulink file and .stl files. i dont know how to palce a debug point on the line as you mentioned, so i skiped it. when the simulation ran the workspace variables are showing the data of .h5 given but i dont see the diplaced volume in it and i dont know how to use the body.dispVol fucntion. 

I created the .h5 by write_hdf5 in WEC-Sim application master, i use soliworks to create the stl file for the geometry. I dont use any meshing for this and i dont know about a such procedure.i just created the .h5 file and .stl files and insert them

 and do necessery changes in the wecsiminputfile.m .i think the error may be due to that. i will attach a zip file of the directory so that you can clearly identify any problem if i have missed. 
[Sphere1 - Copy.zip](https://github.com/WEC-Sim/WEC-Sim/files/5600785/Sphere1.-.Copy.zip)

dforbush2, 2020-11-30:
Hi @Rodimus88 

This was helpful. I was able to take a look at your *h5 file directly and I notice that it is very similar to the *h5 file developed for the RM3 by the write_hdf5 RM3 applications case. As mentioned above, if simu.nlHydro =0, then WEC-Sim determines hydrodynamic coefficients from the *h5 file. Because your h5 file looks like the RM3, your results will as well in this case, regardless of the geometry in your *stl file.  This explains why your WEC-Sim results look as they do, but is not likely what was intended!

Looking at the write_hdf5 applications case, you will need to come up with a hydroData.mat file like the one used in this example _but for your spherical geometry_. It looks to me instead like the RM3 buoy hydroData.mat file was resampled at different frequencies and for a single body and single wave direction. You have correctly modified the inertia, in your wecSimInputFile, but for this single degree of freedom simulation this will have little to no effect. Typically, the information needed in hydroData.mat will come from a boundary element method code like NEMOH, WAMIT, or AQWA (as discussed in our documentation https://wec-sim.github.io/WEC-Sim/man/overview.html). In particular, NEMOH is free and easily used for axi-symmetric meshes. There is even an included example for a spherical geometry you could work from.  **This is the approach I would recommend.** 

 From there ./source/functions/BEMIO/Read_NEMOH.m can be used to create your *h5 file.

Rodimus88, 2020-12-02:
Hi @dforbush2 

it was a well detailed explanation. i will tryout the NEMOH to create the hydrodata. thank you for the help given.

closed 2020-12-02 by Rodimus88