WEC-Sim/WEC-Sim #471: ERROR WHILE TRYING TO START WECSIM

SofiasKonstantinos, 2020-11-25:
I'm new in matlab and WecSim , so i'm trying to go with the spectrum import waveclass . I have this error message 

**Undefined operator '>=' for input arguments of type 'struct'.

Error in waveClass/waveSetup (line 239)
                            freq_loc = freq_data>=min(obj.bemFreq)/2/pi &
                            freq_data<=max(obj.bemFreq)/2/pi;

Error in wecSim (line 143)
waves.waveSetup(body(1).hydroData.simulation_parameters.w,
body(1).hydroData.simulation_parameters.water_depth, simu.rampTime, simu.dt,
simu.maxIt, simu.g, simu.rho,  simu.endTime);**

I am using the "write_hdf5.file'' to provide the right h5 file  , but i dont get it . Fo r instance i have the water_depth variable inside the h5 hydro file but i still get this error . 
I also created a spectrumData file by using the command : 
**if true
  Input=xlsread('MATLAB.xlsx');
  save spectrumData1.mat
end**
Your advice would be very helpful !!!! 
Thanks in advance 
[SofiasKonstantinos wecsim.zip](https://github.com/WEC-Sim/WEC-Sim/files/5598641/SofiasKonstantinos.wecsim.zip)

nathanmtom, 2020-11-25:
@SofiasKonstantinos Thank you for describing your issue. I have downloaded your .zip file and read your text, but will need to ask a few questions to better understand what you are trying to achieve. 

1.  Are you having issues with the spectrum import option within the waveClass or generating a separate .h5 file?
2.  If I've read your text accurately, then the MATLAB.xlsx file contains the spectrum data you want imported to WEC-Sim? If so, can you please state what the units are of the 1st and 2nd columns? Note the first column needs to be the wave frequency in Hz and should be in ascending order, currently it looks like it is random.
3. The write_hdf5.file you have provided does not seem to have changed from the WEC-Sim applications case. Therefore, what other hydrodynamics are you trying to add to the rm3.h5 file?
4. The files you have provided unfortunately do not include the other files required to run WEC-Sim so it is difficult to understand if you have potentially used incorrect inputs or missing information. If possible, please include the wecSimInputFile.m and `.slx` file and other files required to execute your example. 

Hopefully, answering my questions above and more information on your model files will help us get to the cause of your issue. 

Cheers, 
Nathan 
National Renewable Energy Laboratory

SofiasKonstantinos, 2020-11-27:
@nathanmtom Thank you very much for the immediate response and sorry for aswering today. 
1. I have issues with the spectrum import . I think . 
2. I will attach the xlsx file again with the right order , and the first column beeing the frequency in Hz and the second one beeing the spectrum density according to the PM spectrum method . 
3.I have data from the Poseidon System of the Hellenic Centre for Marine Research , so i just need to work with this wave data , i have Hs , Hmax , Wave_dir, T(s), Tp , and depth . These measurement are for every 3h . To be completely honest i dont know which hydrodynamics i need to change , i thought i was going to import my wave characteristics and use the hydro coefficients provided by the examples . 
4. I am going to attach all the file i use in the path to make wecsim run . 
Thank you very much 
Konstantinos 
University of west Attica 
[SofiasKonstantinos wecsim vol.2.zip](https://github.com/WEC-Sim/WEC-Sim/files/5608858/SofiasKonstantinos.wecsim.vol.2.zip)

nathanmtom, 2020-12-01:
@SofiasKonstantinos Thank you for providing updated information on your model. After reviewing the files I have the following comments: 

1. To clarify, you want to use the provided WEC-Sim files for the RM3 with a hydraulic PTO, but only import your own custom wave spectrum? Please check that your Excel file as it provides !div/0! indexes and the frequency spacing is non-uniform and there are certain frequencies that repeat with different spectral values. Please double check the spectrum is in the form that you desire as the spectrum doesn't appear to follow the traditional PM definition. 
2. Did you start with the files provided in the WEC-Sim_Applications-?PTO-Sim->RM3->RM3_HydraulicPTO?
3. Your `spectrumData1.mat` file include several other variables that are not needed. The waves.spectrumDataFile needs to only include one variable with a single two column vector with the first column the frequency in Hz and the second column in m^{2}/Hz. I've attached a .zip file that includes your original .mat file and a version I created that would be accepted into WEC-Sim.
4. I am still not sure why you generated a new .h5 file with the create_h5File.m? If you are not trying to change the hydrodynamics than I would use the rm3.h5 file already provided in the example case. 
5. In your wecSimInputFile.m you have included the `simu.nlHydro` option; however, it is not clear if you have sufficient resolution with the provided .stl to run this option as no .stl files were included in your .zip file. Therefore, I turned this option off during troubleshooting. 
6. In your wecSimInputFile.m, the mass of the float and the plate were empty, not defined, which caused problems with the simulation running. I'd suggest using `body(1).mass = `equilibirum';` and the same for body(2). 

After making the changes above, I was able to run WEC-Sim with the custom spectrum; however, I cannot check for the accuracy of the device response given the unusual spectrum input. 

Hopefully, following my comments above will help you troubleshoot your problem or provide additional guidance that allows me to better understand your simulation desires and how to address continued issues. 

Cheers, 
Nathan 
National Renewable Energy Laboratory

[spectrumData1.zip](https://github.com/WEC-Sim/WEC-Sim/files/5624613/spectrumData1.zip)

SofiasKonstantinos, 2020-12-05:
@nathanmtom Thank you once again for your help , so i need some further guidance while i am working on this project 
1. I want to provide my simulation with wave data that i have for 1 year every hour . So it is innevitable that they are random and the frequency spacing is non-uniform . I need to make the simulation as if it where in a random sea state place. So i cant understand maybe if i'm using the wrong command with "spectrumimport" . Maybe i need to use etaImport but it needs surface elevation and only having significant wave height i dont know how to calculate it and from which benchmark. 
2.Yes i started with those exact files 
3.I understood this step and your file was very helpful 
4.I want to change certain hydrodynamics , such as wave depth but i cant because it says it is read only . I downloaded HD5 reader but i can only read them in a right order from there 
5.Im no sure yet if i change the body element , but if i do , is scalling them enough to just make the WEC smaller without havong to provide new hydrodynamics ? 
6.I also used body(1).mass = equilibirum'; and it worked just fine . 

I dont want to be burdensome with asking so many questions but the project is really important to me and i am really interested in working on this program after my graduation . 
Kind regards , 
Konstantinos Sofias 
University of west Attica

nathanmtom, 2020-12-05:
@SofiasKonstantinos Thanks for the additional answers to my question, let me try to provide some guidance to help troubleshoot your issue. 

1. Please clarify what you mean by wave data for every hour for one year? Does this mean every hour you have a significant wave height and a peak period, as these are normally used to define a sea state based on an assumed spectrum? Or for each hour you have a spectral form you want to model (individual vectors of first column [frequency, Hz] and second column [specturm, m^{2}/Hz]). If you want to simulate several 1 hour sea states, with changing parameters then we'd suggest using the multiple condition option, to input the sea state characteristics you have on record. 
2. Understood, thank you.
3. Great, glad they were of use.
4. Understood, in theory changing the water depth can influence the hydrodynamics if the WEC geometric properties do not scale as well. What procedure, or rather where in the code are you trying to define the water depth? If you are trying to define this within the `wecSimInputFile.m` then you are correct that this is a read-only property which is pulled from the .h5 file. The water depth would need to be updated in the .h5 file you are using. 
5. If you plan on scaling the size of the WEC, then you would also need to scale the .stl file if you want to run `simu.nlhydro`; however the .stl provided in this example is likely too coarse to run this option. To your question about scaling, you should scale both the hydrodynamics and the mass properties following Froude Scaling Laws to be the most accurate. 
6. Understood. 

Happy to work with you to try and troubleshoot these issues. So please let me know if you need further clarification or have follow on responses to help me troubleshoot your issues or understand your modeling desire. 

Cheers, 
Nathan

nathanmtom, 2020-12-16:
@SofiasKonstantinos I wanted to check in with you to see if you were able to resolve your issues running WEC-Sim? If you are still having trouble feel free to answer my last series of questions. If you have resolved your issue please let us know so we can close this issue. 

Cheers, 
Nathan
National Renewable Energy Laboratory

SofiasKonstantinos, 2021-01-13:
Dear @nathanmtom , 
I'm sorry it took me so much time to respond but i had some trouble with the pandemic . As i continue to work on my project i need futher guidance . So , i have measurements for every hour for one year , for Significant height , Peak Period , Frequency and Wave direction in deg . I need to describe the sea state for these conditions . Also i need to change the wave depth for sure , and maybe some more hydrodinamics. How can i do that providing my simulation with an error free h.5 file . I have downloaded the HDFview program but cant seem to figure it out . Again sorry if i caused trouble with closing this issue on time , it was due to medical reasons

nathanmtom, 2021-01-13:
@SofiasKonstantinos No problem and glad to hear that you are recovering from your medical conditions. 

1. Yes, if you have measurements for every hour for one year then you can run a multiple conditions run (MCR) that will iterate across the significant wave height, peak period, and wave direction; however, you will need to select what spectral form the wave spectrum should have. Currently in WEC-Sim we have a pierson moskowitz and jonswap spectral forms available and hopefully they fit the spectrum shape used to calculate your wave statistics. 
2. Changing water depth will be more difficult as currently this is pulled from the .h5 file which obtains this information from the hydrodynamic input files from WAMIT or NEMOH. If possible, can you provide more information on what type of water depth changes you want and why they are important?
3. It is still unclear to me why you are having difficulty with the .h5 file as the wave inputs are not defined by this file. The .h5 file contains the hydrodynamic coefficients in the frequency and time domain. Therefore, unless you are using custom imported hydrodynamic coefficients you shouldn't need to touch this file. I believe you have mentioned in the past scaling the hydrodynamic values if your WEC model is smaller than RM3, which will require additional work to convert into a new .h5 file. 

I hope the above answers has helped clarify options within WEC-Sim, but please let me know if still have questions. 

Nathan 
National Renewable Energy Laboratory

SofiasKonstantinos, 2021-01-14:
Dear @nathanmtom ,
Right now i need to comprehend the theory behind the wave spectrum , to be more specific of what i need as an input. Can you suggest any docymentation so i can get a clear image of this ?

nathanmtom, 2021-01-14:
@SofiasKonstantinos Sure, I'll point you to the documentation that should be most relevant to your question. 

1. The theory behind the wave spectrum formulation within WEC-Sim can be found [here](http://wec-sim.github.io/WEC-Sim/man/theory.html#irregular-waves).
2. The inputs needed in the wecSimInputFile.m is described [here](http://wec-sim.github.io/WEC-Sim/man/code_structure.html#irregular).
3. We have a online training course on the Wave Implementation, [here](http://wec-sim.github.io/WEC-Sim/man/webinars.html#wave-implementation), that should be able to walk you through how this object is used to define the wave climate. 
4. Users and utilize our multiple condition runs (MCR) option to simulate a large number of sea states one after another, more information found [here](http://wec-sim.github.io/WEC-Sim/man/webinars.html#wave-implementation).

Hopefully, these sources provided you background knowledge and help you better understand how wave spectrums are used within WEC-Sim. 

Cheers, 
Nathan
National Renewable Energy Laboratory

nathanmtom, 2021-01-29:
@SofiasKonstantinos I wanted to check with you to see if the additional information provided has helped resolve your issues? If not, please let me know what is still confusing you and we can hopefully work together for a resolution. 

Cheers, 
Nathan

SofiasKonstantinos, 2021-02-01:
Dear @nathanmtom  
Apparently i cant seem to comprehend the theory of the PM spectrum and generally i need some more info about my study. All i need to do is to model a WEC in a specific sea state , and a specific depth , and see what results i get in terms of power . Can you recommend me a way of finding these factors ? Im new to matlab so i found it a bit difficult , but besides that , a find it very interesting also and i like WEC-sim as a programm. Hopefully i finish my study with success and i also need you to know that your help is very important and i aprecciate it a lot .

nathanmtom, 2021-02-01:
@SofiasKonstantinos Thank you for your response and happy to provide you some guidance that may help you better understand the theory in WEC-Sim. 

To your first question, you most likely can rely on the default calculation of the PM spectrum provided within WEC-Sim as we follow the IEC TC114 standards. Using the PM option in irregular waves you will be able to set the significant wave height and the peak period which should be provided with any sea state data you have been provided. This lecture course material **[here](https://ocw.mit.edu/courses/mechanical-engineering/2-22-design-principles-for-ocean-vehicles-13-42-spring-2005/readings/lec6_wavespectra.pdf)** may provide additional background information on the PM spectrum.

To you question about specific depth, this gets more complicated because the energy or power contained within a specific sea state is dependent on the wave height, wave frequency, and the water depth. Currently, within WEC-Sim we pull the water depth from the .h5 file which is the output from BEMIO. BEMIO takes the hydrodynamic data output from a BEM solver such as WAMIT, NEMOH, or AQWA, and outputs it in the form required for WEC-Sim. Therefore, technically to be theoretically correct you would need to complete a new BEM run for each new water depth. Now, some approximations can be made if the water depth doesn't change a lot (50 m to 45 m), but if you are talking about say 50 m to 10 m the hydrodynamics have probably changed enough to affect the dynamics. Therefore, I'll need to know more information to better understand parameters you might also be changing to see if this would be significantly altered when changing the water depth. 

Appreciate you comment about WEC-Sim as we hope to provide information and tools to make it easier for new users to pick up quickly. 

Please let me know if there is additional information we can provide to help with you study. 

Cheers, 
Nathan 
National Renewable Energy Laboratory

nathanmtom, 2021-02-10:
@SofiasKonstantinos I hope you are doing well. Please let me know if you have any additional question or need clarification on my previous comment. 

Cheers, 
Nathan
National Renewable Energy Laboratory

nathanmtom, 2021-02-18:
@SofiasKonstantinos I'm going to go and close this issue, but reopen if or when you would like to revisit this topic of conversation.

closed 2021-02-18 by nathanmtom