WEC-Sim/WEC-Sim #473: Generating AQWA file for hydrodynamic data

Rodimus88, 2020-12-02:
Hi,

I was interested about creating hydrodynamic data using AQWA BEM, as this name AQWA is used in simulation software Ansys, does it capable of producing hydrodynamic data. I have referred to the overview of the wec-sim documentation. in there it is mentioned "More details on AQWA setup are given in the AQWA Reference Manual." but not link is provided to the referencing. so if anyone can give me good insight about the AQWA, I would be much grateful.

Thanks.

dforbush2, 2020-12-02:
From google: https://cyberships.files.wordpress.com/2014/01/wb_aqwa.pdf
Though this does not appear to be the most recent version. If you have the software, I am guessing this manual can be found included in an included documentation directory.

Rodimus88, 2020-12-02:
Hi @dforbush2 
thanks you to sharing this information. I had gone thorough this earlier when i installed Ansys on my pc. but it is not very clear how to get the output files required by Bemio to create the.h5 file. If anyone was able to do this would have point out essential things so the work would be much easier for my self :D. but thanks @dforbush2 for your concern over the matter. If I understood the scenario of creating aqwa file properly, I will post the steps in this forum so another one will be able to learn something from it.

dforbush2, 2020-12-02:
Thank you! I will close this issue for now, but feel free to re-open if you uncover a good procedure.

closed 2020-12-02 by dforbush2

Rodimus88, 2020-12-07:
Hi @dforbush2 

I have gone through the aqwa reference manual and figure out how to create the .AH1 and LIS. all is depend on the output setting of the analysis of aqwa software. need to turn on few of the parameter like Output ASCII Hydrodynamic Database to get the .AH1 file. enabling Output Element Properties, Output Source Strengths and Output Potentials allows more information to be added in to .LIS file.

but there was a problem in generating .h5 file using the bemio function. another one have already posted a similar problem in the issues(#476 ), i think it is also related to exceeding of array bounds of the function. I really does not know what the reason for that issue. is it depend on the other settings like wave directions, frequencies or any other parameter. also the required output parameters that I have mentioned above over defining the .LIS and .AH1 files so the bemio,m cannot execute it. 

A good explanation to this would be helpful to any one who is trying to create hydrodynamic data using aqwa. 

thank you.
[image]