WEC-Sim/WEC-Sim #474: Purpose of function "AdjustMassMatrix"?

t-osu, 2020-12-03:
Evening all,

I have in SolidWorks a model of a physical hull I will be working with. It has been successfully tested for a previous students PhD in a wave basin, so it should be stable (or stable-ish). The weight of the entire hull is 24kg, and the draft is about 0.1m, my point being, to check with Archimede's principle, I checked the weight of the displaced fluid in SolidWorks and AQWA, they both give out 24kg, which is fantastic.

I'm having some issues in WEC-Sim though.
When I have a no wave condition, and I set _body(1).mass = 24_ and _body(1).momOfInertia = SolidWorks output_, the properties change. The mass is now about 95 kg, and the MOI is different as well. The simulation fails after 0.15 seconds.
When I set _body(1).mass = 'equilibrium'_, the variable body.mass is now about 24kg (this is what that value is supposed to do anyway, according to WEC-Sim API) and the MOI is what I entered. The simulation runs successfully.

1) What I'm wondering is, if _body(1).mass_ works when set to '_equilibrium'_ (which outputs a value of 24kg), why does it not work when I set it equal to 24kg directly? 
In my efforts, I traced it down to the function **adjustMassMatrix** in the bodyClass.m script. I could assume by its name what it is accomplishing, but I am curious why it is changing the mass if it is supposedly a floating body according to some buoyancy principles. Obviously prototypes aren't perfect, but I'm hoping not to change the design or mass too much so I can use the prototype hull for future research.

2) I also noticed that when using _body(1).mass = 'equilibrium'_ and trying with a regular wave condition, (very small 0.01m Hs and within period of AQWA frequencies) the mass is over 100kg and the simulation fails. I'm unsure of how to go about designing for a more stable system if that is the issue, given that it should be buoyant, the frequency output is not perfect, but not terrible, and it has been successfully tested in a basin before.

Thanks,

- Chris

akeeste, 2020-12-03:
Hi @t-osu, thanks for your question.

WEC-Sim does change the body.mass and body.momOfInertia in the adjustMassMatrix() function. This function combines the mass or moment of inertia with the corresponding added mass before simulating the device. The calculation is reversed in restoreMassMatrix() so that the output is correct. The addedMass set-up is described most thoroughly in issues #161, #308. Essentially, WEC-Sim must handle the added mass in a special way because the combination of the m+AM creates an array of mass values [1:6]. Simscape does not allow a body to have multiple masses in for each degree of freedom, so the special treatment by WEC-Sim is required.

In regards to your specific questions:
1. The added mass treatment should be the same regardless if the 'equilibrium' option is used. Where are you printing the mass value?
2. All of the added mass treatment described above should not cause the simulation to fail, so something else may be causing an issue. Can you give the error message you are receiving and also upload a zip folder of your working directory? With this I can dive into the issue and provide more assistance.

Thanks
Adam

t-osu, 2020-12-04:
Thanks @akeeste for the quick reply.
Ah, yeah, I figured as much just by its name.
1) I go to the workspace where the bodyClass variable name _body_ is, double click, and locate its mass property. Or equivalently in the command line type body(1).mass. 
2) The error message is `Derivative of state 'Vincent.Floating_3DOF.CONSTRAINT_Planar_Joint.Px.v' in block 'Vincent/Floating (3DOF)/CONSTRAINT (Planar
Joint)' at time 0.155 is not finite. The simulation will be stopped. There may be a singularity in the solution.  If not, try
reducing the step size (either by reducing the fixed step size or by tightening the error tolerances)` (you know, the usual error, haha).
Attached below is the .zip folder.
[Vincent.zip](https://github.com/WEC-Sim/WEC-Sim/files/5645423/Vincent.zip)

Thanks again,

- Chris

t-osu, 2020-12-16:
Afternoon @akeeste 
I've tried changing the design of my system which has helped it not fail and reading up on the posts you mentioned above to better understand the function. However I am still getting odd results in my body mass output that are different when manually entering the value vs 'equilibrium' which you stated in should be the same.
I'm still unsure how to go about making sense of this difference in outputs. Have you had any updates on your end?
Thanks

akeeste, 2020-12-16:
Hi @t-osu 

Thanks for following up and sorry for the delay. I have not yet identified anything in your model that would cause the crashing. Although when running the Simulink file step by step it is clear in the SimMechanics Explorer that the body's motion is very unphysical. I tried adding damping and stiffness to the constraint and body but this did not improve the issue. The 'analysis.h5' file looks fine to me aside from a few irregular frequency effects. Though I caution you if trying to use the 'vincent.h5' file in any other cases, this BEM data is largely negative with large irregular frequency effects.

What changes did you make to the system that enable it to run without failing?

As far as the mass--I see the same mass values for both mass=24 and mass='equilibrium' (23.6675kg when rho=1000kg/m^3). Both cases result in mass=95kg on wecSim.m line 320 right before the simulink file is run. Depending where you have paused the simulation to check, maybe the special case of mass='equilibrium' has not been adjusted yet.

Adam

t-osu, 2020-12-28:
I just made it bigger, honestly.
OK, I had this design taken from a student's previous work so I thought since it was tested successfully, it should run in a simulation fine. It was moored in 3 places though, so maybe that was contributing to its success.

I think this is where I am getting confused, maybe I didn't explain it right so I'm adding some attachments.
_eq.png_ is where I set `body(1).mass = 'equilibrium` and _manual.png_ is where I set `body(1).mass = 24`.
I get the same results as you said above in the middle red box, equilibrium yields 23.6675 and manual set to 24 yields 24. After running the simulation for each case (as far as it will go is a tenth of a second, but the Simulink file is still run), the third red box is not anywhere near 24. Although I have set `body(1).mass = 24` before running the simulation, entering `body(1).mass` in the command prompt gives ~172.
Is this due to the AdjustMassMatrix?
[image]
[image]

akeeste, 2021-01-04:
Hi @t-osu 

Yes you are correct! This change comes from the AdjustMassMatrix function, called in wecSim.m right before the Simulink simulation begins. Here the added mass and body mass are lumped together as:
body mass = body mass + added mass 

This is a bit simplified, but is essentially what you are seeing. If Simulink runs successfully to completion, then the original mass and inertia are restored from body(i).hydroForce.storage. postProcess.m calls the function restoreMassMatrix() to correct these values before post-processing and saving output data.

Normally, this all happens behind the scenes and a user has no indication the mass is changed. However, when Simulink crashes postProcess.m is not run, so the change can be seen.

Let me know if you have more questions on this.
Adam

t-osu, 2021-01-04:
Thank you so much @akeeste! I know that we went through a lot to get to a one line answer, but I really appreciate helping me understand what was really going on in the background.

closed 2021-01-04 by t-osu