WEC-Sim/WEC-Sim #475: MoorDyn continues to crash Matlab

t-osu, 2020-12-03:
Hi all,

Recently I have been having an issue using MoorDyn. 
Every time I try to run a WEC-Sim simulation with it, the Moor-Dyn initiatied console window opens and either 1) runs, showing that the fairleads converge or 2) opens but doesn't run. After either scenario, a window opens in the Mathworks Crash Reporter stating Matlab has encountered an internal problem and needs to close. 

I can run the RM3 mooring example file fine, and have copied their text file over to my application, deleted and adjusted some values to make a single line mooring system. I also adjusted the wecSimInput.m mooring class lines to be 1. 
I tried changing some items in the simulation class to match those of the RM3 example, but no luck. I also tried a remote desktop that has more processing capability than my laptop, but again, no luck.

Anyone have ideas on what could be going on? I could not find related results in other related issues posted.

Thanks,
- Chris

yuyihsiang, 2020-12-09:
Can you upload your files for us to check?

Thanks,

--Yi-Hsiang

t-osu, 2020-12-11:
Yes, attached is a zip folder.
It should be noted that this model doesn't work, there were other issues with it, noted in my previous post #474 that Adam is still looking at, I assume, however I don't think that would be directly related to why MoorDyn is crashing.
Thanks!

[Vincent_mooring.zip](https://github.com/WEC-Sim/WEC-Sim/files/5681577/Vincent_mooring.zip)

t-osu, 2020-12-28:
@yuyihsiang just an update, I tried it on my desktop that I got setup in case it was an issue with my laptop or some remote connection issue,, but the problem still occurs. I looked at the report generated, but it doesn't make any sense (at least to me!) so I made sure to send it to Matlab. 

After trying on my desktop with the RM3 MoorDyn example, it worked as it did with my laptop. The only difference I can think of is my `lines.txt`, although I was getting errors with it before in the MoorDyn prompt window (node index number errors) which I fixed so I wonder what else would be the issue.
Matlab also responded saying the error was due to an unknown issue, so that might be a dead end on their side.

dforbush2, 2021-01-29:
Hi @t-osu 

This also crashes my matlab trying to run it... I'm not really sure why this is the case. The fairlead tensions converge (or appear to do so) in the moorDyn generated window: then things sit for about 30 s before crashing my Matlab.

I am not sure exactly the source of this error, but I would recommend that you review your lines.txt file as compared to the hydrodynamic properties of your body. For instance, your MoorDyn lines file has 50 m or chain that is 50 kg / m in air. This chain is stretched, initially, 49 m (based on your vessel and fixed location), against a stiffness on the order of 2E7 N. Checking out your hydrostatic stiffness data, your restoring coefficients in heave are on the order of 300. This implies a massive downwards acceleration at the start of your simulation: I suspect this is causing issue with the simscape model determining its initial conditions. I'm not sure why this crashes matlab...but I would reconsider some of your input parameters.

dforbush2, 2021-02-10:
Hi @t-osu 

Any updates here? Hopefully revising your Lines.txt file to align better with your device properties have solved your issue.

t-osu, 2021-02-10:
@dforbush2 Thanks for taking a look!
I actually switched over to the mooring matrix, so I haven't had the chance to look at MoorDyn again yet, however what you said would seem to make sense. I overlooked those conditions, but since then have a better understanding on stiffness values so I will try that soon.
Just wanted to update.

closed 2021-02-10 by dforbush2