WEC-Sim/WEC-Sim #476: ERROR in reading Read_AQWA file (line194)

akashch293, 2020-12-07:
Hello,
I'm using Aqwa version 2020 R2 and Matlab 2018a
I got the following error when executed the BEMIO code

Index exceeds array Bounds

Error in Read_AQWA (line 194) 
               ind = tmpl(1:3) ; tmpl(1:3)=[ ];

I tried to figure out the problem and also compared my .LIS and .AH1 file with other correct files like aqwa_example_data.The compared files look similar but still Error is produced for my file.
I'm attaching my .AH1 and .LIS file below :
[ANALYSIS.zip](https://github.com/WEC-Sim/WEC-Sim/files/5650860/ANALYSIS.zip)

akeeste, 2020-12-07:
Hello @akashch293 

Thanks for posting your issue and uploading your AQWA files. 

We have had versioning issues with AQWA input to BEMIO in the past. It seems ANSYS makes slight changes in the .AH1 and .LIS files in each new version of AQWA and it can sometimes cause issues with reading the text files correctly with BEMIO. This is not necessarily the problem here, but just something I wanted to point out. We are working to resolve this issue. 

Regardless, I will take a look at your input and see if I can find the issue in the Read_AQWA() function.

Adam

akeeste, 2020-12-08:
Hi @akashch293 

I have found the bug with the read_AQWA script and submitted a fix (PR #478). The script tried to read a line of non-numeric characters as numbers, and caused `tmp1` to be empty and non-indexable in line 194. The fix corrects the lines of read_AQWA that are read for the Froude-Krylov force.

Once this PR is reviewed and merged, the issue will be solved. Please let me know if this does not resolve your issue.

Adam

sugarl14, 2020-12-10:
Hi @akashch293,

In the meantime (assuming you're working under a university or industry license), you could maybe reach out to your technical help desk and ask them if they have access to Ansys Workbench 18.1. GUI isn't too different between versions :)

Good luck!

akeeste, 2020-12-10:
@sugarl14 Thanks for the suggestion! and @akashch293 I hope the issue is resolved with the PR.

We have had a lot of problems of AQWA versioning in the past, so having different output versions is very helpful. I currently have output from Ansys 2015, 17.0, 19.0, 20.0. If either of you do use Ansys 18.1, I would appreciate any sample .AH1 or .LIS files you can provide!

sugarl14, 2020-12-10:
Hi @akeeste,
 I do use Ansys 18.1!

Is there a specific model you'd like me to run in AQWA for you? Specific mesh size? Specific maximum wave frequency? I know there's a lot of output options in AQWA that may affect the .AH1 and .LIS files. Let me know what you need and I'd be happy to provide it!

akeeste, 2020-12-10:
@sugarl14 I opened a new ongoing issue to discuss read_aqwa, let's continue our discussion there.

@akashch293 I will close this issue as PR 478 has been merged. Please respond if you have further problems with the same issue.

closed 2020-12-10 by akeeste