WEC-Sim/WEC-Sim #477: Issue in Using meshConvert

strvalik, 2020-12-08:
Hi all, 

I am trying to convert an STL file to a mesh readable by NEMOH. I want to use meshConvert to convert from .stl to .dat. However, I get errors in matlab when trying to do this:

 meshConvert('myfile.stl','stl','dat','convertedfile',2,1)

and I get the following error:

Array indices must be positive integers or logical values.

Error in import_stl_fast (line 81)
if isnan(C{4}(end))

Error in meshConvert (line 44)
        [V,normals]=import_stl_fast(meshfile,2,stleol);                     % outputs vertices and normals of ASCII stl mesh, triangular panels

Can anyone help me with this? I have been stuck for hours. If there was some other way that I could have an stl file read by NEMOH that would be terrific!

dforbush2, 2020-12-09:
I appreciate your efforts trying to debug this on your own!

It looks like that error is initially with import_stl_fast.m. Are you sure you have saved your STL file in the ASCII format, rather than binary? I don't think that import_stl_fast will work with a binary STL format.

I would caution you that this workflow is not recommended for the generation of a NEMOH mesh. NEMOH uses a quadrangular mesh, so each panel uses vertices ABCD. An STL file is a triangular mesh, so each panel has vertices ABC. Converting an STL to a NEMOH mesh simply duplicates a vertex to make one that appears quadrangular in file syntax: e.g., ABC --> ABCA. This works only in the lowest functional sense and may not give you the most reliable NEMOH results. Additionally, STL files typically do not adequately panel planar areas for BEM methods (see Refining an STL, https://wec-sim.github.io/WEC-Sim/man/advanced_features.html#bemio).

If you have a simple geometry, especially an axisymmetric geometry, NEMOH has included mesh generation tools. Additionally, if you can build a quadrangular mesh directly, like a GDF file from Rhino, and then convert to the NEMOH mesh format, this is more reliable, since its just reformatting the file, not actually changing any geometry.

You may also explore meshMagick (https://lheea.github.io/meshmagick/command_line/cli_usage.html#translations) for mesh conversion and manipulation: it is much more capable than meshConvert.m

strvalik, 2020-12-09:
Very helpful! Thank you. I have generated the .dat mesh file. Could you help me with the steps after this?

I now have only the .dat file and the matlab routines. When I use axiMesh, it creates many other files (such as Nemoh.cal and hydrostatics) which I assume are necessary.

dforbush2, 2020-12-09:
Once you have your *.dat file, you will want to follow NEMOH instructions to create the output files WEC-Sim needs. https://lheea.ec-nantes.fr/valorisation/logiciels-et-brevets/nemoh-examples. I recommend using one of their example runs as a template. NEMOH has utilities for generating Hydrostatics files, as does meshmagick (link above). 

We are not developers of the NEMOH code, and can offer limited support beyond the above.

closed 2020-12-16 by dforbush2