WEC-Sim/WEC-Sim #479: Variant Subystem Rendering

kmruehl, 2020-12-09:
There's an issue with the rendering of variant subsystem blocks. For example, the Rigid Body block is labeled as a Drag Body when pulled from the WEC-Sim Library (even though it hasn't been initialized yet). This has been temporarily resolved on the `master` branch by "refreshing" the variant subsystem in the library, and retaining the variant subsystem variables in the MATLAB Workspace. This is not an ideal solution, and we need to come up with a more robust solution on the `dev` branch.

t-osu, 2020-12-14:
Hi @kmruehl 
I'm glad I happened upon this post because I was trying to work out all weekend if I had done something wrong!
Would you mind explaining more what you meant by resolving through the `master` branch and "refreshing" the variant subsystem in the WEC-Sim Library? The only option I see to refresh the Library is in the main library, I don't have an option if I select a specific subsystem, if this is what you meant. If it was, refreshing the Library for me still results in the `rigid body` being labeled as drag `body`.
Thanks!

akeeste, 2020-12-15:
Hi @t-osu,

What @kmruehl meant is that she put a quick fix into the master branch, but not the dev branch where we are looking for a more robust solution. The update in master leaves all of the subvariant variables uncleared after simulation which causes the correct labels to appear when opening the Simulink file. Alternatively, after selecting a block with variant subsystems, a new menu tile appears. Under this 'Variant Subsystems' menu, the 'Refresh Blocks' option can be used to remove the incorrect 'Drag body' name. Although I would like to highlight that this label is only incorrect is appearance, the correct functionality is there.

[image]

I have looked into this issue more. It seems that it only occurs when a block has both a mask, and variant subsystems. Other blocks with variant subsystems but no mask (e.g. excitation force) do not seem to have this naming problem. Also, the problem only occurs when the correct Variant variables (sv_b1_dragbody, sv_b1_hydroBody, sv_b1_nonHydroBody) are not yet defined. If these variables are set before opening the library or a model, the correct subsystem name is chosen.

When the Variant variables are not set and without refreshing, Simulink defaults to using the first subsystem name alphabetically. The actual default subsystem is irrelevant to what name appears. This is what is seen when opening the library or a model without any workspace variables. I have found two workarounds, @kmruehl let me know which of these you prefer and I will push the change:
1. Rename the 'Drag Body' block to 'Viscous Drag Body' so that 'Hydrodynamic Body' is first alphabetically and is shown.
2. Use the 'Allow zero active variant controls' option so that the Simulink does not pick any block without the Variant variables defined. This option shows no name unless a subsystem is correctly activated by a Variant variable.

t-osu, 2020-12-16:
I actually found the solution to my problem to be different. After following these instructions, I still was getting the error. I dove into the `variant manager` in Simulink and where they were getting defined in `wecSim.m`. 
It turns out the issue was that I was using an older (not by that much, maybe a month?) version of `wecSim.m` that I had edited and it was unable to correctly identify the blocks into hydro, nonhydro, or drag. Easy fix though, just copy and pasted my old code into the newer `wecSim.m`.

akeeste, 2020-12-16:
Hi @t-osu 

I am glad you were able to resolve your issue! Can you describe what you did to fix this issue more? With your change, if you open up the library or a WEC-Sim simulink model with a clean workspace does the 'drag' label still show up or is it fixed?

Thanks,
Adam

kmruehl, 2020-12-17:
Resolved with #486

commit 820448c8 (2020-12-17): Variant subsystem label fix resolves #479 (#486)

* versioning fix. Label fix only works w/ R2017b+

* miscellaneous cleanup
closed 2020-12-17 by kmruehl