WEC-Sim/WEC-Sim #480: Singularity Issue

sugarl14, 2020-12-10:
Hi there!

I'm a little confused about something and would love some help! To give a little context as to what I'm doing, I'm trying to create a power matrix for an OWSC using the wecSimMCR function. MATLAB seems to be yelling at me when I use a wave height of 2.5m and a wave period of 4 seconds using an irregular PM wave spectrum and the `waves.phaseSeed=  1`. 

I've ran into this issue before, but I can't seem to debug this one. The error is:

Error using wecSim (line 291)
Derivative of state 'wecsim.Rotational_PTO.Pitch_PCC_Revolute_Joint.Rz.w' in block 'wecsim/Rotational
PTO/Pitch PCC  (Revolute Joint)' at time 76.774999999999991 is not finite. The simulation will be stopped.
There may be a singularity in the solution.  If not, try reducing the step size (either by reducing the fixed
step size or by tightening the error tolerances)

Error in wecSimMCR (line 98)
    wecSim; - Show complete stack trace

When I ran into this issue in the past, all I needed to do was change `simu.dt` from 0.1 to 0.01 or 0.001. I've tried changing the `simu.solver` to ode4 and ode45 but still nothing. I've tried changing `simu.rampTime` and `simu.endTime` to 40 & 10, 400 & 100, and 4000 & 1000. But none of these fix the error :(

I think this error has something to do with the wave type (irregular vs regular), because when I run the simulation for regular waves with the same wave height and period, I don't get any errors, and the case runs just fine.

Any idea on why wec-sim isn't liking irregular waves right now for MCR?

Thank you!

dforbush2, 2020-12-11:
This is a weird one if your model works for regular waves at this frequency, and works for other sea states. Some questions:
1) Do you have any damping on your PTO? 
2) Are you using the *h5 file from the example? 
3) Does the model behavior up to that point look reasonable? (If the simulation errors out, you won't have your output variable created, but you can check out the columns of pto1_out that should remain in the workspace. Check the block in the library model for column order, remembering that all 6 DoFs are logged, so the the first 6 columns are position, then, in groups of 6, they are: velocity, acceleration, forceTotal, forceActuation... etc)

If all of the above look good , then what's happening is somehow for that wave, phase realization, and time step, some force component is coming out Inf or NaN.  I would try changing up your phase seed, maybe waves.phaseSeed = 128. I know its not a super elegant solution but that may get you moving in the short term.

sugarl14, 2020-12-13:
1. Yes, I do have defined damping (pto.c) and stiffness (pto.k) in the wecSimInputFile.m code
2. I am using the .h5 file from my own geometry, not the example OSWEC.
3. The model behavior does look reasonable and consistent for that model with almost all other cases. A couple days ago, I was having issues with random sea states like H=3 and T=3, H=2 and T=2, and some others. Sometimes they work fine and sometimes I need to make simu.dt = 0.001 for them to run.  `pto1_out` looks good-- there are 48 columns (so 8 groups) reported.

I tried changing `waves.phaseSeed = 128` but I still got the same error message.

dforbush2, 2020-12-16:
Hmmm. Can you please attach a *zip with your model files? I will take a look.

sugarl14, 2021-01-03:
Hey @dforbush2,

Sorry for the late response. I am waiting for approval from my research advisor to send the zip file. Since my work is university affiliated, I have to have his approval before sending any files. Also, I would assume he wouldn't want me uploading any of my work to a public space like this forum, so would it be OK if I sent it to your email? If so, what would be a good email to reach you at?

Thanks again for all of your help!!

Kindly,
Landon

sugarl14, 2021-01-05:
Just an update: my advisor gave me permission to share my files with you. Also, I've tried running the simulation with regular waves, but at a certain combination of waves.H and waves.T, the simulation fails again with the singularity issue. So I'm thinking the issue may be with my damping and stiffness values? 

If it's ok with you, let me know an email I can zip my files to! Thanks!

dforbush2, 2021-01-06:
Hi @sugarl14 
You can email me at dforbus@sandia.gov

sugarl14, 2021-01-07:
I changed my wave class per your suggestions to waves= 'regularCIC' and it solved my singularity issue for a few cases but not all of them. Whenever the wave height exceeds the wave period, I get the singularity issue again.

dforbush2, 2021-01-07:
I took a look at your *h5 file and there are a couple of oddities. All of your added mass terms are on the order of 10^-6.... which is a surprisingly low for the scale of your device. Same goes for the radiation damping coefficients, which are ~10^-7. I would recommend double-checking all of the units used in your initial model. I would expect something a minimum of 10^3 larger. This is likely the root of your issue. The OSWEC contained in the example files is a good deal larger than your device, but of similar geometry, so can provide a good reference for the expected order of magnitudes and trends in frequency expected for this kind of devices. For this combination of added mass and radiation damping, your system is driven almost entirely by your hydrostatics, which are about the right order of magnitude. 

Your STL files look good, but please recall that these do NOT dictate your hydrodynamics, so it is worth checking that the mesh used in your BEM simulation is in appropriate units, too. You'll probably want to reconsider your PTO damping and stiffness values after revising the BEM data. 

Your impulse response functions actually do look reasonable, but maybe a bit over-truncated given the very slow decay rate.

sugarl14, 2021-01-07:
That is very strange... maybe there is something going on then during the calculation of my .h5 file? Im using AQWA and this is what my raw .LIS file is showing me:

[image]

[image]

It seems like the added mass for RY is on the magnitude of 10^3 and the radiation damping (also for RY) range from 10^-2 to 10^3 (the one in the picture is at 10^3, but there were some that were 10^-2 at the 0.1 rad/s frequency range)

dforbush2, 2021-01-07:
Can you please check your AH1 file? read_AQWA pulls these coefficients from the *.AH1 file.

sugarl14, 2021-01-07:
Sure thing!
Here's a snippet of added mass:

[image]

and here's some of radiation damping:

[image]
[image]

and then further down radiation damping...

[image]

dforbush2, 2021-01-07:
Hmm odd. Can you share these files, your version of Read_AQWA, Write_H5, and other functions in your BEMIO work flow? I'm wondering if somehow the normalization (https://wec-sim.github.io/WEC-Sim/man/theory.html) is somehow being applied twice. You may try re-pulling the WEC-Sim repo and regenerating the *h5 file...there have been some revisions to these files in the past few months.

sugarl14, 2021-01-07:
Just sent it! Do you recommend re-pulling everything? Or just the bemio files?

dforbush2, 2021-01-07:
Ok so on second inspection, your *h5 file and your hydroforce fields of your body look fine. Sorry for the false alarm and any confusion I created. Not sure why I was seeing the strange values. I would recommend continuing to run 'regularCIC'. 

My recommendation should be simpler now: re-run your AQWA run for your as-is geometry out to a higher frequency band. Your impulse response functions do not decay fast enough, and the issue is that your radiation coefficients are not near-enough zero at your highest frequency. If you run the OSWEC example, and plot 

'plot(squeeze(body(1,1).hydroData.hydro_coeffs.radiation_damping.all(5,5,:)))'

You'll see that at the high frequency, radiation damping coefficients are ~1, compared to a maximum of ~1000, about 3 orders of magnitude. Compare this to your radiation damping coefficients, which are separated by a single order of magnitude. 

Same goes for your excitation coefficients and added mass. 

This should generate an impulse response function that decays more rapidly (provided you calculate over a long enough time interval) and should prevent these singularity errors. 

Please let me know if that does not fix the issue: it should.

sugarl14, 2021-01-08:
Right on. I'll try it out!

So just to be clear: 
1) I don't have to re-pull any of the wec-sim code from the repo
2) re-run aqwa with a higher number of wave frequencies and then check to be sure the IRF reaches around 10^3, and then decays rapidly to ~1. 

Does this sound right?

dforbush2, 2021-01-08:
Almost. The point of running out to higher frequencies is to get your radiation damping coefficient to be very nearly zero at the highest evaluated frequencies, and the added mass to approach a constant asymptote. As it stands, neither of these things are quite true enough that you are getting a wholly stable impulse response function, and so for some combinations of waves, your results are going off to infinity. I just cited the OSWEC examples to demonstrate what "very nearly" zero looks like for a case we know to be well-behaved and stable: the magnitude and decay rate of your IRF will be different, but it in theory the IRF should go to very nearly zero over the calculated interval. Expanding the evaluated frequency range in your BEM code will help this to be the case.

sugarl14, 2021-01-08:
Ahhh gotcha, that makes sense. I'm re-running AQWA now and should be done soon, so I'll keep you posted on the results. Thank you!

sugarl14, 2021-01-08:
Hmm... here are the raw results from AQWA. Radiation damping is still a bit off from 0, and the added mass doesn't seem to want to converge to a single number. Maybe refining the mesh would help?

[image]

[image]

sugarl14, 2021-01-08:
and here's what they look like in wec-sim with the code you provided above `plot(squeeze(body(1,1).hydroData.hydro_coeffs.radiation_damping.all(5,5,:)))`

[image]

and added mass (not looking so hot)

[image]

dforbush2, 2021-01-08:
The decay rate I was referring to is that of the IRF, so it is a time-domain decay. With respect to these results, I'm not an AQWA pro, but mesh refinement is a good place to start. That damping certainly does look improved towards the high frequencies, so we're on the right track.

sugarl14, 2021-01-08:
I ran one more case got the highest wave freq to be ~14.2 rad/s. And here were the results for radiation damping and added mass. Pretty similar to what I had on my last thread. I went ahead and used `waves= 'regularCIC'` but still ran into the same error at certain wave conditions. 

[image]

[image]

thanks for the help!!!

dforbush2, 2021-01-08:
@jleonqu do you have any tips for @sugarl14 with respect to getting convergent AQWA damping results? Thanks!

jleonqu, 2021-01-08:
Hello @sugarl14 and @dforbush2 

Refining the mesh is a good place to start. I would recommend to do a convergence study for the mesh, just to make sure that the mesh size does not interfere with the results. Also, I would take a look to the detailed results for all the elements of the mesh, maybe there are some weird results in some of the elements and this is why the BEM results are not what we are expecting.

sugarl14, 2021-01-08:
Hi @jleonqu and @dforbush2,

I conducted a mesh sensitivity analysis for radiation damping, added mass, and wave excitation a while back, and everything seemed to check out OK. Here's one for radiation damping with mesh size ranging from ~4000 diffracting elements to almost 12,000 diffracting elements (the maximum that the aqwa solver can handle).

[image]

Is there a way you prefer to analyze the results for all of the elements in the mesh?

jleonqu, 2021-01-08:
@sugarl14  I don't have any preference for the way to analyze the results for all the elements of the mesh, but I would suggest to take a look to some contours to see if there is any weird results. I am not sure if AQWA has this option, but in other Ansys simulations it is possible to see the results of any variable such as pressure, velocity, etc, for each element, so I would take a look to these results.

dforbush2, 2021-02-10:
Hi @sugarl14 

Any updates here? I will close this issue soon, but feel free to re-open if needed. Hope this is going well.

Best,
Dom

sugarl14, 2021-02-10:
Hey @dforbush2,

I've been a bit pre-occupied with some other work so I haven't really gotten a chance to really look more into the AQWA side of things, but if I run into more issues once I get back to it I'll let y'all know!

I really appreciate all of your help, Dom!

Kindly,
Landon

sugarl14, 2021-02-26:
aaaaaaand we're back! 

Hey again. I've messed around with everything a bit-- double checking my AQWA files, re-calculating my optimum damping values for PTO (I have `pto.k=0`), but I am running into the same problem in my original post.

To recap, that is: When I run a simulation in WEC-Sim in regular waves at `H=3.25` and `T= 8.5` (or similar wave conditions), everything works well. However, when I switch to irregular waves where `waves.spectrumType = 'PM';` and `waves.phaseSeed=1`, the simulation fails. I've played around with different values for `simu.dt`, `wave.phaseSeed`, and `simu.solver` but to no avail. 

Before posting this update, I also made sure to run through everything we talked about previously so I didn't miss something that we've already talked about. I wish I could use regular waves in my simulation, but my advisor would like me to model irregular waves since they're a bit more realistic (compared to regular, sinusoidal waves). 

Sometimes if I include a non-zero value for `pto.k`, the simulation will work with irregular waves. But I was hoping to keep `pto.k=0`. Do you have any other recommendations or suggestions for moving forward? If you need to take a look at anything let me know!

Thank you!

sugarl14, 2021-02-28:
So I think I figured out why the code was not running properly. A long time ago I was fiddling around with my simulink file and I put some Limits on the rotational joint of my OWSC with some pretty large, arbitrary damping and stiffness values. Well, I never changed them...  (see below)

[image]
[image]
 
 
 
When I removed the limits, I was able to run the code successfully without the singularity issue.

However, when I removed the limits and watched the SimMechanics video of my model, it just rotated around and around its hinge (past the 90 degree location, which obviously is unrealistic in the real world). I added limits of 30 degrees with the damping and stiffness values equal to `pto.c` and `pto.k`...

[image]
 
 
but now my model just kind of flops to the 30 deg position and bounces around. See here:
 
[image]

Do you know if this an issue regarding my PTO values or something else? I calculated the optimum damping value for this specific wave state with the equation from Issue #515, and set `pto.k=0`.

nathanmtom, 2021-03-01:
@sugarl14 Thanks for reopening this and providing additional information about your issue. I'm joining this conversation late, but please bear with me:

1.  What is the resonance period (or frequency) of your OWSC? Your error placed 3 days ago appears to be related to converting the Quaternion to Euler Angles. If you are hitting the resonance period you can end up with unrealistically high amplitudes of motion which can cause WEC-Sim to error out. 
2. Have you run `noWaveCIC` to ensure that if you place your WEC in calm water it remains still?
3. Have you tried running `regularCIC` with a very small wave amplitude (~0.01 m = waves.H)? If we are hitting resonance then the amplification in your response at full wave height may be creating issues. If we move to very small wave amplitudes then you may run successfully and we can pinpoint the issue to the wave height. 
4. In addition, you can try to implement a free decay test to verify the radiation hydrodynamics are sufficient to provide a clean response signal. This way we eliminate the influence of the incident waves on the problem to simplify what forces are active and being calculated. 

I feel that performing these checks will help provide additional information on the status of your model to help debug. 

Please let me know if you have any questions or need clarification. 

Cheers, 
Nathan
National Renewable Energy Laboratory

sugarl14, 2021-03-01:
Hi @nathanmtom,

1) The natural frequency of my OWSC is 4.05 rad/s, or 1.55 seconds

2) When I run `noWaveCIC` my OWSC remains completely still
[image]

3) I ran `regular` waves with `H= 0.01` and `T= 8.5`. The flap looked like it fell over and hit the end stops and didn't move after that. I re-ran the simulation again but with `T=1.55`, the natural period, but it did the same thing. 

4) If I'm not mistaken, doesn't BEMIO show the decay rate for the radiation damping term? If I'm not on the right track please let me know!
[image]

nathanmtom, 2021-03-01:
@sugarl14 Thank you for running these checks. 

1. Thanks for providing the resonance frequency. Can you clarify if the simulations that were crashing were occurring for sea states near the resonance frequency?
2. When you run the WEC-Sim model in regular waves with a small wave input it looks like your hydrostatic restoring force is negative meaning that when there is a slight displacement from vertical that the flap will fall over. What is the locations of the center of gravity and the center of buoyancy for your OWSC? 
2. BEMIO outputs the IRF which is a time series that needs to be convoluted with the time history of the WEC velocity to calculate the radiation forces. In a free decay test, we displace the WEC from its equilibrium condition and watch the decaying time history. The exact decay rate will depend on the lumped mass parameters for a general spring-mass-damper system. In a stable system, the WEC should return to its equilibrium condition after being displaced. This check helps to ensure the hydrostatics and radiation forces are being calculated appropriately. I'm concerned that if you're OSWEC falls over then the restoring coefficient is either not providing sufficient spring to drive your system back to vertical. It looks like most of the discussion in the past might be radiation forces, but we might need to verify the other forces into WEC-Sim and ensure they are not dominating your simulation. 

Let me know if you have other questions or concerns. 

Cheers, 
Nathan
National Renewable Energy Laboratory

sugarl14, 2021-03-02:
Hi @nathanmtom,

1) The simulations that were failing had sea states that were far away from the natural frequency of the OWSC.

2) COG = [0 0 -1.289] m
    COB = [0.000  -0.0038    -1.895] m 
[image]

 
 
Regarding the hydrostatic restoring force, is the picture below what you're speaking about? (I'm not sure if the hydrostatic restoring _coefficient_ and hydrostatic restoring _force_ are interchangeable terms.) If so, how would I change this value to be a positive restoring force? The only thing I can think of is lowering the density of the flap. I already have density at half the density of sea water (512.25 kg/m3)

[image]

3) Performing a decay test makes sense. I guess my question now would be where to start? Is there a way I can do `noWaveCIC` in WEC-Sim and then add an initial displacement of say 15 degrees, and then watch the OWSC response? I can then make a plot of the time it takes for the OWSC to stop oscillating vs. the motion amplitude. 

Also, thank you for all your help so far. It's greatly appreciated!!

Landon

sugarl14, 2021-03-02:
3) EDIT:

I used `noWaveCIC` and set `body(1, 1).initDisp.initAngularDispAngle = 15` to implement an initial displacement angle of 15 degrees... man oh man did my model do some funky stuff. It flipped upside down and is sort of cocked at an angle. Here's a couple screen shots:

[image]

[image]
 

It also looks like my fixed constraint and joint constraint (the rotational PTO) just kind of stopped working after I did this. I'll keep trying to debug until I hear back from you! Thanks again!

nathanmtom, 2021-03-02:
@sugarl14 Thanks for providing the information requested, I think we are starting to get to the bottom of your issue now. 

1. After this new information, I am leaning towards oscillating near resonance and the resulting large amplitudes of motion might not be the issue. Unless you are incorporating a `pto.k` to optimize power output which results in generating an artificial resonance at the wave frequency.
2. Yes, apologies for my loose terminology, but the information you have provided is what I am requesting. You have to be careful with these coefficients because you are calculating about the center of gravity rather than at the hinge. Since your center of buoyancy is below the center of gravity then the restoring coefficient when rotating about the center of gravity should be negative (I'll need to revisit my hydrostatics textbook). Since you have hinged to the seabed and the OWSC rotates about the hinge the center of gravity is actually moving in surge and heave so the total restoring moment about your hinge has more terms than just the (5,5) term you have listed above. You can first try to reduce the body(1).mass term towards 0, this is artificial but adding buoyancy by removing mass should help force the OSWEC back up after being displaced. I'd suggest this still as a check that can be completed without rerunning your BEM solver to update the coefficients you have listed. 
3. I'm still in favor of you completing a free decay test. In your first response above yes, displacing the OWSC by 15 degrees and seeing if it decays back to vertical would be a good check that the hydrostatics and the radiation coefficients are working as expected. Using the `initDisp` option is the correct way to move forward, but remember that WEC-Sim uses radians rather than degrees as input so WEC-Sim thought you wanted 15 rad. Second this `initDisp` rotates the device about the center of gravity, so you will need to also displace the center of gravity in surge and heave to represent the 15 deg rotation about the hinge on the sea floor. This should be relatively straight forward, but an additional calculation to ensure you are modeling the right situation. 

Hopefully this provides enough information to keep you moving through your troubleshooting. 

Cheers, 
Nathan

sugarl14, 2021-03-02:
Hi @nathanmtom 

1) when I first started these simulations months ago I was using a non-zero value for `pto.k`. But up until recently I have now set `pto.k= 0`. 

2) That sounds correct when you said there is a negative hydrostatic restoring coefficient when the COB is below the COG. I went back to AQWA and overrode my GMX and GMY values to be positive (instead of negative) just to see what would happen. And sure enough, they changed from negative to positive values. 
I changed the mass of my device from ~700 kg to 200kg and the SimMechanics video looked wonderful. No weird responses during regular or irregular waves. (picture below is of the OWSC response) 

[image]

Is there a more permanent fix than just reducing the mass in wec-sim? Would I just need to reduce the mass in AQWA and then recalculate?

2.5) Also, just wanted to confirm for you that my BEM results that I import to WEC-Sim are calculated at the COG.

3) That all makes total sense. Thank you. The only part I'm a little fuzzy on is "_...so you will need to also displace the center of gravity in surge and heave to represent the 15 deg rotation about the hinge on the sea floor._ "
How should I displace the COG to the hinge? I don't think I can change `body(1).cg` in the WEC-Sim input file since it gets over ridden after running Wec-sim. Would I need to import BEM files where hydrodynamics were calculated at the hinge > run in BEMIO > then specify `initDisp` ?

nathanmtom, 2021-03-02:
@sugarl14 Thanks for the quick follow up.

1. Based on your recent changes to the model to explore questions about hydrostatics then it appears that your current design has issues with hydrostatic stability. The sizing and mass properties of your current design appear to result in a negative restoring torque about the seafloor hinge. In my opinion you will need to redesign your system so that the buoyancy and mass properties provide a positive restoring coefficient about the hinge when displacing the OWSC from vertical. All of the other fixes you have suggested appear to be superficial and get WEC-Sim to run rather than model your system of interest. 
2. To clarify, if you rotate your OWSC about the center of gravity by 15 degrees the bottom of your OWSC will have moved along the x-axis and the y-axis away from the hinge. The initial displacement option does not account for the hinge attachment (multibody dynamics). Therefore, to model the system where the OWSC is rotating by 15 degrees about the hinge you will need to rotate the OWSC and then move the center of gravity by the x- and y- distances resulting from the 15 degrees rotation of the distance from the center of gravity to the bottom of the OWSC. As an example, if you look at the diagram below and assume point p is the hinge, then as the object rotates about p the center of gravity moves in the x- and y- plane. In WEC-Sim if we rotate about the center of gravity instead the point p on the rectangle will move in space and not stay in place where the hinge location would be. This correction is needed in WEC-Sim to model rotation about the hinge. 

[image] 

Therefore you do not need to displace the COG to the hinge and you do not need to import hydrodynamics calculated about the hinge.

I hope the above explanation helps orient yourself and how you can alter your model to get the results you desire.

Cheers, 
Nathan
National Renewable Energy Laboratory

sugarl14, 2021-03-02:
Hi @nathanmtom, thank you for the quick response.

1) The pictures I provided previously were BEM results calculated about the COG. But I do have BEM results that I calculated about the hinge, and the restoring moments are positive:
[image]

If I'm not mistaken, I believe that answers your concern? Just to avoid confusion here's the response of my results:
- when I calculate the hydrodynamic coefficients about the COG, hydrostatic restoring coefficients are negative.
- when I calculate the hydrodynamic coefficients about the hinge, hydrostatic restoring coefficients are positive.

2) Thank you for the clarification. I went ahead and rotated the OWSC 15 deg about its hinge in SolidWorks and came up with the following new center of mass: [0.469 0 -1.35] m. At this point, my last question is what variable would I assign this new center of mass to in WEC-Sim? would I open the `bodyClass.m` folder and change `cg = [0.469 0 -1.35]`?

Sorry for all of the questions. I didn't think such a seemingly small error initially would result in so many layers of debugging...

nathanmtom, 2021-03-02:
@sugarl14 Thanks for the updates, I'll do my best to provide additional comments. 

1. I'm not familiar with the interface you are cutting and pasting into this thread. But, it does appear that when calculating the hydrodynamics at the hinge your pitch restoring coefficient is positive so it should attempt to return the OWSC to vertical. That being said the coordinate system and other definitions may be different than WEC-Sim so it is hard to trouble shoot given my lack of experience with your interface. 
2. I would step back and model a basic plate like OWSC. In this case there are clear analytical solutions for all of those above parameters. You can then double check your analytical calculations against those from the BEM solver to see if there are any differences or issues with definitions, coordinates, etc. 
3. You do not need to redefine the center of gravity only that at time 0, the center of gravity needs to be displaced. The following body property allows you to define and initial displacement and rotation. 

[image]

But, this will need to be relative coordinates, so the delta between the initial position of the center of gravity as this is the displacement from equilibrium. 

Update us once you are able to run a free decay test to see if you're system returns to equilibrium (vertical). If it does then there is something else causing your WEC to fall over. If the WEC falls over still, then you will need to revisit your BEM set-up to ensure the hydrostatics are being calculated correctly.

Hopefully, these suggestions continue to lead you down a productive path.

Cheers, 
Nathan
National Renewable Energy Laboratory

sugarl14, 2021-03-02:
Hi @nathanmtom,

1) Sorry, they're snippets from Ansys AQWA. From what I've seen, the coordinate system and other definitions transfer appropriately between AQWA and WEC-Sim.

2)Thank you for the suggestion. I will do this just to be sure everything is transferring OK.

3) I was able to get a free decay test. Here, `noWaveCIC` is used with an initial displacement of 0.261799 rads (or 15 degrees). Let me note also that `pto.c` is non-zero.

https://user-images.githubusercontent.com/57150958/109715405-9702dc00-7b71-11eb-94d4-62bab97b5f91.mp4

Here's a plot of the motion response:

[image]

 
 
 Here, `pto.c= 0`. And I can see that the flap returns to vertical over time.

[image]

Looks like it's returning to vertical just fine. So I'm thinking I'm going to have to re-visit my BEM..

nathanmtom, 2021-03-02:
@sugarl14 Thanks for the update. It does look like your hydrostatics are calculating fine. 

1. Thanks for that information, I did see that you had used AQWA in previous posts, but wasn't sure if that was also from the same analysis program. 
2. I'm glad to see that you seem to have been successful at implementing a free decay test. In your first plot, it appears that your system is close to critically damped. While in your second time history trace you have the decay from the radiative wave damping. This would then suggest that your radiative forces are being calculated correctly and are not causing any instability issues.

I'm at a little impasse now since we appear to have checked your hydrostatics and radiation forces. But, in a previous response when you ran `regular` for the wave environment your WEC decided to flop to 30 degrees and oscillate about that angle which makes it seem like the issue is related to the wave-excitation force. 

1. Question: Are you defining `pto.c` and `pto.k` from the wecSimInputFile.m? or are you opening the pto blocks and hard coding them into the interface you attached as a screen shot in a previous post?
2. Request: Can you try and run a `regularCIC` run with `waves.H=0.01` and `waves.T =8.5` while setting `pto.c=0` and `pto.k=0' and see if your WEC still ends up running against your end stops?
3. Request: Can you plot the forceTotal time series output from our free decay run with `pto.c`? This way we can check to see how WEC-Sim is calculating your hydrostatic, mass, and radiation forces.

Sorry to make you go through these extra checks, but it is helpful to get a better grasp on how your model is working. 

Cheers, 
Nathan
National Renewable Energy Laboratory

sugarl14, 2021-03-03:
Hi @nathanmtom, thank you for your response and affirmations. 

1) I am defining `pto.k` and `pto.c` in the wecSimInputFile.m. 

2) I set the wave type to `regularCIC`, changed `H= 0.01`, `T=8.5`, `pto.k=0` and `pto.c= 0`. I ran this for 2 cases:
-Case 1: When `body(1).mass= 200` kg, the flap maintains it's vertical, upright position. (see below)
-Case 2: When `body(1).mass = 700` kg (the original mass of the flap), the flap runs against the end stops. (see below)

Case 1 position results:
[image]

 
 
 Case 2 position results:
[image]

The mass seems to be affecting whether or not the flap flops to its end stops, or maintains vertical positioning.

3) I re-ran the free decay test with initial displacement of 15 degrees with `noWaveCIC` enabled. I did one case for `body(1).mass= 200` and another case for `body(1).mass = 700`. I wasn't sure if you wanted force results from the body  (`output.bodies(1).forceTotal(:,5)`) or the PTO (`output.ptos.forceTotal(:,5)`), so I plotted both.

**case 1:  `body(1).mass =200`**
[image]

**case 2: `body(2).mass = 700`**
[image]

Don't be sorry for all the checks! I see these hiccups as a very valuable learning experience. Your help has been so insightful and much appreciated!

Landon

nathanmtom, 2021-03-03:
@sugarl14 Thanks for running these additional checks. 

1. In my opinion there still appears to be an issue with the mass properties you have defined. I see that your original `body(1).mass = 700` can you let me know what `body(1).dispVol=?` would be? What is your target displaced volume? In the end if we have Cg, Cb, mass, and buoyancy we should be able to do double check to see if there is a positive restoring coefficient in pitch. 
2. Apologies for not being more clear, but these plots were a good start. For the 200 kg mass the force on the body goes to 0 which is consistent with returning to vertical, but for the 700 kg mass the force does seem to grow indicating some forcing is pushing the WEC towards the end stops. For the 700 kg mass, I'd suggest plotting the individual force components such as `output.bodies(1).forceHydrostatics, radidation, mass` (I'm forgetting the exact variable labels but opening up `output.bodies(1).` you should be able to find these forces/torques I'm referencing. Remember that these forces are calculated at the center of gravity and so the forces in surge and heave contribute to a moment about the hinge. Make sure to review these other degrees of freedom to see if there is something causing a net torque about the hinge. 

Update me when you've had a chance to review and hopefully we can get one step closer to solving your problem. 

Cheers, 
Nathan

sugarl14, 2021-03-03:
Hi @nathanmtom, thank you for the suggestions.

1) with `regularCIC`, `H=0.01`, `T=8.5`, and `body(1).mass = ~700`, the displaced volume is `body(1,1).dispVol = 0.9410`. 
`cg = [0; 0; -1.289]`
`cb = [0; -0.0038; -1.895]`
`mass= 692.5`  (this is the real mass of the "~700kg" model)
 
 
2) Here are the individual force components for the ~700kg body taken from `output.bodies(1)` with the same wave settings mentioned above :

### Hydrostatic Restoring Force
. 
[image]
.
.
### Added Mass Force
. (Note: Here, I changed `xlim([0 200])` so we could see the forces closer)
[image]

.
.
### Radiation Damping Force
.(Note: Here, I changed `xlim([0 200])` so we could see the forces closer)

[image]

. Here, the force magnitude due to the heave motion is higher than torque magnitude due to the surge motion.
.
### Excitation Force
.
[image]
.
.

Thank you

Landon

sugarl14, 2021-03-03:
Also, here is the plot for position in the surge and heave motions.
for the surge motion, the y-axis is in degrees (e.g. the flap starts vertical at 0 degrees and then flops down to ~18 degrees)
for the heave motion, the y-axis is in meters (e.g. the flap starts vertical at -1.289 m and then flops down to -1.385)

[image]

nathanmtom, 2021-03-03:
@sugarl14 Thanks, I've been trying to digest your latest plots; however, your latest plot has confused me. 

If you are plotting surge and heave these are translational modes of motion and thus the output from WEC-Sim would be in meters not degrees. WEC-Sim outputs are not in degrees, they would be output from WEC-Sim as radians and you then would convert to degrees. I'll need you to confirm your labeling before I can understand your latest set of plots. 

Also, at this point in time there are a lot of plots and having to track back to reference previous information is getting difficult. Therefore, may I ask that you try and add as much detail (redundancy is fine) to the plots and description as possible to ensure we are tracking the same simulations, inputs, etc.. Some of the plots jump between regular wave excitation and free decay which is making it difficult to remember what is driving your system. For the time being can we go back to looking at the force, moment, and position time histories for the free decay tests. Your latest excitation force plot makes is seem like those force traces are consistent with regular wave excitation and would not be contributing to a net moment that would drive your WEC to hit and hover near an end stop. 

Essentially, we are trying to find what force/torque has a zero-mean value which is the only way the WEC would fall and sit on your defined end stop.

Cheers, 
Nathan
National Renewable Energy Laboratory

sugarl14, 2021-03-03:
Hi @nathanmtom ,

1) Sorry-- I did not mean to add confusion. I will clarify with the following plots: 

**For Surge:**
I converted the `output.position(:,5)` from radians to degrees and then plotted. I used `(180/pi) *output.bodies(1).position(:,5)`. Basically, here we can just see that the OWSC just kind of flops to 18 degrees after ~100 seconds.

[image]

**For Heave:**
I simply plotted `output.position(:,3)` (which is in meters) against time. Here, we can see that the position of the OWSC changes from -1.289 meters to -1.385 meters after ~100 seconds.

[image]

2) I agree- all the plots are getting a bit confusing. I will be sure to specify the different parameters in each figure window also! 

Here are some plots of the free decay test with mass= 700 kg.

**Position (converted from radians to degrees)**
[image]
.
.
.
.

**Hydrostatic Restoring:**

[image]
.
.
.
.

**Added Mass Force**
[image]
.
.
.
.

**Radiation Damping Force**
[image]
.
.
.
.
**Wave Excitation Force**
I'm assuming this plot is zero since we I ran `noWaveCIC` here, and therefore there was no wave to cause an excitation force Would that be correct?
[image]
.
.
.
.

**forceTotal**
[image]
.
.
.
.

please let me know if there's anything else you'd like for me to specify in these plots in the future. Thank you!

nathanmtom, 2021-03-03:
@sugarl14 Thanks for these updates and thanks again for accommodating my request for additional details placed on the plots. 

Can you clarify why in your first set of plots the end stop appears to be at 18 degrees while for the free decay test it is at 35 degrees? Are you using the same end stops for both simulations?

So it does look like the hydrostatic restoring torque is the dominating force about the center of gravity of the WEC. The restoring torque is in the same direction as displacement, i.e. positive displacement and positive torque, when we want positive displacement and negative torque. Therefore, the WEC should keep falling to one side. 

To shed more light can you make a new `.slx` file where the rotational PTO is replaced with a fixed constraint. Then run the free decay tests again as above. The WEC shouldn't move, but now we can look at the fixed constraint forces and look for the moment at the hinge. Since we know all of the other forces at the center of gravity of the WEC you should be able to reproduce the moment about the hinge from a static force and moment calculation. I'm hoping this can show us how the forces calculated at the center of gravity (as you have plotted) get transferred to the hinge. 

Cheers, 
Nathan
National Renewable Energy Laboratory

sugarl14, 2021-03-03:
Hi @nathanmtom,

I used the same end-stops for both simulations. 

In Case 1, where `regularCIC` waves are used with `H= 0.01` and `T= 8.5`, the OWSC seems to settle at 18 deg.
[image]

In Case 2, the free decay test, where `noWaveCIC` is used, and the OWSC is initially displaced 15 deg, the OWSC seems to settle at 35 deg.
[image]

This only happens when the mass of the flap is 700 kg. When I change the mass of the flap to 200, the flap returns to its original vertical position.

.
.
.

2) Here is a screenshot of my new `.slx` file followed by the corresponding wecSimInputFile.m code for the extra fixed constraint.
[image]

[image]
.
.
.
I ran the decay tests again, where `body(1).mass= 700 kg` , `noWaveCIC` is used, and an initial displacement about the hinge of 15 degrees is used.

Here, constraint(2) is the fixed constraint at the hinge. So I have made 2 plots: one for the force in the heave motion, and one for the torque in the surge motion. I have posted them below and tried to add as much clarity as possible.

**Static Force in Heave Motion:** 
[image]
.
.
.

**Static Torque in Surge Motion**
[image]
.
.

Please let me know if I misinterpreted what you were asking me to provide and I will do my best to correct it!

nathanmtom, 2021-03-03:
@sugarl14 Thanks for these updates. So the constraint is applying a negative torque to stop the WEC from moving in the positive rotation direction, which is consistent with the previous free decay response. 

Now, that we have the heave force and the pitch restoring coefficient values, you should be able to use your displacement, hydrostatic coefficients, mass and buoyancy to reproduce what WEC-Sim in outputting in your plots above. My guess is that the heave force should mostly be from the difference between gravity and buoyancy, but this can be quickly checked based on the information you have on hand. 

It may be helpful to include a free body diagram of this system in your next post to show what forces are being calculated and considered in your calculations. 

Cheers, 
Nathan
National Renewable Energy Laboratory

sugarl14, 2021-03-03:
Hi @nathanmtom, thank you for the update and quick response. I have to step away from my computer for a few hours but I will continue to pick this up when I get back! Thank you!

sugarl14, 2021-03-04:
Hi @nathanmtom 

My only question is that I got a positive value for my heave force, and WEC-Sim calculated a negative value. I know I'm probably wrong somewhere but can't seem to find out where..

Here's my FBD and calculation of **Static Heave Force**
[image]

and here is my FBD and calculation of **Static Surge Torque**
[image]

nathanmtom, 2021-03-04:
@sugarl14 Thanks for these schematics! Very helpful to see what is going on with the mass properties. 

1. The force you have calculated is the force on the body which is net positive, but the constraint is the force required to keep the OSWEC hinged to the sea floor. So the output from the constraint should be equal and opposite to your calculations above. 
2. For your moment calculation, given the WEC-Sim coordinate system the flap force generates a positive moment and the buoyancy force creates a negative moment. The net results is a positive moment that will want to keep your WEC rom rotating in the positive direction until you hit your end stop. Similar to point 1 above, since you have a positive moment about the hinge, the hinge needs to apply a negative moment to keep it from moving. 

These results then lead me to believe that essentially that the combination of mass and location of the center of gravity relative to the center of buoyancy means your system does not have a positive restoring coefficient in pitch which is causing your WEC to not return to vertical after being displaced. This was somewhat verified when you reduced the mass of your WEC. The flap force would have decreased by about 70% from your runs (200 kg vs 700 kg) which likely allows the buoyancy force to exceed your flap force and result in a negative moment about Point O returning your system to vertical. 

Therefore, I feel at this point that there is not a bug within WEC-Sim and it is calculating the forces appropriately given your inputs. It seems to me that you will need to redesign your WEC to have a positive restoring coefficient when displacing the WEC as you have done in this example free decay test. 

Cheers, 
Nathan
National Renewable Energy Laboratory

sugarl14, 2021-03-04:
Hi @nathanmtom ,

Thank you for the reply. I will get back to the drawing board and try switching up the material so the mass of the flap is lower. I'll then perform another free decay test to make sure it returns to vertical after an initial 15 degree displacement.

I really cannot express my gratitude for all your help!! Your patience is very much appreciated. Thank you for contributing to a wonderful learning experience for me. I'll keep you updated on my progress!

Take care!

Landon