WEC-Sim/WEC-Sim #481: Read AQWA

akeeste, 2020-12-10:
I am opening a new ongoing issue to discuss modifications of the Read_AQWA() BEMIO function. WEC-Sim+AQWA users please feel free to discuss suggestions here. If you have a specific error or bug with AQWA BEMIO, please open a separate issue.

In summary: the AQWA *.LIS and *.AH1 files have proven difficult to read consistently with a single function. AQWA seems to make small versioning changes to the .LIS format that adversely impacts our ability to read the files with BEMIO. Also, the .LIS and .AH1 files can be changed by a variety of user options.

akeeste, 2020-12-10:
Hi @sugarl14

Thank you for offering to submit ANSYS 18.0 output (continuing our discussion in issue #476). There is not a specific model or mesh that we require. Right now the most important point is to have a sufficient number of wave frequencies and headings (200-300 and 5-10) as this seems to affect the .LIS table set-up.

Also, the AH1 file seems to have a more consistent format than the .LIS output. Is it possible to output either the Froude-Krylov or diffraction forces in the AH1 file instead of the LIS?

sugarl14, 2020-12-10:
Hi @akeeste,

Sounds good. Let me know if there's anything incorrect about this: I will run an analysis on a simple cube. I'll also define a range of wave frequencies so that 300 wave frequencies are available. I'll have 8 wave heading angles.

Yes the .AH1 file does seem to be a bit more consistent with previous versions of Aqwa. I've seen some .AH1 files from previous Aqwa versions, and they all look pretty consistent. 

I'll look into your question about outputting FK and diffraction forces in the .AH1 file instead of the .LIS file!

Landon

akeeste, 2020-12-10:
Hi Landon,

This sounds good, this case will be very helpful! I will look into the .AH1 output as well.

Thanks,
Adam

sugarl14, 2020-12-10:
Running the simulation now. But the AQWA diffraction solver is limited to solving for 100 different wave frequencies. I'll run it at this maximum number. So far I have not found anything pertaining to the FK + Diff forces in the .AH1 file. But ill keep looking around!

akeeste, 2020-12-17:
Hi @sugarl14 

I just wanted to check in and see if you had updates on the AQWA 18 case? I am still working on getting AQWA and looking into the .AH1 possibility.

sugarl14, 2020-12-19:
Hey @akeeste 

My apologies, I have been out of town with limited service, and my computer is at home. I will be back home on the 23rd, so I can send it then. I remember that the file is a bit too large for GitHub, even when I throw it in a zip folder. I was going to put it on a cloud and then share it with you. Do you have an email that works best?

akeeste, 2020-12-21:
Hi @sugarl14 

Thanks for getting back to this and running this case for us. Please share this with akeeste@sandia.gov

Adam

closed 2021-02-10 by kmruehl