WEC-Sim/WEC-Sim #482: Pretension tether?

mleary29, 2020-12-11:
Hi,

I am currently working on a two-body point absorber where the float is attached to a heave plate through a tether. Now, I know two-point cable connections are approximately connected by rotational constraints in WECSim creating a mass less rod.

With this I am struggling to apply the appropriate pre-tension to the rod such that the heave plate's motion is appropriately dampened. I have tried incorporating a translational PTO with pretension, but the result is that the heave plate doesn't move at all regardless of the pretension I set for pto(1). The simulink model for this is shown directly underneath:

[image]

Furthermore, I have tried to dampen the heave plate motion through an additional PTO to the heave plate, but I fear this dampening is disconnecting the relative motion between the heave plate and the float. The added PTO has a very large stiff component which dampens motion.

[image]

Any point in the right direction would be greatly appreciated for simulating a two-body point absorber connected by a tether. Thank you for your time, and please let me know if you have any questions.

Best,
Matt

Aquaharmonics, 2020-12-11:
Hi Matt,

Try downloading this example I quickly made to get the basic concept functioning based on the RM3 two body example:
[  (https://github.com/Aquaharmonics/RM3_PT_Tether)   ]

This is what the Simulink model looks like:

[image]

The convention I picked was starting from the Global Reference frame working up to the spar and then the float, labeling rotational constraints 1-4 and pto's 1 & 2.

If I understand correctly what you are trying to do, you need a rotational constraint at either end of a translational PTO to get the "tether" representation.

 You need to change the PTO's to " Translational PTO Actuation Force" to allow for pre-load.

In this example I use a ramp and saturation block multiplied by a constant preload force that is applied to the spar from the sea-floor, and from the spar to the float. The pre-load value is the same on each body.

Note that for "pre-tension" to pull the spar down or float down the applied actuation force must be negative.

I also added "File Solids" to the model between the body and references of each constraint/PTO/body/frame to help with the "degenerate mass on body/follower side" errors that tend to crop up. 

There are probably a myriad of other ways to do this but this has been my method. You need to direct the file solid to a reference file and I just chose the float.stl file, and then give the point mass under the inertia drop down a value,  in this case something small, say 1 kg. 

[image]

The RM3 example has a depth of 200m, so the first rotational constraint is placed on the seafloor, the second one at the base of the spar, the third one at the middle of the spar, and the fourth one a little below the float. I chose these specific places to ensure some kind of stability between each body and between the spar and the sea floor.

Here are the locations from the input file:

[image]

Constraint 1 and Constraint 2 represent each point of the first tether from the spar to the sea floor, with PTO1 placed between them.

Constraint 1 origin:

[image]

Constraint 2 origin:

[image]

PTO1 origin:

[image]

Constraint 3 and Constraint 4 represent each point of the second tether from the float to spar , with PTO2 placed between them.

Constraint 3 origin:

[image]

Constraint 4 origin:

[image]

PTO2 origin:

[image]

The example has no waves to get the pre-tension concept across.

The delays in the pre-tension allow one to see the applied pre-tension to each body one at a time. Note that I have not changed the damping in the PTO from the RM3 example, so there is damping, but no spring in each PTO.

At 2 seconds the pre-tension starts ramping up on the spar, and you can see the displacement on the spar and some displacement on the float due to damping existing at PTO2 between the spar and float. The float slowly moves back to its starting position due to the restoring force from displacing it downwards.

At 30 seconds, pre-tension ramps up on the float, noting that there will be an opposite reaction to the spar. 

You can see that since the pre-tension on each body is identical, the spar settles to its starting position and only the float remains displaced downward from its initial position.

Body 1 response: 
[image]

Body 2 response:
[image]

Anyways, just thought I might be able to help with a simple quick example. Let me know what you think or if you have trouble accessing the example files.

Best,
Alex

mleary29, 2020-12-11:
Wow this is extremely helpful. Thank you. I guess where I'm still confused now is with the PTO. My only questions are just tomake sense of this PTO. 

For the tether to be represented both PTOs should be identical even if there were a stiffness component? 

These PTOs represent the power take off system for the WEC and are not just present to represent the tether? The repetition is just making me confused as to the physical representation of the PTO system for this example.

Thank you again!

dforbush2, 2020-12-11:
@Aquaharmonics thank you so much!
This is about the approach I was going to recommend and your example is very helpful. 

We have had some luck with treating tethers via a sort of simplified "lumped mass model" where we model the tether as a series of discrete point masses (modeled as drag bodies) separated by rotational ptos. This is a typical approach used to model cables/tethers, but generally they use a fairly large number of discrete elements, but I would recommend only a few here. Obviously, this doesn't accurately the actual motions of the tether very well, but it should do a could job getting the tether forces exerted on the bodies. Tweaking the masses of the drag bodies, and adding a little stiffness/damping at the rotational constraints can approximate any tether inertias and stiffnesses.

Great work to @Aquaharmonics for putting together this example. Would you have any interest in it joining the WEC-Sim Applications repo?

Aquaharmonics, 2020-12-11:
Hi @mleary29,

I may have misunderstood your initial concept a little from the original simulink model. I thought you were looking to have a pto/pretension between the float and the spar, **and** between the spar and the sea floor. 

That is why I applied pre-tension to both PTO's to help explain the physics, namely that if you pre-load the spar with the same force as the float and do the process in steps, you can clearly see that the spar is first pulled down to the seafloor, and then pulled back up to its initial position since the float is reacting against the spar as it is pulled down with pre-tension. In short, the stepwise approach just helped to verify the physics of the concept.

It looks like you are looking instead to have pre-tension and a PTO between the float and spar only, and then a passive mooring line to the seafloor from the spar. **Is this correct?**

As @dforbush2 points out, using the method described, MoorDyn to model the spar plate mooring to the seafloor is a more realistic representation of a passive mooring line element to the seafloor than the method I show in the model.

Also, just to clarify, I am taking "pre-tension" to mean a constant force pre-load , **not** a spring. In the model, you can still apply damping and spring via pto.c and pto.k for pto2 as modeled between the float and spar. Note that these apply symmetrically in the 1 dof of the translational PTO.

If you wanted asymmetry, say  damping only as the float rises relative to the spar, or spring only as the float rises for example, you cannot use the built in spring and damping within the translational PTO block via pto.c and pto.k, and would need to apply those corresponding forces via the same "actuation force" as the pre-tension is applied. 

Let me know if that makes sense, or if I am misunderstanding your concept.

Best,
Alex

Aquaharmonics, 2020-12-11:
Hi @dforbush2 ,

If it is helpful,  adding it to the Wec-Sim Applications repo is fine with us.

Best,
Alex

mleary29, 2020-12-11:
Hi @Aquaharmonics 

Thank you so much again for taking the time to help me with this.

So the WEC I’m concerned with is free floating with a tether connecting the float and spar. Thus, I am concerned with the PTO and pretension tether between float and spar, and so the PTO1 portion portion reacting with the spar reacting against the seabed is confusing at present.

Thank you again!

Aquaharmonics, 2020-12-11:
Hi @mleary29 ,

Take a look at this version:
[ https://github.com/Aquaharmonics/RM3_PT_Tether/tree/master/RM3_1PT_Tether_FreeFloating]

I think it is close to what you are trying to do. I removed constraint 1&2 &  pto1, and replaced all with a 3DOF floating constraint. This would then have 2D planar motion, with the WEC free floating, and a pre-tensioned tether/pto between the spar and float. 

Note, this does not model any mooring to the sea floor, which would still have to be added. 

Let me know your thoughts.

Alex

Aquaharmonics, 2020-12-11:
@mleary29 ,

I would also add a note that if you took the original example and removed the damping and pre-load from pto1, it would essentially be the same model as the one with a 3DOF floating constraint.

mleary29, 2020-12-11:
@Aquaharmonics 

Thank you! This is very clear and makes sense to me! Have a good weekend!

Aquaharmonics, 2020-12-14:
No problem @mleary29 .

@dforbush2, let me know if you need anything to add the 2 examples to the applications repo.

Best,
Alex

mleary29, 2021-02-10:
Hi @Aquaharmonics  and @dforbush2 ,

With this tether or umbilical functionality, I am wondering the ability to measure tension?

Best,
Matt

Aquaharmonics, 2021-02-10:
Hi Matt,

Just to refresh my memory, are you coupling with MoorDyn, or just assuming
the PTO is representing the mooring tether as well?

mleary29, 2021-02-10:
I am assuming the PTO is representing the mooring tether.

Aquaharmonics, 2021-02-10:
In the workspace, *output.pto.forceActuation(:,3)* should contain the
tension on the tether in this case, column 3, heave, 1 DOF.

However, if you do not model any physical mass of the tether itself, it
will be "massless". In that case, it is not a perfect representation of the
true tension on the mooring line, but it can be a reasonable estimate
depending on what the mooring line construction is, i.e., neutrally buoyant
synthetic mooring line VS link chain,  etc.

Alex

mleary29, 2021-02-10:
Thank you Alex that makes good sense! Have a good rest of your day.

closed 2021-02-10 by mleary29