WEC-Sim/WEC-Sim #489: Wave Class: Peak vs Energy Period

Rjan821163, 2021-01-06:
Hello,

I am still getting used to WEC-Sim, and I am currently using the RM3 tutorial to obtain load responses for a certain sea state.
I want to run an irregular wave case with JONSWAP spectrum, but I noticed that the wave period input needs to be a Peak Period (Tp).  The data for my chosen site is given as Energy Period (Te). 

Is there a simple way to make WEC-Sim process the period input as Te instead of Tp? 
Or is there a simple way to convert Te into Tp? 

Thank you!

kmruehl, 2021-01-06:
@Rjan821163 The [JONSWAP spectrum is defined by Peak Period (Tp)](http://wec-sim.github.io/WEC-Sim/man/theory.html?highlight=jonswap#jonswap-js). The Energy Period can then be calculated from a ratio of the -1 and 0 spectral moment, Te = m_{-1}/m_0, using the generated JONSWAP spectrum. We don't provide this functionality in WEC-Sim, but [MHKiT ](https://mhkit-software.github.io/MHKiT/index.html)has functions you can use to quickly generate a JONSWAP spectrum and then calculate the energy period. 

```python
from mhkit import wave
import numpy as np

# Define Frequency Range, Hs, and Tp
omega = np.arange(0.1,3.5,0.01)
f = omega/(2*np.pi)
Hs = 2.5    # Define Hs
Tp = 8      # Define Tp
# Generate JONSWAP Spectrum
S = wave.resource.jonswap_spectrum(f, Tp, Hs)   
S.plot()
# Calculate Energy Period
Te = wave.resource.energy_period(S)
print(Te)
print("Tp = ",Tp)
```

```python
                         Te
JONSWAP (2.5m,8s)  7.238472
Tp =  8
```

Alternatively, if you have data for your chosen site you can load the spectra or time-series data directly into WEC-Sim using the [Wave Class](http://wec-sim.github.io/WEC-Sim/man/code_structure.html#wave-class).

kmruehl, 2021-01-13:
@Rjan821163 Does this answer your question?

Rjan821163, 2021-01-13:
Yes this does! Thank you so much!

closed 2021-01-13 by kmruehl