WEC-Sim/WEC-Sim #492: Unable to execute Free_decay application because of error

BinoBK, 2021-01-08:
Hi all, 
Thanks for helping me out in advance. I am a beginner and learning WECSim. 

I am trying to perform free decay analysis of RM3 model to know the natural frequency of the system. After successfully running the RM3 example file, I tried modifying the same with Free_decay application code given in ---> https://github.com/WEC-Sim/WEC-Sim_Applications/tree/master/Free_Decay

I created folders with 0m, 1m, 1m-ME, 3m, 5m. When I run the runFreeDecay.m file, I get the error:

WEC-Sim Read Input File ...   
Error using bodyClass/checkinputs (line 406)
The hdf5 file hydroData/rm3.h5 does not exist

To my surprise, there exists an "rm3.h5" file in "hydroData" folder. Can someone let me know the mistake? Is there any other documentation which I am supposed to refer? 
Thanks for the help.
Bino

closed 2021-01-09 by BinoBK

BinoBK, 2021-01-09:
I realized that I just had to replace as shown below to solve the issue. 

body(1) = bodyClass('hydroData/rm3.h5');          ---->                       body(1) = bodyClass('E:\WEC-Sim-master\examples\FreeDecayRM3\hydroData\rm3.h5')

Thanks.

ChrisJohnsonJ, 2021-01-28:
Hi Bino,
I am a new user to WEC-Sim and doing Free Decay tests on a heaving sphere. I saw your query regarding the error when you ran Free decay. I was wondering if we could get in touch to discuss how you went about generating the results. I saw the responses to your query but I would like to know the procedure behind it. It would be very helpful for me to relieve a lot of stress in the project I am doing, if we could chat. 

I look forward to your reply. Thanks in advance.

BinoBK, 2021-01-28:
Hi Chris,
Sure, I will definitely like to help. Even I am learning everyday. I could successfully perform FreeDecay test with sphere. Now I am trying with a multi-body model. 

I am sure other members from WEC-Sim would too help you if you post any queries. 
Thanks,
Bino

ChrisJohnsonJ, 2021-01-28:
Hi Bino,
It's great that you could successfully do the FreeDecay test with sphere. In my case, however, I am encountering errors with the CoG and the graph is oscillating about a non zero point. Is there any way to get in touch with you via private message? My email id is 'chrisjohnsontud@gmail.com'. It would be easier for me to discuss the issue I am facing. I tried posting a query on the forum but I couldn't get a satisfactory answer. I hope we could connect soon.

Thanks,
Chris