WEC-Sim/WEC-Sim #493: Error while using MCR: "Matrix dimensions must agree"

BinoBK, 2021-01-10:
Hello everyone,

I have an issue while running MCR applications (both RM3_MCROPT1 and RM3_MCROPT2). I am getting following error:

***********************************************************************************************
Error using  / 
Matrix dimensions must agree.

Error in waveClass/waveSetup (line 201)
                        obj.w = 2*pi/obj.T;

Error in wecSim (line 144)
waves.waveSetup(body(1).hydroData.simulation_parameters.w, body(1).hydroData.simulation_parameters.water_depth, simu.rampTime, simu.dt, simu.maxIt, simu.g, simu.rho,  simu.endTime);
*************************************************************************************************

The same error was also discussed earlier in ISSUE no #438 (Below is the link to it)
https://github.com/WEC-Sim/WEC-Sim/pull/438#pullrequestreview-519072236

Has this error been resolved? Thanks a lot for the help. 

Regards,
Bino

kmruehl, 2021-01-13:
@BinoBK I am also unable to reproduce this error with the [Multiple_Condition_Runs application case](https://github.com/WEC-Sim/WEC-Sim_Applications/tree/master/Multiple_Condition_Runs). Similar to my comments in #494 for debugging:

- Do you have the latest version of WEC-Sim from the master branch of https://github.com/WEC-Sim/WEC-Sim?
- What version of MATLAB are you running?
- Are you running this case by executing `wecSimMCR` or `wecSimPCT`?

kmruehl, 2021-01-13:
@BinoBK following up, because I think this issue relates to #494 

- I recommend using the latest version of WEC-Sim from master, instead of the v4.2 release. It's possible that the v4.2 release is causing the issue, and if that's the case we can tag a bug-fix release.
- Also, I recommend verifying that WEC-Sim has been properly installed and that only the WEC-Sim/source directory has been added to the MATLAB path.

BinoBK, 2021-01-14:
Thank you so much @kmruehl. My case is working now. 

I made following changes: 
1. Used the latest version of WEC-Sim instead of V4.2.
2. Changed my Matlab version from R2020a to R2020b. Also made sure ONLY source directory has been set to Matlab path. 
3. for execution used "wecSimMCR" instead of "wecSim".

kmruehl, 2021-01-14:
@BinoBK that's great news. Do you know which one of those things resolved your issue, or if it was a combo? I'm going to close this issue since it's been resolved, but am curious what resolved the issue. Thanks.

closed 2021-01-14 by kmruehl

BinoBK, 2021-01-15:
@kmruehl, I think it is the use of "wecSim". When I tried executing using "wecSim", I was able to generate the same error.

But when I use "wecSimMCR", the code works fine. 

Thanks again.

BinoBK, 2021-01-15:
I have a new problem now. The output matrix is always 2x2, even when I try to change the wave condition 
From
waves.H = 1.5:1:2.5;                   	% Wave Height [m]
waves.T = 6:2:8;                        % Wave Period [s]

To 
waves.H = 1.5:0.5:2.5;                   	% Wave Height [m]
waves.T = 6:1:8;                        % Wave Period [s]

Is it not supposed to get a 3x3 power matrix?

Thank you.

kmruehl, 2021-01-15:
@BinoBK post-processing plots are generated in the `userDefinedFunctionsMCR.m` script. If you change the number of cases run, you also need to revise `userDefinedFunctionsMCR.m` to account for the new cases when plotting the power matrix.