WEC-Sim/WEC-Sim #494: Error while using FreeDecay: "Index exceeds the number of array elements"

BinoBK, 2021-01-10:
Hello all,

While using the FreeDecay application with sphere, I am getting following error:

*************************************************************************************
Index exceeds the number of array elements (1).

Error in responseClass/plotResponse (line 247)
            t=obj.bodies(bodyNum).time;

Error in userDefinedFunctions (line 14)
output.plotResponse(2,3);

Error in wecSim (line 303)
    userDefinedFunctions;

Error in runFreeDecay (line 9)
    wecSim
*************************************************************************************

1) Since I am new to WEC-Sim, I am unable to find the solution. Please help me out.

2)  What is "simu.adjMassWeightFun = 0.47761120825;" that is included in the code wecSimInputFile.m? 

Kindly provide the support. Thanks in advance. 

Regards,
Bino

kmruehl, 2021-01-13:
@BinoBK 

1. I am unable to reproduce this error when running the [Free Decay applications case](https://github.com/WEC-Sim/WEC-Sim_Applications/tree/master/Free_Decay). A couple of possibilties:
- Have you verified that you have the latest version of WEC-Sim? 
- Are you running from the case directory? 
- What version of MATLAB are you using?

[image]

2. [`simu.adjMassWeightFun`](http://wec-sim.github.io/WEC-Sim/dev/man/api.html#objects.simulationClass.adjMassWeightFun ) is the weighting function for adjusting added mass term in the translational direction due to the way we handle added mass in WEC-Sim.

BinoBK, 2021-01-13:
Hi,
I am using WecSim Version 4.2 now and Matlab version R2020a. I am still getting the same error. 
[image]

I have attached below my model file. Thanks for the support. 
Regards,
Bino 

[Uploading FreeDecay_Sphere.zip…]()

BinoBK, 2021-01-13:
[FreeDecay_Sphere.zip](https://github.com/WEC-Sim/WEC-Sim/files/5810415/FreeDecay_Sphere.zip)

kmruehl, 2021-01-13:
Hi @BinoBK. I'm not able to reproduce this issue, but I did make some minor revisions to get this case running (e.g. using relative paths). [FreeDecay_Sphere.zip](https://github.com/WEC-Sim/WEC-Sim/files/5810725/FreeDecay_Sphere.zip)

I recommend using the [latest version of WEC-Sim from master,](https://github.com/WEC-Sim/WEC-Sim) instead of the v4.2 release. It's possible that the v4.2 release is causing the issue, and if that's the case we can tag a bug-fix release. 

Also, I recommend [verifying that WEC-Sim has been properly installed](http://wec-sim.github.io/WEC-Sim/dev/man/getting_started.html#install-wec-sim) and that only the `WEC-Sim/source` directory has been added to the MATLAB path.

kmruehl, 2021-01-14:
@BinoBK has this issue been resolved as well? related to #493

BinoBK, 2021-01-15:
Hi @kmruehl , I am able to generate the graph, but getting the following error:
[image]

Thanks for help.

kmruehl, 2021-01-15:
@BinoBK this is an issue with labeling the legend, that should have been resolved in the zipped file I sent you. If not, please replace the `plotFreeDecay.m` script with this:

```matlab
%Example plotting free decay

clear all; close all; clc;
casedir = {"0m","1m","1m-ME","3m","5m"};

figure

for ii = 1:length(casedir)
    cd([string(casedir{ii}) + '/output'])
    load sphere_matlabWorkspace.mat
    plot(output.bodies.time,output.bodies.position(:,3)-body(1).cg(3))
    hold on
    cd ../..
end

title('IEA OES Task 10')
xlabel('Time (s)')
ylabel('Displacement (m)')
for i = 1:length(casedir)
    leg{i} = ['z_0 = ' + casedir{i}];
end
legend(leg,'location','northeast')
```

closed 2021-01-15 by kmruehl

BinoBK, 2021-01-17:
Thank you @kmruehl , The zip file worked.