WEC-Sim/WEC-Sim #495: Error while using PTO: "Reference to non-existent field 'time'."

BinoBK, 2021-01-10:
Hello all,

As I am slowly learning and getting familiar with WECSIM, I discovered an issue that I am facing while simulating a custom PTO model with "sphere" floating body. I had to modify the simulink model also. Once I execute "wecSim", I am getting following error:

**********************************************************************************************
Post-processing and saving...   
Reference to non-existent field 'time'.
Error in responseClass (line 170)
                    obj.constraints(ii).time = constraintsOutput(ii).time;
Error in postProcess (line 87)
output =responseClass(bodiesOutput,ptosOutput,constraintsOutput,ptosimOutput,mooringOutput,waves.type,waves.waveAmpTime,hspressure,wpressurenl, wpressurel, simu.yawNonLin);
Error in wecSim (line 300)
postProcess

**********************************************************************************************

Please note: I has to insert the following in the wecSimInputFile:

simu.CITime = 10;

I am attaching the model ZIP file along with this post. Kindly help. 

Thanks and Regards,
Bino
[Sphere.zip](https://github.com/WEC-Sim/WEC-Sim/files/5793124/Sphere.zip)

akeeste, 2021-01-13:
@BinoBK based on the responseClass inputs, it does not look like this is the current version of WEC-Sim. What version are you using?

jleonqu, 2021-01-13:
Hello @BinoBK 

Looks like you are initializing a constraint block in your wecSimInputFile, but you don't have one in your Simulink model. You just need to comment the constraint initialization lines in your wecSimInputFile (line 33 and line 34) and your model will run. Please look at the attached ZIP file with the script corrected.

[Sphere.zip](https://github.com/WEC-Sim/WEC-Sim/files/5810278/Sphere.zip)

BinoBK, 2021-01-13:
Hi,
I used the latest version 4.2, but still I am getting the same error as shown below:

********************************************************************
Reference to non-existent field 'time'.
Error in responseClass (line 193)
                    obj.constraints(ii).time = constraintsOutput(ii).time;

Error in postProcess (line 98)
output =
responseClass(bodiesOutput,ptosOutput,constraintsOutput,ptosimOutput,mooringOutput,waveOutput,
simu.yawNonLin);

Error in wecSim (line 366)
postProcess
 ***********************************************************************

**I am attaching the model file below.** Thanks a lot for the support. 

Regards,
Bino

[Sphere.zip](https://github.com/WEC-Sim/WEC-Sim/files/5810292/Sphere.zip)

BinoBK, 2021-01-13:
Thanks a lot @jleonqu. The error is resolved now.

kmruehl, 2021-01-13:
Thanks @BinoBK and @jleonqu  I'm closing this issue now.

closed 2021-01-13 by kmruehl