WEC-Sim/WEC-Sim #496: OSWEC Example SW file?

sugarl14, 2021-01-13:
Hi again!

I am looking to do a comparison study between AQWA and WAMIT and was wondering if you all could possibly provide the SolidWorks geometry files used in the OSWEC example? I know you all used WAMIT for the OSWEC example and so I'd like to match the results using AQWA. Also, any useful information such as density, water depth, surface piercing amount (if any), number diffracting elements used in the mesh, and wave frequency range would be great! 

Thank you! :)

kmruehl, 2021-01-13:
@sugarl14 I do not have access to the SolidWorks geometry used for the OSWEC example, but we do provide the WAMIT run in the [WEC-Sim/examples/BEMIO/WAMIT/OSWEC](https://github.com/WEC-Sim/WEC-Sim/tree/master/examples/BEMIO/WAMIT/OSWEC) directory and a NEMOH run of the OSWEC in the the [WEC-Sim/examples/BEMIO/NEMOH/OSWEC](https://github.com/WEC-Sim/WEC-Sim/tree/master/examples/BEMIO/NEMOH/OSWEC) directory. These runs include information about the water density, and water depth, along with the geometry files and mesh used for each run. 

The OSWEC dimensions are also provided here: http://wec-sim.github.io/WEC-Sim/dev/man/tutorials.html#oscillating-surge-wec-oswec, and could be use to develop a SolidWorks model of the geometry. I hope this helps!

kmruehl, 2021-01-15:
@sugarl14 I found a CAD model of the OSWEC geometry, see attached.

[OSWEC.zip](https://github.com/WEC-Sim/WEC-Sim/files/5817800/OSWEC.zip)

closed 2021-01-15 by kmruehl

sugarl14, 2021-01-15:
Hey Kelley,

Sorry I forgot to reply to your first response. But thank you so much!!