WEC-Sim/WEC-Sim #498: Error while using WECCCOMP_Fault_Implementation

BinoBK, 2021-01-17:
Hi all,
I am facing an issue while running WECCCOMP_Fault_Implementation case. Following is the error that I am getting:

[image]

I am using Matlab R2002b and latest WEC-Sim. Kindly extend help. 

**Also, Is it possible to share the WEC-Sim model that was shared for the competition "A competition for WEC Control Systems" in 2017.** 

Thanks,
Bino

BinoBK, 2021-01-17:
Thanks to the solution provided in #497 , after adding body(1).cg=[0 0 0], I was able resolve the above mentioned error. 

**As requested above, Is it possible to share the WEC-Sim model that was shared for the competition "A competition for WEC Control Systems" in 2017.**

nathanmtom, 2021-01-18:
@BinoBK Thank you for this notification, we will need to update the WECCCOMP files in the WEC-Sim Application Repository. 

Please see the following repository that holds the WEC-Sim Model used during WECCCOMP

https://github.com/WEC-Sim/WECCCOMP

Cheers, 
Nathan
National Renewable Energy Laboratory

BinoBK, 2021-01-19:
Thank you @nathanmtom

closed 2021-01-20 by nathanmtom