WEC-Sim/WEC-Sim #500: FFT feature in WEC-Sim Free decay test

BinoBK, 2021-01-19:
Hi all,
I was curious if Fast Fourier Transform feature is available for identification of Natural frequency from free-decay results. 

Is FFT feature in-built in WEC-Sim? 

Thanks,
Bino

nathanmtom, 2021-01-19:
@BinoBK  Thank you for your question about a WEC-Sim FFT function. WEC-Sim currently doesn't have a built in function to call to perform and FFT on a given time signal of interest; however, MATLAB has the base `fft` function with examples [here](https://www.mathworks.com/help/matlab/ref/fft.html) to help you build your own FFT script.

Cheers, 
Nathan
National Renewable Energy Laboratory

BinoBK, 2021-01-21:
Thanks a lot @nathanmtom

kmruehl, 2021-02-01:
I opened an issue on the MHKiT repo to request this feature.

closed 2021-02-01 by kmruehl