WEC-Sim/WEC-Sim #504: crane ship simulation

ParallelMech, 2021-01-23:
I'm planning develop a simulation software of a crane ship with a suspended load.
This is a coupled simulation of the floating ship(decoy) on the sea (fluid part) and a crane with a suspended load on the ship (dynamic part).
Is it possible to develop the above software by WEC-Sim ?

nathanmtom, 2021-01-24:
@ParallelMech Thank you for your question. If I have understood your question, you want to ensure that gravitational forces are accounted for in the dynamics of your suspended block? If so, then yes, WEC-Sim includes a non-zero gravity term when calculate the dynamics of all bodies. 

Cheers, 
Nathan
National Renewable Energy Laboratory

ParallelMech, 2021-01-25:
Thank you for your quick response. 
I want to develop a system combining the WEC-Sim and SimscapeMultibody. I just made a sample system as attached pdf.
WEC-Sim Part (surrounded by red box) is from the RM3, and I added a inverted pendulum using blocks of SimscapeMultibody (blue box). I modified the wecSimImput, then run wecSim.
However, It doesn't seem no gravitational force act on the pendulum.
I guess the reason that the gravity of the Mechanism Configuration in the Global Reference Frame is set as [0 0 0].
So, I temporary added a external force on COG of the pendulum as a gravity force (green box).
I guess my method is may be ok, because it goes well. However, I don't think my method is not sophisticated. 
If you have another solution for the handling the gravity for the body, please let me teach.
[WEC-Sim_and_Multibody.pdf](https://github.com/WEC-Sim/WEC-Sim/files/5865149/WEC-Sim_and_Multibody.pdf)

nathanmtom, 2021-01-28:
@ParallelMech Thank you for your response and sharing your image of the modified WEC-Sim model. I may have misunderstood your initial question, but agree that some of the standard Simscape Multibody may need to be modified slightly to interface with WEC-Sim. However, our library includes options for adding nonhydrodynamic bodies which essentially are masses described by buoyancy and gravity and we have incorporated them into masses moving above the water similar to your case. The best available example we have would be our WECCCOMP example found here: https://github.com/WEC-Sim/WECCCOMP

After reviewing this model and seeing how we implement nonhydrodynamic bodies may provide you a better idea on how to achieve your desired modeling. 

Let us know if you have additional questions or comments. 

Cheers, 
Nathan
National Renewable Energy Laboratory

ParallelMech, 2021-01-29:
Thank you for your quick response and kind comment.
I have some errors while running the WECCCOMP model on MATLAB 2020b.
The error messages are as follows. (The original messages are in Japanese, I translated them into English).
I'd appreciate you if you would help me with fixing the problem.

Error: wecSim (line 354)
    Invalid setting in block'WaveStar / Arm / Non-Hydro Body / Net Buoyancy Force / Center of Buoyancy' for parameter'Value'
    Error: wecSim (line 354)
    Invalid setting in block'WaveStar / BC / Non-Hydro Body / Net Buoyancy Force / Center of Buoyancy' for parameter'Value'
    Error: wecSim (line 354)
    Invalid setting in block'WaveStar / Frame / Non-Hydro Body / Net Buoyancy Force / Center of Buoyancy' for parameter'Value'
    Error: wecSim (line 354)
    Invalid setting in block'WaveStar / Motor / Non-Hydro Body / Net Buoyancy Force / Center of Buoyancy' for parameter'Value'

kmruehl, 2021-01-29:
WECCCOMP question related to #498

nathanmtom, 2021-01-29:
@ParallelMech My apologies about the error that you encountered with the WECCCOMP model. This is because of a recent update to the code structure that @kmruehl has cited above in reference to issue #498. The fastest way to fix this error is to update the wecSimInputFile.m and adding body(ii).cg=[0 0 0], where ii corresponds to the index of the non-hydro bodies. 

Let us know if that solves your problem and if not we can work together to resolve your issue. 

Cheers, 
Nathan 
National Renewable Energy Laboratory

ParallelMech, 2021-01-30:
Thank you again for your kind reply. 
I added body(ii).cb=[0 0 0] (not body(ii).cg=[0 0 0]), then the program run with no error.
(However, I need further understanding about the Non-Hydro Body.)

Thanks.

nathanmtom, 2021-02-01:
@ParallelMech 

Glad that you were able to run the WECCCOMP example, the best way to see what forces are being calculated in the non-hydro body is to go under the mask in the non-hydro body block. 

I've taken a screen shot and placed it below. You will see that the forces calculated will be gravitational forces that are applied to the body. As you have pointed out below, WEC-Sim does not automatically activate gravity, so gravitational forces will need to be calculated separately and then these forces feed into the dynamics solver. Therefore, we suggest if you have a hanging mass above the water line to use the non-hydro body rather than masses from the Simscape Multibody library. 

[image]

I hope this information helps understand WEC-Sim's implementation, but let me know if you need additional information. 

Cheers, 
Nathan
National Renewable Energy Laboratory

ParallelMech, 2021-02-04:
Thank you very much for your comment.
I'll try to use the non-hydro body.

Best regards

nathanmtom, 2021-02-10:
@ParallelMech Glad that my description helped you understand how WEC-Sim is working. Let us know if you have any questions and would definitely like to hear move about the results of your efforts. Do not hesitate to open a new issue if you have other questions related to WEC-Sim.

Cheers, 
Nathan
National Renewable Energy Laboratory

closed 2021-02-10 by nathanmtom