WEC-Sim/WEC-Sim #505: Wave-Powered Desalination System (From Webinar 3)

gglosson, 2021-01-26:
Hello,

I was wondering if I could potentially get access to the file for the desalination system from the third webinar on the WEC-Sim page. I know that there is a desalination application under the WEC-Sim_Applications repository but I am relatively new to Simulink and the one from the webinar is much simpler to understand. Any help would be greatly appreciated!

Thank you!

yuyihsiang, 2021-01-29:
Unfortunately, that was an old model, developed using PTO-Sim, as oppose to using SimScape Fluids toolbox, and we don't have one available right now. 

We are doing an upgrade on PTO-Sim right now. After it is done, we should be able to roll out a desalination example using PTO_Sim.

--Yi-Hsiang

gglosson, 2021-01-29:
Ok. Thank you for taking the time to look.

closed 2021-01-29 by gglosson