WEC-Sim/WEC-Sim #507: Use MATLAB merge tools in CI testing

H0R5E, 2021-02-01:
I was wondering if it might be possible to integrate the [MATLAB git external merge tool](https://uk.mathworks.com/help/simulink/ug/customize-external-source-control-to-use-matlab-for-comparison-and-merge.html) into the testing workflow?

We could run a fake merge in Travis (and/or another CI framework), just to indicate if there is a conflict before merging a PR (which will overwrite the existing file without comparison).

akeeste, 2021-04-07:
I finally got around to looking at this tool and testing it out. It is very useful to see and merge library conflicts locally, but I'm unsure if it's beneficial to use on Travis or a CI framework. 

The tool: 
This setup allows command line git to optionally use the Matlab diff, merge and automerge tools to compare matlab binary files. It is convenient and easy to merge two branches of a library with different changes:
- If only one branch has made a change to each subsystem, Matlab can automatically combine these versions in the command line
- If both branches have made changes to the same subsystem, a Simulink-based GUI opens. This interface allows users to easily track and view disparate changes across two library versions simultaneously. 

The tools can quickly be set-up with the following steps:
1. Run the following commands in Matlab
	comparisons.ExternalSCMLink.setup()
	comparisons.ExternalSCMLink.setupGitConfig()
2. Add line to .gitattributes
	*.slx binary merge=mlAutoMerge
3. Use the tools in git command line with the commands:
	git difftool -t mlDiff <revisonID1> <revisionID2> myModel.slx
	git mergetool -t mlMerge myModel.slx

akeeste, 2021-04-07:
The CI environment could use this matlab merge in a couple ways. It could test if the library can be automerged, or has a conflict. Then it could either:
1. Comment on Github indicating if the PR can be automerged or not.
2. Comment on Github, and then actually perform the automerge and commit to the PR if possible.

The user workflow without a CI environment is very similar:
1. User completes commits to a PR locally.
2. User pulls the trunk branches locally and uses the matlab merge tool to indicate any conflicts.
3. User automerges if possible, or uses Matlab merge tool to manually merge
4. User pushes merged library to Github

In my opinion, the CI tool just commenting status is not as valuable because a user would still have to pull the branches, merge conflicts with the tool and commit/push a new change. If the CI tool could automerge and commit to the branch it would save users time. However, this itself was very quick for me to do when testing. I think it will be easier to integrate this into our PR workflow for the libraries, but leave out the CI environment.

H0R5E, 2021-04-08:
> In my opinion, the CI tool just commenting status is not as valuable because a user would still have to pull the branches, merge conflicts with the tool and commit/push a new change.

Thanks for looking into this Adam. I would argue that preventing conflicting merges is mainly beneficial because it stops other people's work being overwritten.

akeeste, 2021-04-09:
@H0R5E I completely agree, conflicting merges with the library has been problematic and we want to avoid overwriting other changes. It would be nice to have a CI tool set-up to check for this. 

I'm more unsure whether this is the right tool to do that with. I haven't tested the tool extensively yet, but from what I saw the automerge was tricky to get working correctly. I wouldn't want to set it up and add to the CI tests if its not robust. That being said, if we go through with the Simulink additions in PR #548, library changes will be even more critical to manage.

I'll continue using it with some other PRs and see if I can get it working robustly.

Adam

akeeste, 2021-06-18:
Hi @H0R5E 

Following up on our discussion on Wednesday -- I created two branches off of dev with test library commits. Each has added one comment in a block mask initialization: 
Flex body mask ([library_test1](https://github.com/akeeste/WEC-Sim/tree/library_test1))
Fixed constraint ([library_test2](https://github.com/akeeste/WEC-Sim/tree/library_test2))
The .slx files are also uploaded here: [WECSim_Lib_libraryTest12.zip](https://github.com/WEC-Sim/WEC-Sim/files/6678838/WECSim_Lib_libraryTest12.zip)

When attempting to merge these branches locally, the MATLAB merge tool will not automerge the two disparate mask changes. The tool catches some minor differences in how each library is saved. 

When opening the simulink merge interface manually, the important mask changes are merged correctly. The problematic changes can be overwritten easily, but are not automerged. I usually see this under the hidden category "/BdWindowsInfo/WindowsInfo/EditorsInfo/Object/". There may be some Simulink setting that determines what is ignored, but I have not found how to set this in the automerge tool yet.

My workflow is something like this:
``git checkout library_test1``
``git merge library_test2``

     Auto-merging source/lib/WEC-Sim/WECSim_Lib.slx
     CONFLICT (content): Merge conflict in source/lib/WEC-Sim/WECSim_Lib.slx
     Automatic merge failed; fix conflicts and then commit the result.

``git mergetool -t mlMerge source/lib/WEC-Sim/WECSim_Lib.slx``

     (Opens Simulink interface....)

I confirmed the tools are installed with "git mergetool --tool-help"

Any suggestions are appreciated, it would be nice to have this integrated into a test.

Adam