WEC-Sim/WEC-Sim #510: output/pto/powerInternalMechanics ???

aglassFHKiel, 2021-02-09:
Can anybody explane what kind of Mechanical Power "powerInternalMechanics" is?

Is this the mechanical power input for the PTO-System e.g. for PTO-Sim?

best regards
Andreas

yuyihsiang, 2021-02-10:
It is just the force multiplied by the velocity. Since it indicates absorbing power by the joint, it is negative given based on the coordinate system.

kmruehl, 2021-02-24:
@aglassFHKiel did this answer your question?

aglassFHKiel, 2021-02-25:
Is it the **forceTotal** multiplied by the velocity?

kmruehl, 2021-03-03:
pto.powerInternalMechanics = pto.forceInternalMechanics*pto.c

aglassFHKiel, 2021-03-04:
Thank you Kelley,

i suppose that pto.c is the relative velocity between float and spar,
further is pto.forceInternalMechanics the force between float and spar.

But where can I find information about InternalMechanics? 
How is the equation of pto.forceInternalMechanics?

best regards
Andreas

kmruehl, 2021-03-11:
@aglassFHKiel my apologies, I responded to this post too quickly and I provided you incorrect information. 

`output.ptos.powerInternalMechanics = output.ptos.forceInternalMechanics.*output.ptos.velocity`

- `output.ptos.forceInternalMechanics` is the pto force between float and spar.
- `output.ptos.velocity` is the relative velocity between float and spar
- `pto.c` is the pto damping between the float and the spar

kmruehl, 2021-03-17:
@aglassFHKiel did this resolve your issue?

aglassFHKiel, 2021-03-18:
Hi Kelley, thanks for your support. Yes this resolved the issue.

best regards
Andreas

closed 2021-03-18 by aglassFHKiel