WEC-Sim/WEC-Sim #515: Optimizing PTO and damping values for each wave frequency

sugarl14, 2021-02-17:
Hi,
I have a few questions about something I've seen in a few publications and was hoping to learn more about:

I have noticed that authors sometimes calculate the optimum PTO damping coefficient for each wave frequency when using a linear PTO. They use the following equation:
[image]

What is the main reason for finding the optimum damping values? Is it to find a starting point for `pto.c` in the wecSimInputFile? In a real-life PTO (without any controllers), wouldn't you have to keep the damping and stiffness values constant after you build the WEC?

Furthermore, is there an option already available in WEC-Sim that would calculate the plot for optimum PTO damping for each wave frequency or would I need to calculate it manually using my BEM results from AQWA?

nathanmtom, 2021-02-18:
@sugarl14 The equation you have listed is correct, assuming I is the inertia (mass if translational), A is added mass, C is restoring coefficient, \omega is wave frequency, and B is wave radiation wave damping and under regular wave-excitation and unconstrained motion. Also, you will need to be careful about where the hydrodynamic coefficients are calculated when inserting into the above equation. The equation above is normally used for a single-degree of freedom system, which for a heaving device doesn't require modification, but for a rotating system like an oscillating surge wave energy converter the coefficients would need to be defined about the point of rotation rather than the center of gravity required by WEC-Sim. You can transfer the hydrodynamics from the center of gravity to the hinge, but this option is not available in WEC-Sim.

The reason for finding the optimum damping value would be to maximize power for a given sea condition. Perhaps the required damping value is too high for a physical or desired PTO system can match, so you may need to perform a parameter search to identify the best damping coefficient for your system. 

If the PTO does not have it's own controller and cannot be adjusted after deployment, then yes, the damping and stiffness values would be constant; however, most PTOs (hydraulic or direct-drive) are normally accompanied by a control system that allows them to have a variable damper and possibly stiffness coefficient. There is quite a bit of literature on this topic and a fairly quick search should provide several examples of real-life PTO systems with this capability. 

At this stage in time there is no option within WEC-Sim that can calculate the optimum PTO damping at each wave frequency, but hope to incorporate some option in the future to assist with this calculation. Therefore, at the moment you would need to calculate the optimum PTO damping at each frequency yourself, using your choice hydrodynamic coefficients, and then input using `pto.c` and `pto.k` in a PTO block.

I hope this explanation was clear, but let me know if you have additional questions or need clarification. 

Cheers, 
Nathan 
National Renewable Energy Laboratory

sugarl14, 2021-02-18:
Hi Nathan,

Thank you for the response. I have replied to some of your comments just to make certain I am understanding everything correctly! Please correct me if anything I say is incorrect/misleading..

`The equation you have listed is correct, assuming I is the inertia (mass if translational), A is added mass, C is restoring coefficient, \omega is wave frequency, and B is wave radiation wave damping and under regular wave-excitation and unconstrained motion. Also, you will need to be careful about where the hydrodynamic coefficients are calculated when inserting into the above equation. The equation above is normally used for a single-degree of freedom system, which for a heaving device doesn't require modification, but for a rotating system like an oscillating surge wave energy converter the coefficients would need to be defined about the point of rotation rather than the center of gravity required by WEC-Sim. You can transfer the hydrodynamics from the center of gravity to the hinge, but this option is not available in WEC-Sim.`

Gotcha! So at this point I would have two different BEM files: 
1) Contains the hydrodynamic coefficient results about the OWSC hinge to be used to calculate the optimum damping coefficient (using the above equation). And
2) Contains hydrodynamic coefficient results about the COG to be used in WEC-Sim.

`The reason for finding the optimum damping value would be to maximize power for a given sea condition. Perhaps the required damping value is too high for a physical or desired PTO system can match, so you may need to perform a parameter search to identify the best damping coefficient for your system.`

This makes sense.  In your last sentence you mentioned a parameter search. This can be accomplished with wecSimMCR, correct?

Thanks again!

Landon

nathanmtom, 2021-02-18:
@sugarl14 Thanks for the for the quick follow-up.

Yes, probably the easiest approach is to have two hydrodynamic solutions at different points on your WEC. Since this should be a minor change in your input files. As mentioned above, theoretically you can shift the coefficients from the center of gravity to another location but the effort to write up a script and verify will likely take longer than running your BEM code twice. 

Yes, you could use wecSimMCR to iterate over a range of PTO damping and stiffness values that are feasible for your chosen PTO design. This is a common practice among WEC-Sim users and we should have some examples in the WEC-Sim Applications Repository. 

Let me know if you have any other questions or feel free to close this issue if you're happy with our response. 

Cheers, 
Nathan
National Renewable Energy Laboratory

sugarl14, 2021-02-22:
@nathanmtom Thank you for the response. It's been very helpful! I will close the issue now and re-open it if I have any further questions!

Take care,
Landon

sugarl14, 2021-02-22:
My apologies, I just thought of another question.

I noticed in some publications and even the WEC-Sim Applications Repository that the PTO stiffness coefficient, `pto.k = 0`. Is there a specific reason for this?

nathanmtom, 2021-02-22:
@sugarl14 No problem, happy to try and answer your latest question. 

The inclusion of a non zero `pto.k` requires that your PTO act as a motor and return energy to the waves for a portion of the wave period. This allows for better phase control of the WEC velocity and can theoretically improve power capture; however, it is well known that this is highly dependent on the efficiency of your PTO system. The lower the efficiency the less return you get for having this motoring capability. Therefore, a lot of PTOs are designed or controlled such that they only apply a resistive force against the motion direction which can be just represented with a `pto.c` term. 

If desired, can provide some references to aid in understanding this requirement. 

Cheers, 
Nathan
National Renewable Energy Laboratory

sugarl14, 2021-02-22:
@nathanmtom 

Thank you for the explanation. That clears things up pretty nicely. I would love for some references to read up on it a bit more. 

Thank you!

nathanmtom, 2021-02-22:
@sugarl14 Sure, happy to provide a few references that should help provide additional background information on this topic. 

1. [Energy-Maximizing Control of Wave-Energy Converters: The Development of Control System Technology to Optimize Their Operation](https://ieeexplore.ieee.org/document/6898109).
2. [Revisiting Theoretical Limits for One Degree-of-Freedom Wave Energy Converters](https://asmedigitalcollection.asme.org/energyresources/article/143/9/091301/1091987/Revisiting-Theoretical-Limits-for-One-Degree-of).

These two references should be a good start and should provide other technical documents to review as desired. 

I'm going to close this issue, but reopen if another question arises on this topic.

Cheers, 
Nathan
National Renewable Energy Laboratory

closed 2021-02-22 by nathanmtom