WEC-Sim/WEC-Sim #517: How to compute A_inf(added mass at infite frequency) in WEC-Sim?

linmeihong, 2021-02-21:
How to compute A_inf(added mass at infite frequency) with parameters(Tp, Hs, start frequency and end frequency) in WEC-Sim? thank you.

nathanmtom, 2021-02-21:
@linmeihong Thank you for your question and will try to answer your question as best I can. 

The added mass at infinite frequency will not have any frequency variation and will be a constant matrix that is dependent on your body geometry. The parameters Tp, Hs are used to define the wave spectrum used when simulating irregular seas. When you mention start frequency and end frequency are you denoting the bounds from the hydrodynamic data or the wave spectrum? 

The calculation of the added mass at infinite frequency is normally handled by the BEM software, i.e. WAMIT, NEMOH, AQWA. For certain geometric shapes this value does have an analytical formulation, but if you have a custom geometry this will likely need to be calculated via a BEM software. 

Please let us know if this has answered your question, if not please respond to our points above and provide additional details about your question and request. 

Cheers, 
Nathan 
National Renewable Energy Laboratory

linmeihong, 2021-02-22:
> @linmeihong Thank you for your question and will try to answer your question as best I can.
>
> The added mass at infinite frequency will not have any frequency variation and will be a constant matrix [...]

Thank you for your reply.  According to your guidance I find the formula from NEMOH. thanks for your help.
In the NEMOH, use FD2TD.m to compute A_inf, in the function, it use the average value of all the omegas. my previous opinion is just use one omega value such as 2*pi/Tp. 
the formula is A(Inf) = A(w) + 1/w ∫^Inf_0 K(t) sin(w * t)dt.
in the function, A(Inf) =sum [A(w_i) + 1/w_i ∫^Inf_0 K(t) sin(w_i * t)dt]/num_of_w.

But I still have one doubt. just for a discussion.Maybe I don't understand enough about the theory. may be the result is the same(with some small error) .
in the computation of irregular wave , Start frequency and end frequency are use inthe generation of frequency array. and NEMOH provides FD2TD.m for computation of A_inf,  in the function, it also has a frequency range. I have some doubt when the range use in WEC-Sim is different with the range in FD2TD.m. Should the A_Inf need to Recalculate?  Though the range is use as approximation of ∫^(Inf)_0 , such as: Start frequency and end frequency [0.5, 2.0] may be a sub-range of [0.1,3.0] is use for computation of irregular wave, and [0.1,3.0] is the BEM model compute frequency range. 

Thank you for your help, I have got loss on this for a long time. now I understand how to compute the A_inf value.
FD2TD.m from NEMOH is show bellow
--------------------------------------------------------------------------------------------
% 
% --> function [K, Mu]=FD2TD(w, A, B, T)
%
% Purpose:  Calculation of radiation coefficients in time domain from
%           frequency domain coefficients using Ogilvie's formulae
%
% Inputs :
% - w   : Frequency vector
% - A   : Added mass coefficient
% - B   : Damping coefficient
% - T   : Time vector
%
% Outputs :
% - K   : Retardation function K
% - Mu  : Added mass
%
% Copyright Ecole Centrale de Nantes 2014
% Licensed under the Apache License, Version 2.0
% Written by A. Babarit, LHEEA Lab.
%
function [K, Mu]=FD2TD(w, A, B, T)
clear K Mu;
n=length(T);
dt=T(2)-T(1);
nw=length(w);
dw=w(2)-w(1);
K=zeros(n,1);
for j=1:n
    for k=1:nw
    	K(j,1)=K(j,1)+B(k)*cos(w(k)*T(j))*dw;
    end;
end;
K=K*2./pi;
CM=zeros(nw,1);
Mu=0;
for k=1:nw
    for j=1:n
    	CM(k,1)=CM(k,1)+K(j)*sin(w(k)*T(j))*dt;
    end;
    CM(k,1)=A(k)+CM(k,1)/w(k);
    Mu=Mu+CM(k,1);
end;
Mu=Mu/nw;%%%average value of all omega
for k=1:nw
    CM(k,1)=Mu;
end;
figure;
plot(w,B,'r',w,A,'b',w,CM,'b.');
xlabel('frequency (rad/s)');
ylabel('FD coefficents');
figure;
plot(T,K,'r');
xlabel('Time (s)');
ylabel('Retardation function');
%fid=fopen('Kt.dat','w');
%for j=1:n
%    fprintf(fid,'%E %E %E %E %E %E %E \n',T(j),0.,0.,0.,0.,K(j),0.);
%end;
%close(fid)
end
------------------------------------------------------------------------

dav-og, 2021-03-01:
Hi @linmeihong,

Thank you for bringing this to our attention. At the moment when reading Nemoh results, BEMIO will just take the final added mass value in the series and define A(inf) as this value:

https://github.com/WEC-Sim/WEC-Sim/blob/cd40499eab6f6cccfd4b3c66afebb2bd1a54fe3c/source/functions/BEMIO/Normalize.m#L52

We usually advise people to select a sufficiently large frequency range to allow the added mass results to converge, and this hasn't been identified as a significant source of error for people using Nemoh-generated hydrodynamic coefficients in WEC-Sim. However, the more robust approach would be to implement the equations you've shared above - we'll look into it! 👍

kmruehl, 2021-04-07:
relates to #534

nathanmtom, 2021-04-21:
Refer to Pull Request #534 for code changes to address the question in this issue.

closed 2021-04-21 by nathanmtom