WEC-Sim/WEC-Sim #518: NEMOH to WECSim

okhsallam, 2021-02-21:
Dear all 

I am trying to use NEMOH to get the required files for  WECSim

All files are already done except  (Hydrostatic.dat) and (KH.dat) these 2 files do not appear in the MESH after the NEMOH solver  and postprocessor finish. 

Can you please help how I can generate these 2 files via any opensourse codes 
I wrote to the NEMOH representatives to help for the issue but I did not get any reply ? 

Thanks in advance

dav-og, 2021-02-24:
Hi @okhsallam,

You can use the Nemoh 'Mesh' programs: https://lheea.ec-nantes.fr/valorisation/logiciels-et-brevets/nemoh-mesh

Or you can use Meshmagick to compute these data:
https://github.com/LHEEA/meshmagick

dav-og, 2021-03-10:
Hope that answered your question - please re-open if not.

closed 2021-03-10 by dav-og

okhsallam, 2021-03-13:
Thanks a lot for your reply
Yes this helped me