WEC-Sim/WEC-Sim #519: PTO actuator assistance: model won't stay still

t-osu, 2021-02-22:
Hello,
I have a 3 part system, a nacelle with a heave plate and a forward and aft float connected to it with 2 rotational PTOs. I am trying to stabilize the nacelle heave plate system such that the heave plate does not move at all in pitch/roll. I tried putting a PTO rotational actuator block in between the 6DOF constraint and where the aft/float PTOs connect to the nacelle rigid body. I set the position equal to a constant 0, but this does not make a difference. I then broke the link in the nacelle rigid body so I could edit it and implemented a PID system for the pitch DOF in displacement to follow a 0 reference signal and outputted the control signal to the position input of the PTO actuator and this seemed to have some affect, but the nacelle heave plate system still was pitching/rolling even though the position control signal is accomplishing this (it is basically the opposite of the input measured signal).

Also, from what I've read, I think breaking the link means the block won't reference the library, but that would only be an issue when WEC-Sim library's are updated though, right?

Any ideas on stabilization?
Thanks,

- Chris

akeeste, 2021-02-23:
Hi @t-osu,

Thanks for your question, I will do my best to address it.

For your latter question first, yes breaking the link means that the block won't reference the library. This should only be an issue if you pull a library update. In that case, you may be able to get around this by copying your changes, restoring the link to update the blocks, breaking it again and pasting in your old changes. I would not recommend this workflow long term, but it may provide a workaround while working through this issue.

For the main part of your question, I have a hard time picturing your set-up. Can you uploading a screenshot of your block diagram or the case directory? Based on your description, it seems like there may be an issue with how the fore and aft floats are constrained to the nacelle. You should be able to use a position controller with the rotational-actuator PTO block without breaking the rigid body, so maybe some axis of the constraints is incorrect.

Adam

t-osu, 2021-02-23:
Morning @akeeste 

Gotcha, thanks for the confirmation on links.

I attached what the Simulink setup looks like with the PTO actuator block. I don't have it setup where the aft and fore PTOs are (i.e. instead of pto1 and pto2 have them both be actuator blocks) because I don't want the floats to have a 0 position, I want the nacelle/heave plate to, which I took as placing the actuator PTO where it is now. But I would appreciate you taking a look, I could have missed some constraint or axis setup like you commented. 
Let me know if there is anything else I can do to help!
[image]

akeeste, 2021-02-23:
Hi @t-osu 

Thanks for the diagram! Based on this, I think part of this issue is about which blocks are being constrained, and possibly the constraint locations. In this example, the nacelle has a 6DOF constraint relative to the seabed, so it is free to pitch/roll and move in any dof. There is also the rotational actuation pto, but the PID input does not define the nacelle's motion. It will constrain the followers (floats) rotation relative to the base (nacelle), so it will only prevent the floats from spinning in a certain degree of freedom. 

Are you able to upload the input file and simulink file? I think this would be the easiest way for me to look at the locations, axes and cgs of everything. If not, can you give the 4 constraint locations, 3 body cgs and 3 constraint axes?

t-osu, 2021-02-24:
Hm that's interesting. Yeah I always got confused with the "B" and "F" in the blocks due to my lack of knowledge with multi-body (that is ever growing). It is also interesting because when I had it set to a constant value of 0 instead of command_p_d the floats were not staying at a 0 position. 
Sure, here are the files. Would love to know any details on what you find and why it is causing an issue just so I understand everything.
[MECC.zip](https://github.com/WEC-Sim/WEC-Sim/files/6038576/MECC.zip)

akeeste, 2021-02-25:
Hi @t-osu 

I took a look at your case and from how the pto's are placed right now, I don't think there will be much resistance to the nacelle pitching. 

The rotational ptos are located at, respectively:
pto_aft.loc = [0 0 -0.03];
pto_fore.loc = [0 0.01 -0.03];

However the axis of each is not defined, so it defaults to allowing rotation in +y = [0 1 0]. This allows the nacelle to rotate with some small damping about the [0 1 0] axis at [0 0 -0.03]. Even if the hydrodynamics of the floats were such that they were static (and therefore supportive), if the nacelle is unstable in roll/pitch it may slowly roll as there is no stiffness.

Without more details on the case, the physical set-up you're modeling, and seeing the spinning behavior, it is difficult for me to say how exactly to fix this issue. My recommendation is to experiment with the constraint/float placement. Especially, look at varying placement in x to support the nacelle in pitch, vs the current variation in y to support in roll. Also, look at applying some small stiffness to the rotational ptos and see how this affects behavior.

Adam

akeeste, 2021-03-03:
Hi @t-osu,

I just wanted to check in and see, has your issue been resolved? I will close the issue in a week if so. If not, feel free to respond or reopen the issue and I can discuss with you further.

Adam

t-osu, 2021-03-05:
@akeeste apologies, the information you provided above was insightful, it's been a busy few weeks so I haven't been back on the help page in a while. We are trying out a few different things and your suggestions, thanks!

closed 2021-03-05 by t-osu