WEC-Sim/WEC-Sim #521: Clarification on wave direciton

t-osu, 2021-02-25:
Hello all,

I have a quick question. I know that +x is 0 degrees, but does the degree increase in the _clockwise_ or _counter-clockwise_ direction? So if I were to put in +90, would that be (based on the global coordinate system) +y or a -y direction?
Can we assume also that +x would be East ?
I have data from wave roses and I'm trying to determine how to place that information into WEC-Sim.
thank you!

- chris

nathanmtom, 2021-02-25:
@t-osu Thank you for your question and will do my best to answer your question. 

At a zero degree wave heading, as you have mentioned, the waves travel from the -x-axis to the +x-axis. The wave direction, \Beta, is then defined positive by rotating counter clockwise about the z-axis. Therefore, a wave direction of +90 would have the waves traveling from the -y-axis to the +y-axis. A useful plot of this definition can be found in the WAMIT user manual, **[here](http://www.wamit.com/manualupdate/history/V70_manual_old.pdf)**, on the 41st page of the .pdf or on page 4-10.

The definition of East, or any compass direction, is dependent on how you decide to orient your floating body. Therefore, the +x-axis would not always be associated with East. 

If you have wave rose data, then you should just need to take that direction data and covert it to an incident angle relative to the orientation of the floating body. Once you know this convention you should be able to input the right wave direction within WEC-Sim. 

I hope this has answered all of your questions, but let me know if I need to clarify any of my statements above. 

Cheers, 
Nathan 
National Renewable Energy Laboratory

nathanmtom, 2021-03-03:
@t-osu I wanted to check-in to see if our response above satisfies your question?

t-osu, 2021-03-04:
Yes Nathan the item about the wave rose helped a lot, thank you!

closed 2021-03-04 by t-osu