WEC-Sim/WEC-Sim #522: Wave Direction MCR

ckberinger, 2021-02-25:
Hello,

I have a few goals to analyze effects of wave direction on the OSWEC.

1. I am trying to run regular waves from 0 degree wave direction to 90 degrees. I want to run one direction at a time and use the MCR function. I get the error below. Seems like the MCR function does not have the ability to run through different wave directions? How can I edit the code to make this possible? Or can I use MCR to rotate the OSWEC as needed and keep wave direction at 0 degrees?
"Unable to perform assignment because the left and right sides have a different number of elements.

Error in bodyClass/regExcitation (line 471)
                    obj.hydroForce.fExt.re(ii) = interp2(X, Y, squeeze(re(ii,:,:)), w, waveDir);"

2. I also want to look at waveSpread effects on power production in irregular waves. I understand I have to specify a list for waveDir and specify the percent in waveSpread for each direction.  What exactly is this doing? Making an x percent of the waves come from a theta direction over the time series? Where does the cos squared spreading come in? I think my plan at the moment is to run irregular waves with 0 spreading, spreading equally distributed from 0-45 degree, and spreading equally distributed from 0-90 degrees. I have a feeling there is not a way to use the MCR function to run different sets of wave spreadings? 

Thank you,
-Courtney

dforbush2, 2021-03-01:
Hi @ckberinger 

This is definitely something that should work, as per the examples: https://github.com/WEC-Sim/WEC-Sim_Applications/tree/master/Multiple_Condition_Runs.

Depending on your objective, you may also consider using the passive yaw option https://github.com/WEC-Sim/WEC-Sim_Applications/tree/master/Passive_Yaw 

In the short term: a quick fix may be to reload the *h5 file every iteration by setting simu.reloadH5Data =1 in your input file.

If this doesn't work: I think it would be easiest for me to help you if you could share your code? A *zip containing your simulink model, wec-sim input file, any associated MCR files, and the *h5 file containing your hydrodynamic data would be the easiest.

ckberinger, 2021-03-01:
Hi @dforbush2 ,

Looks like those examples don't use waveDir like I am trying to, but they do use direction via a spectrum? I don't know what the three columns of the waveSpectrum files are but I assume wave height, period, and heading? 

I think I am eventually going to use passive yaw but want the static case first. 

I tried to reload the H5 data using the line you provided and that did not work unfortunately. I still get the error below.

"Unable to perform assignment because the left and right sides have a different number of elements.
Error in bodyClass/regExcitation (line 471)
                    obj.hydroForce.fExt.re(ii) = interp2(X, Y, squeeze(re(ii,:,:)), w, waveDir);"

Here is my attached zip folder, thank you!
[OSWEC PTO damping direction 3.zip](https://github.com/WEC-Sim/WEC-Sim/files/6063948/OSWEC.PTO.damping.direction.3.zip)

dforbush2, 2021-03-02:
Ah I see. Unlike waves.T and waves.H, there is not a flag in wecSimMCR for waves.waveDir. So to run this as a wecSimMCR, you'll want need to use a different approach, as the wecSimMCR tries to run waveDir of length > 1 as a vector, rather than iterating one at a time. 

I agree this is confusing, I'll see what I can do about it in the documentation. In the meantime, you can run this through the mcrFile approach detailed in the applications case link documented above. 

You'll want to first add to your wecSimInputFile:

`simu.mcrCaseFile = 'mcrExample.mat';`

and you'll also want to set waves.waveDir and pto.c to be length=1. Doesn't matter what values you use, they will be overwritten.

And if I understand your intent, you want to run a series of PTO.c in various wave directions. If thats correct, you can do something like this in the command line

```
mcr = struct();
mcr.header{1,1} = waves.waveDir;
mcr.header{1,2} = pto.c;
mcr.cases(:,1) = [0; 45; 90; 0 45; 90; 0; 45; 90];
mcr.cases(:,2) = [0.5; 0.5; 0.5; 1; 1; 1; 10; 10; 10];
save('mcrExample' , 'mcr');
```
or whatever your values are. Then when you run wecSimMCR, it will reference this mcr case file and should do what you want. This will also work with passive yaw turned on. Hopefully this addresses your issue!

ckberinger, 2021-03-03:
Thank you! This code and the MCR application example were a great help. I got it all working!

closed 2021-03-17 by kmruehl