WEC-Sim/WEC-Sim #523: How to define PTO stiffness & damping

timmaflanagan, 2021-02-28:
Hello! I am trying to define PTO stiffness and damping coefficients for a fixed OSWEC. Could you please explain what the difference is between the stiffness and damping coefficients are and how wecSim interprets them? Do you have any resources with experimental data for the optimal PTO stiffness and damping coefficients?

Thanks,

Timma

sugarl14, 2021-03-01:
To WEC-Sim Team: If any of my response has incorrect / misleading information, please correct me!

Hey Timma, 

Let me first say that I am by no means an expert. I'm still learning like you. But here are some things that helped me wrap my brain around it a little better:

`Hello! I am trying to define PTO stiffness and damping coefficients for a fixed OSWEC. Could you please explain what the difference is between the stiffness and damping coefficients are and how wecSim interprets them?`

`pto.k` = PTO Stiffness (Spring stiffness): In MTALAB, stiffness is the torque required to rotate the OSWEC by a unit angle. The higher the stiffness, the more wave force is required to move the OSWEC. Think about it this way: You can easily compress the spring of a pen with your fingers because it has a small stiffness. But you probably can't compress the springs of a car's suspension since it has much higher stiffness. 

`pto.c` = PTO Damping: In MATLAB, this is the torque required to maintain a constant joint privative angular velocity between the base and its following frames. Dampers really just limit the amount of oscillation. High damping= not much oscillation. Low damping= a lot of oscillation. Here's an example: For an over-damped system, think about the spring of a pen. If you compress it and let it go, it'll just shoot out, but wont really go "boing" a lot. However, let's say you have a slinky and you let it go. The slinky will bob up and down for a while since it has a low damping value. It has a lot of "boing" in it-- this is an underdamped system. 

Let me also clarify that the slinky example is one with low spring stiffness and low damping.

Here's a pretty helpful tool for visualizing spring damping and stiffness: https://www.khanacademy.org/computing/pixar/simulation/hair-simulation-101/a/mass-spring-system

The PTO is what converts the motion of the OSWEC into power. So, it's important to optimize the stiffness and damping values. A lot of work has been done experimentally to determine the optimum damping and stiffness values, but sometimes we're just left to figure it out ourselves. Luckily WEC-Sim has a neat feature called wecSimMCR ([webinar](https://www.youtube.com/watch?v=C9R_mGNI5yA)) where we can find which combination of damping and stiffness values lead to the highest power capture at a given sea state. 

I had a similar question with PTOs in Issue #515 that may help you out a little too
 
 
 

` Do you have any resources with experimental data for the optimal PTO stiffness and damping coefficients?`

As far as resources, try checking out Chapter 6 of the NumWec Project. Also, if you go to Issue #515 and scroll down, @nathanmtom provided 2 publications that have great explanations. 

Hope this helps a little! :)

timmaflanagan, 2021-03-01:
Wow! This is fantastic, thanks so much! I saw some of your other posts, and
questions on PTO stiffness and found it really helpful to look through that
information!

Thanks again. I will look at all of these resources you've provided!

Hope you have a great day!

nathanmtom, 2021-03-01:
@timmaflanagan Thank you for your question and @sugarl14 thank you for providing a response to this issue. Your response is well-written and thank you for linking to #515 as the discussion with this issue is very similar to that thread. 

I will only add that when using `pto.k` and `pto.c` the PTO blocks will then calculate a resistive force (or torque) based on the following feedback loop: force = `-pto.k`*X - `pto.c`*d/dt(X), where X is the displacement (position) of the mode of motion. Note that the inclusion of a `pto.k` will result in some energy being injected back into the waves. Therefore, your PTO will need to be able to have bi-directional energy flow to implement this control in a laboratory or open water scenario. 

Lastly, the optimal PTO stiffness and damping coefficients are strongly dependent on the WEC system you are modeling. Therefore, additional details would need to be provided to see if there is an analytical solution for the optimum damping and stiffness values or you will need to complete a parameter search such as @sugarl14 has suggested. 

Please let us know if you have more questions on this topic? If not, then we ask that you close this issue and reopen later if you have more questions.

Cheers, 
Nathan
National Renewable Energy Laboratory

timmaflanagan, 2021-03-02:
Thank you both!

This is very helpful to know :)

closed 2021-03-02 by timmaflanagan