WEC-Sim/WEC-Sim #525: Does the Excitation force depende on the Position of the floater?

Liam-Guerrero, 2021-03-03:
Hi Guys, 
This is more a theoretical question, and I wonder if I can get any help:

I am using the model used for WECCCOMP for some case studies, but I am having some trouble to understand some concepts:

From what I understand, under linear theory, the excitation and radiation forces are computed using BEM for the specific shape and sea conditions. So the radiation and excitation forces do not depend on the actual position of the float? 

Do the radiation, excitation force are the same independently if the PTO is exerting force? I mean, those forces are the same does not matter if the float/arm are being controlled or moving freely?

Should the excitation moment be in phase with the wave elevation?

Any help is welcome!

Cheers, 
Juan

nathanmtom, 2021-03-03:
@Liam-Guerrero Thank you for your questions and I'll do my best to answer your questions. 

1. You are correct that under linear hydrodynamic theory the excitation and radiation forces are computed from a BEM solver that assumes the oscillating body is at its equilibrium position in the water. This position is defined by the mesh and where the user tells the BEM where on the solver to output the global forces and torques. 
2. The radiation force coefficients are calculated assuming small displacements from the equilibrium position of the body. Therefore, the radiation coefficients do not update based on the instantaneous position of the body in the water. 
3. The excitation force coefficients are calculated assuming the body is held in place and sinusoidal waves impact the body. These forces also will not update based on the instantaneous position of the body. 
4. The radiation forces are dependent on the acceleration and velocity of the body and so if you control the float/arm to move more or less the radiation forces will correspondingly more or less. The excitation force will be independent of the PTO exerting a force since this is related to the pressure generated around the body because of incident waves. 
5. I'll need you to clarify where you are defining the excitation moment? For the float or about hinge A in the WECCCOMP set-up?

I hope these answers have helped shed light on your questions. But let us know if you still have questions. 

Cheers, 
Nathan

Liam-Guerrero, 2021-03-04:
Hi Nathan, 

Yes, it did help me to understand a bit more about the radiation/excitation forces. 

Is there any analytical limit for the displacement of the body for which linear hydrodynamics theory can no longer be considered? For example, for the wecSim model for WECCCOMP, in the competition, Hm0 = 2,08 / 6.25 / 10.42 cm could be considered under linear theory, but what about 15 cm or 20 cm?

I'll need you to clarify where you are defining the excitation moment? For the float or about hinge A in the WECCCOMP set-up?
R/ excitation moment defined about hinge A. What I wanted to clarify is if the excitation force on the float (which produce an excitation moment about hinge A) has to follow the wave elevation. Here the excitation moment is computed considering the excitation forces on the float in the Z and X direction and the excitation moment for the float about Z-axis. 

[image]

Finally, I still have a doubt about the direction of the radiation/excitation forces in general. For example, for a generic sphere semi-submerged in water, shouldn't both forces point in the same direction? 
That is how I thought before, but I found one of our papers ('Experimental Confirmation of Nonlinear-Model-Predictive Control Applied Offline to a Permanent Magnet Linear Generator for Ocean-Wave Energy Conversion', 2015) in where the radiation force is in opposite direction as the excitation force, and I do not quite understand the reason for that. 

[image]

Thank you much for your help, it really help me.

Cheers
Juan Guerrero

nathanmtom, 2021-03-04:
@Liam-Guerrero Thank you for your response and happy to try and add more context and explanations about my response. 

1. In response to when linear theory is valid, in my opinion this is dependent on the WEC system and also on what information you are hoping to obtain from your simulation. The more accurate of a time history trace the larger response amplitudes opens more opportunities for nonlinearities to be present but if you are looking for statistics across a sea state then you might not see as a large of an influence. For the WECCCOMP specifically, the WEC-Sim model and results from the competitors appear to show that linear hydrodynamics did a sufficient job for this application. 
2. Since the float is the only wave-excited body in this WEC-Sim model let us focus there. Since the surface piercing float is similar to a point absorber, then the heave excitation force for most sea states is likely to be in phase with the wave elevation height. The surge and pitch wave-excitation force and torque is likely to be out of phase with the wave elevation and even then this is more related to the largest length of your device relative to the wave length. I have not done my own checks, but since the x-lever arm to the hinge is larger than the z-lever arm and generally the heave force is probably larger that the surge force. In this case when you look at the excitation moment about the hinge, since it is driven mostly by the heave excitation force on the float, it will mostly be in phase with the wave elevation. 
3. There are nuances when trying to generalize this question to all WEC systems, but for the schematic you have shown here the directions are correct. In this snap shot the WEC is moving upwards vertically, there is a wave crest at the center of the body which results in a positive heave-excitation force, and the heave radiation forces work opposite to the direction of motion so they would be pointing down. In general radiation forces are resistive forces which means they act to resist the motion of the WEC (not necessarily just a damping force). 

I hope my responses above have either answered your questions or at least helped clarify some of the topics under discussion. 

Cheers, 
Nathan
National Renewable Energy Laboratory

nathanmtom, 2021-03-10:
@Liam-Guerrero I wanted to check in to see if your latest questions were answered with my response above?

Cheers, 
Nathan
National Renewable Energy Laboratory

Liam-Guerrero, 2021-03-10:
Yes, definitely It really helped me. 
Thanks for the support.

closed 2021-03-10 by Liam-Guerrero