WEC-Sim/WEC-Sim #527: Changing water depth in WEC-Sim

SofiasKonstantinos, 2021-03-04:
Hi everyone , 
I am currently modeling different sea states with the MCR option on the RM3 model . For the purposes of my project i want to change the water depth from infinite to 25m . Unfortunately i **do not** have access to a BEM solver since the input files for WAMIT and NEMOH are available so i could run this configuration myself . 
Thank you in advance 
KS

akeeste, 2021-03-04:
Hi @SofiasKonstantinos,

In the WEC-Sim [BEMIO examples](https://github.com/WEC-Sim/WEC-Sim/tree/master/examples/BEMIO) we do include the input files and mesh for the RM3 in several different BEM codes. If you do not have access to a WAMIT license, I would recommend altering the input files for Nemoh or Capytaine, both of which are open source. Once either code is installed, it should be straightforward to modify the water depth and rerun the RM3 at your desired depth.

Unfortunately, we cannot run any cases and are fairly limited in the support we can give on these BEM codes. Please let me know if this addresses your issue.

Thanks,
Adam

SofiasKonstantinos, 2021-03-06:
@akeeste , 
I hope you are doing well . I installed bemio and python and now i am trying to alter the water depth. So for start i went in the bemio master file , i opened tutorials/nemoh and then by using the command in cmd , python run.py i was able to get the programma going . But when i went into the data file , one_body/Nemoh.cal and changed the water depth from there i got this error .
[image] 
What am I doing wrong ?

SofiasKonstantinos, 2021-03-06:
Is using the write HDF5 option any different ?

akeeste, 2021-03-08:
Hi @SofiasKonstantinos 

I'm sorry, it is difficult to tell what your problem with running Nemoh is, and I cannot provide direct support as we are not developers of the code. I would recommend using Capytaine if you are familiar with Python as there are also BEMIO examples and the developer is more active in maintaining the code.

You are not able to simply change the depth after running BEMIO in the hydro struct, or with the write_HDF5 functions. In the BEM code, the water depth affects the calculated Green's functions and potential, and therefore the hydro coefficients for the body. Forcing the depth to change in WEC-Sim will only affect the wavenumber calculated for the period and wave height. This wavenumber is input to the nonlinear hydro options, but the body will still be missing the correct hydrodynamic data.

SofiasKonstantinos, 2021-03-13:
Dear @akeeste , 
Your guidance was very helpful but i have some questions while i dive deeper into the solver . 

1. The first question is about the RM3.py . So i ran the solver from my Ubuntu laptop , downloaded capytaine and everything but i get this error (file attached below). I also got this error from windows on my PC but solved it by taking the call_capytaine.py into the RM3 folder directory. Doing the same thing in Ubuntu didnt solve it . This question is mostly for educational purposes if you happen to know. 
[image]
[image]

2. When succesfully running the BEM solver in windows i get this message , which i suppose is ok . 
[image]
So then after that i take this .nc file that was produced by capytaine and paste it into my matlab hydrodynamics working folder . I am going to also attach the RM3.py file that i used because from there i changer the depth and the wave frequencies . 
[image]
When i do this and then try to run BEMIO i get this error :
[image]
Thank you very much for your time , your advice helps me proceed with my project

SofiasKonstantinos, 2021-03-20:
Everything works fine now . Thank you so much @akeeste for your help !

closed 2021-03-20 by SofiasKonstantinos