WEC-Sim/WEC-Sim #528: Specifying damping in MCR option 2?

sugarl14, 2021-03-04:
Hi!

I have created an excel file that contains a JPD plot of wave height, period, and freqency of occurrence. (Similar to what is spoken at 11:30 of this video https://www.youtube.com/watch?v=C9R_mGNI5yA).

I was wondering if there was a way that I could also specify my optimum  `pto.c` values for each wave period in this excel file? 
i.e., when `T= 1.5`, I want `pto.c= 6000`. Or  when `T= 8.5` I want `pto.c = 20000`. (Also, I'm just using arbitrary numbers for `pto.c` here)

Here's a screenshot of what I have / what I'm trying to do:

[image]

thanks!!

sugarl14, 2021-03-04:
On second thought, couldn't I just use MCR option 3 and create a .mat file that looks something like this?

[image]

yuyihsiang, 2021-03-10:
Yes, that is the best way to do it.

—Yi-Hsiang

closed 2021-03-17 by kmruehl

hossein-mehdipour, 2021-07-06:
Hello @sugarl14 , excuse for bothering you, I had a question about this issue & I was wondering if you could help me
I was wondering when you use option 3, you no longer can specify percentage of occurrence of any sea states. Doesn't this influence the results of your simulations? Or its effects negligible? 
Or you just repeat the rows concerning sea states that occur multiple times?

sugarl14, 2021-07-06:
Hey @hossein-mehdipour, great question. I'll do my best to answer but I'd like to have a WEC-Sim expert verify any answer I give. So take what I say with a grain of salt :)

Option 3, as far as I am aware, only accounts for `waves.H`,  `waves.T`, `pto.c`,  and `pto.k`, so you are correct that option 3 doesn't include the percentage of sea state occurrence. 

Here's what I did for my work (which may or may not be the preferred method) -- 
I used MCR option 3 and to obtain the mechanical power into the PTO for all the seas states being considered. Then I took these values and multiplied them by the percentage of sea state occurrence to obtain my power contribution matrices. 

For example,

Let's say we have `waves.H = 1.5`, `waves.T = 5.5`, `pto.c = 20000`, and 'pto.k= 0`. After running this simulation let's just guess that the average mechanical power into the PTO is something like 5,000 Watts.  But lets say that specific sea state only  occurs 3% of the time , over a 1 year span. We can do 5,000W * 0.03 to get 150W. So this sea state will contribute 150 W of power over a span of 1 year. 

I hope this helps you in some way. Don't hesitate if you have any more questions! I'll be happy to do my best to answer. But again, I recommend having a WEC-Sim guru to double check.

Thanks,

Landon

hossein-mehdipour, 2021-07-07:
@sugarl14 Thank you so much for taking the time to answer, it was an interesting approach that I think makes sense. I'm going to double-check it just in case.
And I assume "Power Contribution Matrices" and "Power Matrices" are not same thing (according to description you just mentioned). Am I right?
Hello @yuyihsiang, I was kindly wondering if you could check the comment above to see if there's anything we are missing.