WEC-Sim/WEC-Sim #529: Doubt in wecSim results and issue in Mechanics explorer

Pisdan96, 2021-03-05:
I'm new to wecSim and I've followed the examples given. I've tried to do a MCR simulation on a sphere and the code ran successfully. However, I faced the following problems.
1. How can I make sure that I got that I got the correct simulation results even if the code ran smoothly?

2. Mean free surface and seabed was shown as a tiny part in the Mechanics Explorer. does that mean my simulation is wrong ?
[image]

akeeste, 2021-03-08:
HI @Pisdan96 

Thanks for asking, I will answer your questions as best I can.

To answer your second question first, your simulation is not necessarily wrong. The domain size (simu.domainSize) defaults to 200m in length and width, so if it is not visible then the .stl file is definitely too large. By default, the body .stl file is only used for Mechanics Explorer for visualization and wouldn't impact the simulation results. However, if you use any nonlinear hydrodynamics option (simu.nlHydro >= 1) then the stl file is used in the force calculation (see [theory manual](https://wec-sim.github.io/WEC-Sim/man/advanced_features.html#nonlinear-buoyancy-and-froude-krylov-excitation)) and the simulation will not be correct.

The stl is likely output in the wrong units (Solidworks often defaults to mm in .stl output). WEC-Sim assumes the unit is in meters. I recommend checking this before proceeding.

In regards to your first question, this is much broader and you can go about this several ways. As a preliminary check, use the Mechanics Explorer visualization and user defined functions to look at the response and force time series for abnormal behavior. For certain geometries, you may find journal references with comparable cases. For example, spheres are often a basic case assessed and several simulations can be found in IEA OES wind task 10 (see applications repo [case](https://github.com/WEC-Sim/WEC-Sim_Applications/tree/master/Free_Decay)).

Beyond these initial assessments, CFD simulation or experimental data should be used to thoroughly validate your WEC-Sim results.

Please let me know if you have more questions on this.
Adam

akeeste, 2021-03-26:
Hi @Pisdan96 

There has been no traction on this in two weeks, so I am closing the issue. If my answer has not resolved your issue, you can reopen it and we will continue our conversation. Otherwise feel free to open a new issue if a different problem comes up.

Adam

closed 2021-03-26 by akeeste