WEC-Sim/WEC-Sim #530: Passive Yaw Application

ckberinger, 2021-03-09:
Hello,

I have a few questions on the passive yaw application.
1. I downloaded the file, ran BEMIO to make the H5 file, and tried to run the "runYawCases". It had an error because it was trying to find the userDefinedFunction in the wecSim source. Once I commented out this line, I think it worked as intended. (I will uncomment when I move away from passive yaw.) But can I get help troubleshooting why it did not bypass the userDefinedFunction section in the source code? 
2. What are the lines at the end of the PTO section of the input file doing? I realize it is defining the z pto as the y pto and the y pto as the z one. But when you look at the powerInternalMechanics it still had power values in the pitch rotation. Are these actually the pto power values in the yaw rotation? The doc ptoClass doesn't have much information on orientation. 
- pto(1).orientation.z=[0 1 0]; % switching so device will yaw
- pto(1).orientation.y=[0 0 1]; % switching so device will yaw

dforbush2, 2021-03-12:
Hi @ckberinger 

Check out the README file on the top level of the Passive_Yaw applications case. There are 2 sub-directories, PassiveYawON and PassiveYawOFF each containing a WEC-Sim model and input file. The example is actually intended to run from the top level directory, using runYawCases.m

1) There are no userDefinedFunctions.m in these sub-directories, so if you have another one on your path, MATLAB will try to run that one instead. And it may or may not be successful, depending on your search path. I often will include a blank "userDefinedFunctions.m" in a directory to avoid this error, but I did not do this in this case.

2). Your instinct is totally correct. The pre-made rotational PTO block we used in the OSWEC model by default rotates in pitch, so the two lines you refer to change this orientation to create the yawing motion in the example here. It is confusing, but as it stands the powerInternalMechanics field refers to the local (un-rotated) coordinate system.

dforbush2, 2021-03-12:
Bear in mind this example is mostly intended to illustrate the impact of the passive yaw implementation and is not intended to be an effective WEC design.

closed 2021-03-12 by dforbush2