WEC-Sim/WEC-Sim #532: Floating Two Degree of Freedom Constraint

7jcapper7, 2021-03-12:
I am trying to create a two-body attenuator WEC with a rotational PTO between the two bodies. I want each body to be able to move in the heave (Z) and pitch (RY) DOF's only. I see there is a standard floating 3DOF constraint available, but I do not want the bodies to move in surge (X), which happens if I use the floating 3DOF constraint. How can I make each body move only in heave (Z) and pitch (RY)?

[image]

ttran18, 2021-03-12:
Hello @7jcapper7,

I did the similar constraint (only heave & pitch) for my simulation. You can actually customize desired DOFs using the existing constraint block in WEC-Sim and SimScape Multibody Joints. The attached file is an example of a 2-DOFs constraint (heave&pitch). Hope that help.

[Constraint_2DOFs.zip](https://github.com/WEC-Sim/WEC-Sim/files/6131521/Constraint_2DOFs.zip)

Cheers,
Toan

7jcapper7, 2021-03-12:
Hello @ttran18 ,

I am having trouble using the file you attached. I am getting the below error message. Are there additional steps I need to take after downloading the file so that it will work?

[image]

ttran18, 2021-03-12:
Hello @7jcapper7 ,

I forgot to disable the WEC-Sim library link. You should able to download and use it from the previous link now.

Best,
Toan

7jcapper7, 2021-03-12:
@ttran18 , your most recent link looks empty. Can you please try again?

ttran18, 2021-03-12:
Hi @7jcapper7 ,

Try this link.
[Constraint_2DOFs_T.zip](https://github.com/WEC-Sim/WEC-Sim/files/6132705/Constraint_2DOFs_T.zip)

Let me know if this works for you.

7jcapper7, 2021-03-12:
That works, thanks.

kmruehl, 2021-03-17:
Thank  you @ttran18 !

closed 2021-03-17 by kmruehl