WEC-Sim/WEC-Sim #539: Linear to Rotary Motion Conversion - Advice

rodrijardim, 2021-03-15:
Hello everyone,

I am working on the RM3 model but I would like to change the PTO to one that has a linear to rotary motion converter combined with a rotary generator. In PTO-Sim there is only the conversion from rotary to linear and I would like to know what would be the best solutions to implement this PTO.

Thanks

jleonqu, 2021-03-18:
Hi @rodrijardim 

In this case I would recommend to implement your own equations for the conversion from linear to rotary motion. Taking as a reference the RM3 case available in the applications repository ( https://github.com/WEC-Sim/WEC-Sim_Applications/tree/master/PTO-Sim/RM3/RM3_cHydraulic_PTO ) , you will need to modify the PTO-Sim subsystem. You can add a subsystem that takes as an input the linear position and speed, and implement your kinematic equations for the motion conversion.

You can use as a reference this comment: https://github.com/WEC-Sim/WEC-Sim/issues/247#issuecomment-740228118
In this comment I explained how the torque in the OSWEC is calculated as a function of the angles.

rodrijardim, 2021-03-19:
Thanks for your help @jleonqu, I will do what you recommended !

closed 2021-03-19 by rodrijardim