WEC-Sim/WEC-Sim #541: Viscous Drag Characteristic Area - WEC-Sim Webinar #2 (nlhydro, non-hydro, b2b)

7jcapper7, 2021-03-17:
I am looking at the nlhydro heaving ellipsoid tutorial in Webinar 2. How were the drag characteristic areas determined? It looks like maybe term 3 is the cross-sectional area, but I don't understand how terms 1 and 5 were determined.

[image]

[image]

yuyihsiang, 2021-03-25:
The characteristic area for the translational directions (1,2,3) is defined by the projected area. Therefore, for heave (direction 3), it is ab*pi. For surge (direction 1), it should be equal to bc*pi. My guess is that the person who created the example, simply just took an estimation and used a value of 25. For pitch, it is not an area, we often use pi*r^5.

Note that you will need both the characteristic area and the Cd value. The webinar is just presenting an example. The actual values are often determined from empirical solutions or CFD simulations.

7jcapper7, 2021-03-25:
For pitch, is it the heave projected area times r^3? 

(pitch characteristic area)=(heave projected area)*r^3

For pitch, which dimension should be used for the characteristic length r? (Would it be  dimension a on the diagram?)

I am trying to simulate a floating, rectangular two-body attenuator with a center hinge and I am trying to determine what the characteristic area should be for pitch for each body based on the heaving ellipsoid example. 

In WEC-Sim, should the pitch characteristic area always be based around the body's COG? In my case, each body is rotating around a hinge, not rotating around the COG.  

[image]

kmruehl, 2021-03-25:
The drag coefficients for the translational DOFs can determined from a shape coefficient:
https://en.wikipedia.org/wiki/Drag_coefficient
https://www.grc.nasa.gov/www/k-12/airplane/shaped.html
 
In reality the Cd value will change with the body's orientation of each translational DOF. But since we don’t handle that directly in WEC-Sim, a way to approximate that is to add drag in rotational DOFs. These values can be used a tuning parameter to account for fluid viscosity, which is not accounted for in BEM coefficients.

7jcapper7, 2021-03-25:
I am still having some trouble understanding the drag characteristic area for the heaving ellipsoid. For pitch, is it the heave projected area times r^3? 

(pitch characteristic area)=(heave projected area)*r^3

For pitch, which dimension was used for the characteristic length r? (Would it be dimension a on the diagram?)

In WEC-Sim, should the pitch characteristic area always be based around the body's COG? In my two-body attenuator case, each body is rotating around a hinge, not rotating around the COG.

kmruehl, 2021-03-25:
For translational DOFs (surge/sway/heave) the area corresponds to the projected area in that DOF, and a shape coefficient for drag. For rotational DOFs (roll/pitch/yaw) there is not a physical projected area for the corresponding DOFs. For rotational DOFs this drag term is used to compensate for the fact that there is fluid viscosity, and this term can be used as a tuning parameter. 

For this particular WEC concept, you can always leave the rotational DOFs values set to the default value of zero, and only apply drag in the translational DOFs for which there is a physical mapping between the characteristic area and the shape parameter. This is an approach that is often used for modeling devices in WEC-Sim. For this case applying drag in the heave DOF will likely be sufficient because rotational about the hinge corresponds to both a translation in heave and a rotational in pitch about the cg of each body. 

http://wec-sim.github.io/WEC-Sim/man/api.html#objects.bodyClass.viscDrag

7jcapper7, 2021-03-25:
In Issue 130, Dr. Yu stated that the rotational characteristic area "should be the corresponding projected area multiplies R^3, where R is a characteristic length." Do you know what the "corresponding projected area" would be for pitch? And which dimension would R be for the heaving ellipsoid (a, b, c)?

https://github.com/WEC-Sim/WEC-Sim/issues/130

kmruehl, 2021-03-25:
Characteristic length is an approximation, so any of a geometry's lengths are justifiable. For an ellipsoid the characteristic length could be either a, b, or c. In [this WEC-Sim ellipsoid application](https://github.com/WEC-Sim/WEC-Sim_Applications/tree/master/Nonlinear_Hydro), r = 5, so either a or b. 

Yes, in #130 @yuyihsiang provided an explanation for using r^5 based on units consistancy, and also provided a reference for how to estimate the drag coefficients and define the characteristic areas:
Cozijn H., Uittenbogaard R., and Brake E. ter, 2005, “Heave, Roll and Pitch Damping of a Deepwater CALM Buoy with a Skirt,” 15th International Offshore and Polar Engineering Conference, ISOPE, Seoul, Korea, pp. 388–395.

https://github.com/WEC-Sim/WEC-Sim/issues/130#issuecomment-253644834

yuyihsiang, 2021-04-07:
@7jcapper7, hope that is helpful. did that resolved your question?

7jcapper7, 2021-04-08:
Yes, that helps, thank you.

closed 2021-04-16 by nathanmtom